kaloz/mwlwifi #70: Low TX power output on WRT1900ac 5GHz

honicky, 2016-03-31:
I have tried both .16 and .17 firmware in the UNI3 bands with the same results:

Changing the txpower using:
```
iw wlan1 set txtpower fixed 0 
```
and 
```
iw wlan1 set txpower fixed 3000
```
has no effect on the actual transmitted power, as measured by a cabled spectrum analyser.  I have confirmed that the driver believes the power has been changed using the command:
```
iw wlan1 info
```
Regardless of the power settings, the power output seems to be about 2dBm per antenna.

This may explain the poor performance some people are seeing (e.g. #64 and #68)

honicky, 2016-03-31:
Interestingly, the bottom UNI1 channel seems to be outputting at -10dBm per antenna and UNI2 at 2-4dBm per antenna.  That could be because of some restrictions on the bottom channel that are not capture in regdb.txt, but are encoded in the firmware

So 

Band |Channel Center| Power
-------|----|------
UNI1 | 5210 MHz | -10dBm/antenna
UNI1 | 5290 MHz | 1dBm/antenna
UNI2 | 5530 MHz | 4dBm/antenna
UNI2 | 5590 MHz | 2dBm/antenna
UNI3 | 5775 MHz | 2dBm/antenna

Obviously Channel 36 is (the top row) is the outlier.

johnnysl, 2016-04-05:
Did you ever test the 2.4ghz channel as well? I'm curious how strong that one is

yuhhaurlin, 2016-04-11:
Power table is loaded according to setting in dts. Mwlwifi driver does not support specific tx power setting now. I will close this issue.

closed 2016-04-11 by yuhhaurlin

honicky, 2016-04-11:
@johnnysl sorry, I didn't test 2.4Ghz, because I'm not using it.  Sorry...