kaloz/mwlwifi #71: Wireless connection problems with a Samsung UE40F6740 and sometimes notebook

mmatuska, 2016-04-05:
Hi I am experiencing wireless connection problems with a Linksys WRT1200AC (Marvel Armada 385 Caiman) to a Samsung UE40F6740 Smart TV or a notebook (Atheros AR9462).
 
What does happen? My Samsung UE40 Smart TV connects to the router and is unable to communicate or reconnect after about one minute while connected. A reconnect from the the TV is not possible, I have to restart the router device. Happens on both 802.11a or 802.11g, encryption WPA2-PSK AES-CCMP. The same thing has happened to my Ubuntu notebook with a Atheros AR9642 chipset but very rarely.

I have tried OpenWRT 15.05.1 and the latest David's build (Bleeding Edge, r49037) using the 10.3.0.17 mwlwifi. On my second router TP-Link TL-WDR3500 (Atheros 9300/9340) the identical configuration works without problems.

How can I help debugging this problem? Where can I look for the cause?
Thanks

johnnysl, 2016-04-05:
Before blaming the driver, i would first properly check your wireless setup.
How does /etc/config/wireless look like?
Are there any neighbours that might be usng the same wireless channel?
Any errors in the kernel log?
And what driver you now exactly using? Are you sure you are using both the latest .17 driver, as well as the firmware??

mmatuska, 2016-04-06:
I am not blaming the driver, I was just following hints I have been told by the OpenWRT people. The main question is why an identical configuration on a TL-WDR3500 does not cause this issue. I am a coder myself (but not a kernel driver developer) so I would really like to help looking into this. I will provide more information later today.

yuhhaurlin, 2016-05-26:
@mmatuska Would you please check latest driver (10.3.0.17-20160523)?

yuhhaurlin, 2016-05-30:
Without update. I will close this one.

closed 2016-05-30 by yuhhaurlin