kaloz/mwlwifi #72: 1200AC - Poor performance on 802.11n 2.4GHz & 5GHz

lachesis, 2016-04-12:
I'm receiving poor performance on 802.11n on my WRT-1200AC. Any 802.11n devices (tested with Lenovo X1 Carbon, X220, and W530, plus RPi3 and RPi2+cheap USB wifi dongle) get < 30Mbps on wireless. However, 802.11ac devices (tested with my girlfriend's Lenovo T440 and my Moto X Pure) are able to get > 100Mbps.

Both 2.4GHz and 5GHz radios seem to be impacted.

I've tried CC 15.05.1, DD r49161, and davidc502's snapshot build.

See below for lots of documentation. I'm happy to add more as needed! Also, there's some more info in [my reddit post](https://www.reddit.com/r/HomeNetworking/comments/4e9jsn/openwrt_cc_wrt1200ac_advancedn_6205_20mbps/d1yday8), when I was still on CC.

Example iperf session, from my laptop `reseune` to my home server (on ethernet) `tron`.
```
reseune ~$ iperf -c tron                                                                               
------------------------------------------------------------
Client connecting to tron, TCP port 5001
TCP window size: 85.0 KByte (default)
------------------------------------------------------------
[  3] local 192.168.60.160 port 44224 connected with 192.168.60.2 port 5001
[ ID] Interval       Transfer     Bandwidth
[  3]  0.0-10.1 sec  23.5 MBytes  19.6 Mbits/sec
```

```
 -----------------------------------------------------
 DESIGNATED DRIVER (Bleeding Edge, r49161)
 -----------------------------------------------------

root@vertigo:~# uname -a
Linux vertigo 4.1.20 #1 SMP Mon Apr 11 20:02:03 UTC 2016 armv7l GNU/Linux
```

```
root@vertigo:~# strings /lib/modules/*.*/mwlwifi.ko | grep "^10.3"
10.3.0.17-20160324
10.3.0.17-20160324
```

```
root@vertigo:~# cat /var/run/hostapd-phy0.conf 
driver=nl80211
logger_syslog=127
logger_syslog_level=2
logger_stdout=127
logger_stdout_level=2
country_code=US
ieee80211d=1
ieee80211h=1
hw_mode=a
channel=36

ieee80211n=1
ht_coex=0
ht_capab=[HT40+][LDPC][SHORT-GI-20][SHORT-GI-40]
vht_oper_chwidth=1
vht_oper_centr_freq_seg0_idx=42
ieee80211ac=1
vht_capab=[RXLDPC][SHORT-GI-80][SU-BEAMFORMER][SU-BEAMFORMEE][RX-ANTENNA-PATTERN][TX-ANTENNA-PATTERN][RX-STBC1][MAX-MPDU-7991][MAX-A-MPDU-LEN-EXP7]

interface=wlan0
ctrl_interface=/var/run/hostapd
ap_isolate=1
disassoc_low_ack=1
preamble=1
wmm_enabled=1
ignore_broadcast_ssid=0
uapsd_advertisement_enabled=1
wpa_passphrase=XXXXXXXXXX
auth_algs=1
wpa=2
wpa_pairwise=CCMP
ssid=vertigo
bridge=br-lan
wpa_key_mgmt=WPA-PSK
okc=0
disable_pmksa_caching=1
bssid=XX:XX:XX:XX:XX:91
```

```
driver=nl80211
logger_syslog=127
logger_syslog_level=2
logger_stdout=127
logger_stdout_level=2
country_code=US
ieee80211d=1
hw_mode=g
channel=6

ieee80211n=1
ht_coex=0
ht_capab=[LDPC][SHORT-GI-20][SHORT-GI-40]

interface=wlan1
ctrl_interface=/var/run/hostapd
ap_isolate=1
disassoc_low_ack=1
preamble=1
wmm_enabled=1
ignore_broadcast_ssid=0
uapsd_advertisement_enabled=1
wpa_passphrase=XXXXXXXXXX
auth_algs=1
wpa=2
wpa_pairwise=CCMP
ssid=vertigo
bridge=br-lan
wpa_key_mgmt=WPA-PSK
okc=0
disable_pmksa_caching=1
bssid=XX:XX:XX:XX:XX:90
```

yuhhaurlin, 2016-04-12:
What is the number you got when tested with stock firmware under same test environment? Thanks.

lachesis, 2016-04-12:
I will flash back to stock and give it a try tonight.

lachesis, 2016-04-12:
Under stock `FW_WRT1200AC_1.0.4.167471_prod.img`:

http://www.speedtest.net/result/5246207247.png
http://www.dslreports.com/speedtest/3594248.png

I'll flash back to DD and repeat those two tests after I post this.

lachesis, 2016-04-12:
Using DD:

http://www.speedtest.net/result/5246223844.png
http://www.dslreports.com/speedtest/3594327.png

Dramatically slower. :(

yuhhaurlin, 2016-04-14:
I will build updated openwrt for WRT1200AC and test with 11n client later. I wonder if anyone encounters the same problem?

johnnysl, 2016-04-15:
@yuhhaurlin, seems to be the same problem as with the V2. If you check the thread over at the openwrt forum, you'll see that many are complaining about the ~20mbit ceiling.

yolanjoy, 2016-05-09:
I see the same throughput problem with the latest Stock firmware. The issue exists on both bands but it's predominant on 2.4GHz band. Suspect the Marvell driver issue.  

[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-1.00   sec  2.25 MBytes  18.9 Mbits/sec
[  4]   1.00-2.00   sec  1.00 MBytes  8.39 Mbits/sec
[  4]   2.00-3.00   sec  3.00 MBytes  25.2 Mbits/sec
[  4]   3.00-4.00   sec  2.25 MBytes  18.9 Mbits/sec
[  4]   4.00-5.00   sec   640 KBytes  5.24 Mbits/sec
[  4]   5.00-6.00   sec  1.00 MBytes  8.39 Mbits/sec
[  4]   6.00-7.00   sec   512 KBytes  4.19 Mbits/sec
[  4]   7.00-8.00   sec   768 KBytes  6.29 Mbits/sec
[  4]   8.00-9.00   sec   640 KBytes  5.24 Mbits/sec
[  4]   9.00-10.00  sec  2.62 MBytes  22.0 Mbits/sec
- - - - - - - - - - - - - - - - - - - - - - - - -

bbogey, 2016-05-16:
I have also the issue with wrt1900acs. With both bands.

Kicking out any client will resume the speed.

Macbook-Jukka:Documents jukka$ ./iperf301  -c 192.168.1.33 -i 1  -p 5201
Connecting to host 192.168.1.33, port 5201
[  4] local 192.168.1.14 port 49903 connected to 192.168.1.33 port 5201
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-1.02   sec  4.00 MBytes  33.0 Mbits/sec              
[  4]   1.02-2.08   sec  3.00 MBytes  23.7 Mbits/sec              
[  4]   2.08-3.01   sec  4.00 MBytes  36.3 Mbits/sec              
[  4]   3.01-4.00   sec  3.88 MBytes  32.6 Mbits/sec              
[  4]   4.00-5.19   sec  3.88 MBytes  27.4 Mbits/sec              
[  4]   5.19-6.04   sec  3.50 MBytes  34.6 Mbits/sec              
[  4]   6.04-7.01   sec  3.75 MBytes  32.2 Mbits/sec              
[  4]   7.01-8.00   sec  3.88 MBytes  32.8 Mbits/sec              
[  4]   8.00-9.02   sec  4.00 MBytes  33.0 Mbits/sec              
[  4]   9.02-10.01  sec  3.75 MBytes  31.6 Mbits/sec              
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-10.01  sec  37.6 MBytes  31.5 Mbits/sec              sender
[  4]   0.00-10.01  sec  37.2 MBytes  31.2 Mbits/sec              receiver

Macbook-Jukka:Documents jukka$ ./iperf301  -c 192.168.1.33 -i 1  -p 5201 -R
Connecting to host 192.168.1.33, port 5201
Reverse mode, remote host 192.168.1.33 is sending
[  4] local 192.168.1.14 port 50056 connected to 192.168.1.33 port 5201
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-1.08   sec  2.75 MBytes  21.3 Mbits/sec              
[  4]   1.08-2.07   sec  2.75 MBytes  23.3 Mbits/sec              
[  4]   2.07-3.06   sec  2.75 MBytes  23.3 Mbits/sec              
[  4]   3.06-4.05   sec  2.75 MBytes  23.3 Mbits/sec              
[  4]   4.05-5.08   sec  2.38 MBytes  19.3 Mbits/sec              
[  4]   5.08-6.08   sec  2.62 MBytes  22.1 Mbits/sec              
[  4]   6.08-7.07   sec  2.75 MBytes  23.2 Mbits/sec              
[  4]   7.07-8.07   sec  2.75 MBytes  23.1 Mbits/sec              
[  4]   8.07-9.08   sec  2.75 MBytes  22.9 Mbits/sec              
[  4]   9.08-10.07  sec  2.75 MBytes  23.2 Mbits/sec              
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bandwidth       Retr
[  4]   0.00-10.07  sec  30.3 MBytes  25.2 Mbits/sec    1         sender
[  4]   0.00-10.07  sec  27.8 MBytes  23.1 Mbits/sec              receiver

iperf Done.

root@OpenWrt ~# iw dev wlan0 station dump 
Station 1c:1a:c0:a5:53:8d (on wlan0)
    inactive time:    63560 ms
    rx bytes:    2720785
    rx packets:    24823
    tx bytes:    2255151
    tx packets:    7583
    tx retries:    0
    tx failed:    0
    signal:      -67 dBm
    signal avg:    -66 dBm
    tx bitrate:    150.0 MBit/s MCS 7 40MHz short GI
    rx bitrate:    150.0 MBit/s MCS 7 40MHz short GI
    authorized:    yes
    authenticated:    yes
    preamble:    long
    WMM/WME:    yes
    MFP:        no
    TDLS peer:    no
    connected time:    137222 seconds
Station 30:d6:c9:d2:72:88 (on wlan0)
    inactive time:    13950 ms
    rx bytes:    2970212
    rx packets:    37020
    tx bytes:    36936998
[... 154 lines cut ...]
[  4]   0.00-10.01  sec   364 MBytes   305 Mbits/sec              sender
[  4]   0.00-10.01  sec   364 MBytes   305 Mbits/sec              receiver

iperf Done.
Macbook-Jukka:Documents jukka$ ./iperf301  -c 192.168.1.33 -i 1  -p 5201 -R
Connecting to host 192.168.1.33, port 5201
Reverse mode, remote host 192.168.1.33 is sending
[  4] local 192.168.1.14 port 50717 connected to 192.168.1.33 port 5201
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-1.00   sec  30.1 MBytes   252 Mbits/sec              
[  4]   1.00-2.01   sec  31.6 MBytes   265 Mbits/sec              
[  4]   2.01-3.01   sec  31.6 MBytes   264 Mbits/sec              
[  4]   3.01-4.01   sec  31.9 MBytes   268 Mbits/sec              
[  4]   4.01-5.01   sec  31.9 MBytes   267 Mbits/sec              
[  4]   5.01-6.01   sec  31.5 MBytes   264 Mbits/sec              
[  4]   6.01-7.00   sec  31.6 MBytes   266 Mbits/sec              
[  4]   7.00-8.01   sec  31.5 MBytes   263 Mbits/sec              
[  4]   8.01-9.00   sec  31.5 MBytes   265 Mbits/sec              
[  4]   9.00-10.01  sec  31.6 MBytes   265 Mbits/sec              
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bandwidth       Retr
[  4]   0.00-10.01  sec   318 MBytes   266 Mbits/sec    0         sender
[  4]   0.00-10.01  sec   316 MBytes   265 Mbits/sec              receiver

iperf Done.

Chadster766, 2016-05-16:
@bbogey interesting I will see if I can replicate this.

bbogey, 2016-05-16:
Using openwrt r49166 (arokh build)

Marvell 802.11ac Wireless Network Driver version 10.3.0.17-20160324

Now testing CHAOS CALMER (15.05.1, r48532) (mwlwifi 10.3.0.16-20160105)
 
But seems having same issue.

bbogey, 2016-05-17:
Today 5GHz was slow again.
I left iperf running and after about 12 minutes speed was increased:

http://pastebin.com/qj2HVmWd

Router was visible at 4 meters away.

I have one neighbor + I have another Wifi at garage. 
For 5GHz wrt1900acs is on chan 36 (80MHz), garage is on chan 52 (40MHz).  No neighbors at 5GHz.
For 2.4GHz wrt1900acs is on chan 11 (20MHz), garage on chan 1 (20MHz), neighbor is on chan 1 (40MHz).

bbogey, 2016-05-17:
Again had the same. Now at davidc502 build r49377 with latest wifi driver.
Right after I kicked one client, speed was normal again.

`root@OpenWrt:~# ubus call hostapd.wlan0 del_client '{"addr":"80:c5:e6:fe:d3:8c", "reason":1, "deauth":true, "ban_time":10}'`

Here is snip from iperf

`  4] 600.12-601.09 sec  2.00 MBytes  17.2 Mbits/sec              
[  4] 601.09-602.07 sec  2.00 MBytes  17.3 Mbits/sec              
[  4] 602.07-603.11 sec  2.12 MBytes  17.1 Mbits/sec              
[  4] 603.11-604.08 sec  2.00 MBytes  17.2 Mbits/sec              
[  4] 604.08-605.11 sec  2.12 MBytes  17.3 Mbits/sec              
[  4] 605.11-606.08 sec  2.00 MBytes  17.3 Mbits/sec              
[  4] 606.08-607.06 sec  2.00 MBytes  17.1 Mbits/sec              
[  4] 607.06-608.09 sec  2.12 MBytes  17.3 Mbits/sec              
[  4] 608.09-609.07 sec  2.00 MBytes  17.2 Mbits/sec              
[  4] 609.07-610.11 sec  2.12 MBytes  17.2 Mbits/sec              
[  4] 610.11-611.08 sec  2.00 MBytes  17.2 Mbits/sec              
[  4] 611.08-612.06 sec  2.00 MBytes  17.1 Mbits/sec              
[  4] 612.06-613.10 sec  2.12 MBytes  17.2 Mbits/sec              
[  4] 613.10-614.07 sec  2.00 MBytes  17.3 Mbits/sec              
[  4] 614.07-615.10 sec  2.12 MBytes  17.2 Mbits/sec              
[  4] 615.10-616.08 sec  2.00 MBytes  17.2 Mbits/sec              
[  4] 616.08-617.12 sec  2.12 MBytes  17.2 Mbits/sec              
[  4] 617.12-618.10 sec  2.00 MBytes  17.0 Mbits/sec              
[  4] 618.10-619.08 sec  2.00 MBytes  17.2 Mbits/sec              
[  4] 619.08-620.06 sec  2.00 MBytes  17.1 Mbits/sec              
[  4] 620.06-621.49 sec   768 KBytes  4.39 Mbits/sec              
[  4] 621.49-622.01 sec  13.8 MBytes   225 Mbits/sec              
[  4] 622.01-623.01 sec  27.1 MBytes   227 Mbits/sec              
[  4] 623.01-624.01 sec  27.0 MBytes   226 Mbits/sec              
[  4] 624.01-625.01 sec  27.1 MBytes   227 Mbits/sec              
[  4] 625.01-626.01 sec  27.1 MBytes   227 Mbits/sec              
[  4] 626.01-627.00 sec  26.9 MBytes   226 Mbits/sec              
[  4] 627.00-628.01 sec  27.1 MBytes   226 Mbits/sec              
[  4] 628.01-629.01 sec  26.6 MBytes   223 Mbits/sec              
[  4] 629.01-630.01 sec  25.9 MBytes   218 Mbits/sec              
[  4] 630.01-631.01 sec  27.2 MBytes   228 Mbits/sec              
[  4] 631.01-632.01 sec  27.2 MBytes   228 Mbits/sec              
[  4] 632.01-633.01 sec  27.0 MBytes   227 Mbits/sec              
^C[  4] 633.01-633.12 sec  2.88 MBytes   218 Mbits/sec              
- - - - - - - - - - - - - - - - - - - - - - - - -
`

davidc502, 2016-05-18:
I've noticed that if I bounce the device I.E. Android, throughput 
*sometimes* return to normal immediately, OR may take an hour or even 
the next day.  After the speeds return to normal, they seem to stabilize 
for days and I've gone a week with great speeds after a device reboot.  
Since new builds are installed almost weekly I haven't tested further.

I didn't want to just boot the router, and found this behavior.

David

---
This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus

bbogey, 2016-05-18:
I noticed the same now.
I connected new connection (another macbook) then disconnected. Nothing.
Then bounced device what already was connected.

Here is the log, with the common ampdu errors

```
Wed May 18 21:11:34 2016 daemon.debug hostapd: wlan1: STA 80:c5:e6:fe:d3:8c WPA: event 2 notification
Wed May 18 21:11:34 2016 daemon.debug hostapd: wlan1: STA 80:c5:e6:fe:d3:8c IEEE 802.1X: unauthorizing port
Wed May 18 21:11:34 2016 daemon.info hostapd: wlan1: STA 80:c5:e6:fe:d3:8c IEEE 802.11: disassociated
Wed May 18 21:11:34 2016 daemon.debug hostapd: wlan1: STA 80:c5:e6:fe:d3:8c MLME: MLME-DISASSOCIATE.indication(80:c5:e6:fe:d3:8c, 8)
Wed May 18 21:11:34 2016 daemon.debug hostapd: wlan1: STA 80:c5:e6:fe:d3:8c MLME: MLME-DELETEKEYS.request(80:c5:e6:fe:d3:8c)
Wed May 18 21:11:34 2016 daemon.debug hostapd: wlan1: STA 80:c5:e6:fe:d3:8c WPA: event 3 notification
Wed May 18 21:11:34 2016 daemon.debug hostapd: wlan1: STA 80:c5:e6:fe:d3:8c IEEE 802.1X: unauthorizing port
Wed May 18 21:11:34 2016 daemon.debug hostapd: wlan1: STA 80:c5:e6:fe:d3:8c IEEE 802.11: deauthenticated
Wed May 18 21:11:34 2016 daemon.debug hostapd: wlan1: STA 80:c5:e6:fe:d3:8c MLME: MLME-DEAUTHENTICATE.indication(80:c5:e6:fe:d3:8c, 3)
Wed May 18 21:11:34 2016 daemon.debug hostapd: wlan1: STA 80:c5:e6:fe:d3:8c MLME: MLME-DELETEKEYS.request(80:c5:e6:fe:d3:8c)
Wed May 18 21:11:34 2016 kern.err kernel: [21530.454170] ieee80211 phy1: check ba result error 1
Wed May 18 21:11:34 2016 kern.err kernel: [21530.459190] ieee80211 phy1: ampdu start error code: -22
Wed May 18 21:11:34 2016 kern.err kernel: [21530.531193] ieee80211 phy1: check ba result error 1
Wed May 18 21:11:34 2016 kern.err kernel: [21530.536204] ieee80211 phy1: ampdu start error code: -22
Wed May 18 21:11:34 2016 kern.err kernel: [21530.581269] ieee80211 phy1: check ba result error 1
Wed May 18 21:11:34 2016 kern.err kernel: [21530.586258] ieee80211 phy1: ampdu start error code: -22
Wed May 18 21:11:35 2016 kern.err kernel: [21530.638161] ieee80211 phy1: check ba result error 1
Wed May 18 21:11:35 2016 kern.err kernel: [21530.643118] ieee80211 phy1: ampdu start error code: -22
Wed May 18 21:11:35 2016 kern.err kernel: [21530.707197] ieee80211 phy1: check ba result error 1
Wed May 18 21:11:35 2016 kern.err kernel: [21530.712136] ieee80211 phy1: ampdu start error code: -22
Wed May 18 21:11:35 2016 kern.err kernel: [21530.742339] ieee80211 phy1: check ba result error 1
Wed May 18 21:11:35 2016 kern.err kernel: [21530.747462] ieee80211 phy1: ampdu start error code: -22
Wed May 18 21:11:35 2016 kern.err kernel: [21530.791123] ieee80211 phy1: check ba result error 1
Wed May 18 21:11:35 2016 kern.err kernel: [21530.796126] ieee80211 phy1: ampdu start error code: -22
Wed May 18 21:11:35 2016 kern.err kernel: [21530.841099] ieee80211 phy1: check ba result error 1
Wed May 18 21:11:35 2016 kern.err kernel: [21530.846123] ieee80211 phy1: ampdu start error code: -22
Wed May 18 21:11:35 2016 kern.err kernel: [21530.901078] ieee80211 phy1: check ba result error 1
Wed May 18 21:11:35 2016 kern.err kernel: [21530.906023] ieee80211 phy1: ampdu start error code: -22
Wed May 18 21:11:35 2016 kern.err kernel: [21530.951091] ieee80211 phy1: check ba result error 1
Wed May 18 21:11:35 2016 kern.err kernel: [21530.956122] ieee80211 phy1: ampdu start error code: -22
Wed May 18 21:11:35 2016 kern.err kernel: [21530.991101] ieee80211 phy1: check ba result error 1
Wed May 18 21:11:35 2016 kern.err kernel: [21530.996008] ieee80211 phy1: ampdu start error code: -22
Wed May 18 21:11:35 2016 kern.err kernel: [21531.041067] ieee80211 phy1: check ba result error 1
Wed May 18 21:11:35 2016 kern.err kernel: [21531.045972] ieee80211 phy1: ampdu start error code: -22
Wed May 18 21:11:35 2016 kern.err kernel: [21531.081070] ieee80211 phy1: check ba result error 1
Wed May 18 21:11:35 2016 kern.err kernel: [21531.085978] ieee80211 phy1: ampdu start error code: -22
Wed May 18 21:11:35 2016 kern.err kernel: [21531.121063] ieee80211 phy1: check ba result error 1
Wed May 18 21:11:35 2016 kern.err kernel: [21531.125967] ieee80211 phy1: ampdu start error code: -22
Wed May 18 21:11:35 2016 kern.err kernel: [21531.161072] ieee80211 phy1: check ba result error 1
Wed May 18 21:11:35 2016 kern.err kernel: [21531.165993] ieee80211 phy1: ampdu start error code: -22
Wed May 18 21:11:35 2016 kern.err kernel: [21531.201069] ieee80211 phy1: check ba result error 1
Wed May 18 21:11:35 2016 kern.err kernel: [21531.205979] ieee80211 phy1: ampdu start error code: -22
Wed May 18 21:11:35 2016 kern.err kernel: [21531.251059] ieee80211 phy1: check ba result error 1
Wed May 18 21:11:35 2016 kern.err kernel: [21531.255962] ieee80211 phy1: ampdu start error code: -22
Wed May 18 21:11:35 2016 kern.err kernel: [21531.291062] ieee80211 phy1: check ba result error 1
Wed May 18 21:11:35 2016 kern.err kernel: [21531.295976] ieee80211 phy1: ampdu start error code: -22
Wed May 18 21:11:35 2016 kern.err kernel: [21531.331056] ieee80211 phy1: check ba result error 1
Wed May 18 21:11:35 2016 kern.err kernel: [21531.335964] ieee80211 phy1: ampdu start error code: -22
Wed May 18 21:11:35 2016 kern.err kernel: [21531.381056] ieee80211 phy1: check ba result error 1
Wed May 18 21:11:35 2016 kern.err kernel: [21531.385961] ieee80211 phy1: ampdu start error code: -22
Wed May 18 21:11:35 2016 kern.err kernel: [21531.421059] ieee80211 phy1: check ba result error 1
Wed May 18 21:11:35 2016 kern.err kernel: [21531.425975] ieee80211 phy1: ampdu start error code: -22
[... cut ...]
Wed May 18 21:11:44 2016 daemon.debug hostapd: wlan1: STA 80:c5:e6:fe:d3:8c IEEE 802.1X: authorizing port
Wed May 18 21:11:44 2016 daemon.info hostapd: wlan1: STA 80:c5:e6:fe:d3:8c RADIUS: starting accounting session 1CE64553-00000018
Wed May 18 21:11:44 2016 daemon.info hostapd: wlan1: STA 80:c5:e6:fe:d3:8c WPA: pairwise key handshake completed (RSN)
Wed May 18 21:11:44 2016 daemon.info dnsmasq-dhcp[18528]: DHCPREQUEST(br-lan) 192.168.1.20 80:c5:e6:fe:d3:8c 
Wed May 18 21:11:44 2016 daemon.info dnsmasq-dhcp[18528]: DHCPACK(br-lan) 192.168.1.20 80:c5:e6:fe:d3:8c Jukka-Lumia-930
```

bbogey, 2016-05-18:
btw today run some endurance test for 7 hours on 5GHz and speed was fine all the time.

[image]

yolanjoy, 2016-05-19:
This is what I see right now the throughput deterioration on the 11n 5GHz band with no other client using it. This would come back to like 150Mbits/sec in a minute or so. But now it's been deteriorated for an hour now.  The router has been up for the last 3 weeks. The 11ac clients ( streaming)  are working fine.

[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-1.00   sec  5.12 MBytes  43.0 Mbits/sec
[  4]   1.00-2.00   sec  4.00 MBytes  33.6 Mbits/sec
[  4]   2.00-3.00   sec  4.38 MBytes  36.7 Mbits/sec
[  4]   3.00-4.00   sec  5.38 MBytes  45.1 Mbits/sec
[  4]   4.00-5.00   sec  4.88 MBytes  40.9 Mbits/sec
[  4]   5.00-6.00   sec  4.50 MBytes  37.7 Mbits/sec
[  4]   6.00-7.00   sec  4.75 MBytes  39.8 Mbits/sec
[  4]   7.00-8.00   sec  3.62 MBytes  30.4 Mbits/sec
[  4]   8.00-9.00   sec  4.88 MBytes  40.9 Mbits/sec
[  4]   9.00-10.00  sec  4.38 MBytes  36.7 Mbits/sec
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-10.00  sec  45.9 MBytes  38.5 Mbits/sec                  sender
[  4]   0.00-10.00  sec  45.8 MBytes  38.4 Mbits/sec                  receiver

yuhhaurlin, 2016-05-20:
Please check 10.3.0.17-20160520.

bbogey, 2016-05-20:
Ok running now
Fri May 20 10:00:48 2016 kern.info kernel: [   10.028849] <<Marvell 802.11ac Wireless Network Driver version 10.3.0.17-20160520>>

yolanjoy, 2016-05-20:
@yuhhaurlin - Thanks for the new version. I hope it will help with the issues. I got the stock firmware on my WRT1200ac router. Linksys hasn't released a firmware update in a long time. At least I would like to see this version - 10.3.0.17-20160324 in the stock firmware. I asked Linksys support about this but never got any replay. Any idea how these changes goes into the stock firmware at some point?

Thanks

yuhhaurlin, 2016-05-20:
@yolanjoy Stock firmware uses proprietary driver. The driver should not have this kind of problem.

yolanjoy, 2016-05-21:
Apparently the stock firmware does have this problem. Is it possible to test the proprietary driver for this issue?

Thanks

yuhhaurlin, 2016-05-21:
Please only report problems of mwlwifi here. Thanks.

yuhhaurlin, 2016-05-26:
Does anyone still have this problem with new driver?

verbit, 2016-05-29:
So I've done some testing of the 5GHz 802.11n performance.

Client: MacBook Pro 2011 Late
Router: WRT1200AC
Firmware: DD-WRT v3.0-r29750M kongmv (05/26/16)
Driver: 10.3.0.17-20160523

WiFi uses channel 48 lower.

MacBook in *send* mode (`iperf3 -c`)
```
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-1.00   sec  21.2 MBytes   178 Mbits/sec
[  4]   1.00-2.00   sec  22.8 MBytes   192 Mbits/sec
[  4]   2.00-3.00   sec  22.6 MBytes   189 Mbits/sec
[  4]   3.00-4.00   sec  22.8 MBytes   191 Mbits/sec
[  4]   4.00-5.00   sec  22.7 MBytes   190 Mbits/sec
[  4]   5.00-6.00   sec  22.7 MBytes   190 Mbits/sec
[  4]   6.00-7.00   sec  22.6 MBytes   189 Mbits/sec
[  4]   7.00-8.00   sec  22.7 MBytes   191 Mbits/sec
[  4]   8.00-9.00   sec  22.7 MBytes   190 Mbits/sec
[  4]   9.00-10.00  sec  22.8 MBytes   191 Mbits/sec
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-10.00  sec   226 MBytes   189 Mbits/sec                  sender
[  4]   0.00-10.00  sec   226 MBytes   189 Mbits/sec                  receiver
```

MacBook in *receive* mode (`iperf3 -c -R`)
```
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-1.00   sec  26.2 MBytes   219 Mbits/sec
[  4]   1.00-2.00   sec  27.7 MBytes   232 Mbits/sec
[  4]   2.00-3.00   sec  27.9 MBytes   234 Mbits/sec
[  4]   3.00-4.00   sec  27.7 MBytes   232 Mbits/sec
[  4]   4.00-5.00   sec  27.7 MBytes   233 Mbits/sec
[  4]   5.00-6.00   sec  27.6 MBytes   232 Mbits/sec
[  4]   6.00-7.00   sec  26.9 MBytes   225 Mbits/sec
[  4]   7.00-8.00   sec  27.5 MBytes   231 Mbits/sec
[  4]   8.00-9.00   sec  27.7 MBytes   233 Mbits/sec
[  4]   9.00-10.00  sec  27.7 MBytes   233 Mbits/sec
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bandwidth       Retr
[  4]   0.00-10.00  sec   278 MBytes   233 Mbits/sec    0             sender
[  4]   0.00-10.00  sec   275 MBytes   231 Mbits/sec                  receiver
```

Pinging Router
```
PING 192.168.2.1 (192.168.2.1): 56 data bytes
64 bytes from 192.168.2.1: icmp_seq=0 ttl=64 time=0.735 ms
64 bytes from 192.168.2.1: icmp_seq=1 ttl=64 time=0.841 ms
64 bytes from 192.168.2.1: icmp_seq=2 ttl=64 time=3.682 ms
64 bytes from 192.168.2.1: icmp_seq=3 ttl=64 time=3.509 ms
64 bytes from 192.168.2.1: icmp_seq=4 ttl=64 time=3.612 ms
64 bytes from 192.168.2.1: icmp_seq=5 ttl=64 time=3.671 ms
64 bytes from 192.168.2.1: icmp_seq=6 ttl=64 time=3.561 ms
64 bytes from 192.168.2.1: icmp_seq=7 ttl=64 time=3.420 ms
64 bytes from 192.168.2.1: icmp_seq=8 ttl=64 time=1.163 ms
64 bytes from 192.168.2.1: icmp_seq=9 ttl=64 time=3.625 ms

--- 192.168.2.1 ping statistics ---
10 packets transmitted, 10 packets received, 0.0% packet loss
round-trip min/avg/max/stddev = 0.735/2.782/3.682/1.230 ms
```

So throughput looks good to me but are somewhat asymmetric (why?). I'm not sure whether the ping times are good. Look slightly high to me (and irregular), regarding that I am very close to the router (<1m).

yuhhaurlin, 2016-05-30:
Thanks. I will close this one.

closed 2016-05-30 by yuhhaurlin