kaloz/mwlwifi #73: WRT1900ACv1 w/ wireless encryption causes crash

jprouty, 2016-04-14:
I've been playing with OpenWRT and DD-WRT the last few days. After installing OpenWRT (CC 15.05), my router will reboot whenever either wireless interface is enabled with encryption enabled. If encryption='none', the unit is fine.

I can no longer revert to stock firmware or DD-WRT, as both of those distributions up an encrypted wifi network on startup. I seem to be stuck with CC 15.05 with wireless disabled. I've tried nearly every build I could get my hands on CC 15.05, CC 15.05.1, OpenWRT trunk build, DD-WRT beta nightlys, stock (various versions).

My hunch is there's something wrong in the 88W8864 chip's config somewhere that I'm not clearing on a re-flash. Is there a procedure for resetting all config/state to stock condition? 

What can I do to provide more information on the crash? I've tried to get more info (stack trace, etc) but must be doing something wrong.

Platform: Linksys WRT1900ACv1 (Mamba)

yuhhaurlin, 2016-04-14:
Please check this kind of problem on dd-wrt or openwrt community.

closed 2016-04-14 by yuhhaurlin

jprouty, 2016-04-14:
Turns out the issue was due to using an underpowered AC brick. The OEM brick is required, otherwise when it goes to turn on the radios, the unit would reboot.