kaloz/mwlwifi #76: Compatible iwlwifi-10.3.0.17 driver with kernel 4.6?

ValCher1961, 2016-05-16:
Hi!
During the installation of modules kernel 4.6 with mwlwifi driver 10.3.0.17-20160324 out warning -
```
drivers/mwlwifi/mac80211.c:746:2: warning: initialization from incompatible pointer type [enabled by default]
  .ampdu_action       = mwl_mac80211_ampdu_action,
  ^
drivers/mwlwifi/mac80211.c:746:2: warning: (near initialization for 'mwl_mac80211_ops.ampdu_action') [enabled by default]

```

Chadster766, 2016-05-18:
Kernel 4.5.4 seem to be working well.

closed 2016-05-20 by ValCher1961

thagabe, 2016-05-20:
So no?

ValCher1961, 2016-05-20:
I did not wait for an answer, but my opinion - is not compatible.
The speed of the wifi was ~ 12-14mbit.

jbsky, 2016-06-01:
So, yes now.

ValCher1961, 2016-06-01:
I have not tried.

thagabe, 2016-06-01:
Yes now bc of the dropping of old mac80211 api's but that means cc mightin the gutter