kaloz/mwlwifi #77: ampdu/ba error log flood after 10.3.0.17-20160520-1 on WRT1900ac

wongsyrone, 2016-05-20:
It appears almost every second, more frequently than the previous version, looks like
```
[40398.631716] ieee80211 phy0: ampdu start error code: -22
[40398.666734] ieee80211 phy0: check ba result error 1
[40398.671710] ieee80211 phy0: ampdu start error code: -22
[40486.386854] ieee80211 phy0: check ba result error 1
[40486.391935] ieee80211 phy0: ampdu start error code: -22
[40486.436821] ieee80211 phy0: check ba result error 1
[40486.441907] ieee80211 phy0: ampdu start error code: -22
[40486.486819] ieee80211 phy0: check ba result error 1
[40486.491829] ieee80211 phy0: ampdu start error code: -22
[40486.526748] ieee80211 phy0: check ba result error 1
[40486.531771] ieee80211 phy0: ampdu start error code: -22
[40486.566804] ieee80211 phy0: check ba result error 1
[40486.571830] ieee80211 phy0: ampdu start error code: -22
[40486.616791] ieee80211 phy0: check ba result error 1
[40486.621792] ieee80211 phy0: ampdu start error code: -22
[40486.656808] ieee80211 phy0: check ba result error 1
[40486.661796] ieee80211 phy0: ampdu start error code: -22
[40552.236965] ieee80211 phy0: check ba result error 1
[40552.242037] ieee80211 phy0: ampdu start error code: -22
[40552.286733] ieee80211 phy0: check ba result error 1
[40552.291823] ieee80211 phy0: ampdu start error code: -22
[40552.326737] ieee80211 phy0: check ba result error 1
[40552.331716] ieee80211 phy0: ampdu start error code: -22
[40552.366732] ieee80211 phy0: check ba result error 1
[40552.371708] ieee80211 phy0: ampdu start error code: -22
[40552.406734] ieee80211 phy0: check ba result error 1
[40552.411709] ieee80211 phy0: ampdu start error code: -22
[40552.446722] ieee80211 phy0: check ba result error 1
[40552.451698] ieee80211 phy0: ampdu start error code: -22
[40552.486738] ieee80211 phy0: check ba result error 1
[40552.491724] ieee80211 phy0: ampdu start error code: -22
[40552.526720] ieee80211 phy0: check ba result error 1
[40552.531702] ieee80211 phy0: ampdu start error code: -22
[40552.566733] ieee80211 phy0: check ba result error 1
[40552.571708] ieee80211 phy0: ampdu start error code: -22
[40552.606736] ieee80211 phy0: check ba result error 1
[40552.611712] ieee80211 phy0: ampdu start error code: -22
[40552.946731] ieee80211 phy0: check ba result error 1
[40552.952055] ieee80211 phy0: ampdu start error code: -22
[40552.986735] ieee80211 phy0: check ba result error 1
[40552.991731] ieee80211 phy0: ampdu start error code: -22
[40553.026740] ieee80211 phy0: check ba result error 1
[40553.031982] ieee80211 phy0: ampdu start error code: -22
[40553.066815] ieee80211 phy0: check ba result error 1
[40553.072013] ieee80211 phy0: ampdu start error code: -22
[40553.106861] ieee80211 phy0: check ba result error 1
[40553.112006] ieee80211 phy0: ampdu start error code: -22
[40553.146799] ieee80211 phy0: check ba result error 1
[40553.152022] ieee80211 phy0: ampdu start error code: -22
[40553.196790] ieee80211 phy0: check ba result error 1
[40553.202070] ieee80211 phy0: ampdu start error code: -22
[40553.236763] ieee80211 phy0: check ba result error 1
[40553.241751] ieee80211 phy0: ampdu start error code: -22
[40553.286816] ieee80211 phy0: check ba result error 1
[40553.292078] ieee80211 phy0: ampdu start error code: -22
[40553.326702] ieee80211 phy0: check ba result error 1
[40553.331686] ieee80211 phy0: ampdu start error code: -22
[40553.376737] ieee80211 phy0: check ba result error 1
[40553.381714] ieee80211 phy0: ampdu start error code: -22
[40553.416736] ieee80211 phy0: check ba result error 1
[40553.421729] ieee80211 phy0: ampdu start error code: -22
[40553.456737] ieee80211 phy0: check ba result error 1
```

wongsyrone, 2016-05-20:
BTW, the previous version is 10.3.0.17-20160324

wongsyrone, 2016-05-20:
kernel: 4.4.11
mac80211: [2016-05-12](https://github.com/lede-project/source/tree/master/package/kernel/mac80211) upgraded by nbd(Felix Fietkau)
codebase: [LEDE](https://github.com/lede-project/source)

yuhhaurlin, 2016-05-21:
If function is all right, I will control the rate to print out this warning messages. Thanks.

wongsyrone, 2016-05-21:
It seems all right, just annoying.

yuhhaurlin, 2016-05-21:
Thanks. I will modify it.

wongsyrone, 2016-05-21:
Any modification I can do manually, or it is in the firmware?

yuhhaurlin, 2016-05-21:
You can just remove the code to print out this message in the driver.

johnnysl, 2016-05-21:
@yuhhaurlin, seeing the same. Any reason to leave this warning code in at all?

wongsyrone, 2016-05-21:
For those who want to get rid of log spam:
```diff
--- a/fwcmd.c
+++ b/fwcmd.c
@@ -2393,8 +2393,6 @@ int mwl_fwcmd_check_ba(struct ieee80211_
 
 	if (pcmd->cmd_hdr.result != 0) {
 		mutex_unlock(&priv->fwcmd_mutex);
-		wiphy_err(hw->wiphy, "check ba result error %d\n",
-			  le16_to_cpu(pcmd->cmd_hdr.result));
 		return -EINVAL;
 	}
 
--- a/mac80211.c
+++ b/mac80211.c
@@ -644,8 +644,6 @@ static int mwl_mac80211_ampdu_action(str
 		spin_lock_bh(&priv->stream_lock);
 		if (rc) {
 			mwl_fwcmd_remove_stream(hw, stream);
-			wiphy_err(hw->wiphy,
-				  "ampdu start error code: %d\n", rc);
 			rc = -EPERM;
 			break;
 		}
```

thagabe, 2016-05-21:
@yuhhaurlin @wongsyrone So what I'm understanding is the driver is reporting this error and openwrt is printing them out on the logs but the functionality of the driver is correct and the message is mostly harmless?

yuhhaurlin, 2016-05-21:
Yes,  from wongsyrone's report. I will modify code later.

wongsyrone, 2016-05-21:
It's better waiting for more reports before making decisions.

yuhhaurlin, 2016-05-21:
I will just modify the code to control the rate to print out these messages.

wongsyrone, 2016-05-21:
nbd did some changes[0], are those you want to modify?

[0]: https://github.com/lede-project/source/commit/29ad4d2a0cadbefea1e7d882b44aa2bbfb711aef and https://github.com/lede-project/source/commit/4926e4b1aeaaeac21bddc6b7fe95ffd221aee499

yuhhaurlin, 2016-05-21:
No, the patch removes the code.

wongsyrone, 2016-05-21:
Then there will be conflicts on the next release. @nbd168

yuhhaurlin, 2016-05-21:
I can modify the code to be compiled without patch with updated compatible wireless module.

wongsyrone, 2016-05-22:
nbd pushed a [fix](https://github.com/lede-project/source/commit/2c9c998b4a9f6aac24ae8b0690319ad1d704a7ae), closing this issue

closed 2016-05-22 by wongsyrone

wongsyrone, 2016-05-23:
@yuhhaurlin 
After 267c5144165ad7bd8da0e0f897a487b82460e95c, I find the error only occurs when connecting Huawei devices
AC:CF:85 HUAWEI TECHNOLOGIES CO.,LTD

yuhhaurlin, 2016-05-23:
If BA can be established without problem, I think you can ignore it.

wongsyrone, 2016-05-23:
Got it.