kaloz/mwlwifi #79: Poor 5GHz upload speed  on WRT1200AC

stamak, 2016-05-27:
Hi
I am experiencing the problem with uploading from my laptop but meanwhile with downloading looks like is ok (~10MB/s).  Router - Linksys WRT1200AC

About upload:
- enabled WMM speed is ~565.0KB/s
- disabled WMM speed is ~2MB/s

root@OpenWrt:~# cat /etc/openwrt_release
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='15.05.1'
DISTRIB_REVISION='r48532'
DISTRIB_CODENAME='chaos_calmer'
DISTRIB_TARGET='mvebu/generic'
DISTRIB_DESCRIPTION='OpenWrt Chaos Calmer 15.05.1'
DISTRIB_TAINTS=''
root@OpenWrt:~# modinfo mwlwifi
module:		/lib/modules/3.18.23/mwlwifi.ko
license:	GPL v2
depends:
root@OpenWrt:~# strings /lib/modules/3.18.23/mwlwifi.ko | grep ^10
10.3.0.17-20160523
10.3.0.17-20160523

verbit, 2016-05-27:
Have you performed an `iperf`/`iperf3` between the router and your laptop?
Can you upload the results, please?

Using the latest mwlwifi build (10.3.0.17-20160523) and latest DD-WRT (DD-WRT v3.0-r29750M kongmv 05/26/16) I do not have such issues in 5GHz 802.11n mode. Maybe the problem is specific to 802.11ac?

yuhhaurlin, 2016-05-28:
It is very hard for me to check this kind of report. Please do not disable WMM. BTW, this version of code can get pretty good speed under our internal test.

yuhhaurlin, 2016-05-28:
Can you check the throughout between LAN (or router) and WLAN first?

yuhhaurlin, 2016-05-30:
Do not know if your problem is the same as #64? Please help to check what client used by you?

yuhhaurlin, 2016-05-30:
I will close this one and focus on issue #64.

closed 2016-05-30 by yuhhaurlin