kaloz/mwlwifi #81: mwlwifi issue samba windows 7 sata HDD

gsustek, 2016-05-30:
Hi, i have WRT1900ACS LEDE r422 build with mwlwifi 10.3.0.17-20160520-1 version, my laptop has Intel 7265 REV=0x210 dual band wifi card, win 7 x64. router HDD is connected through ESATA connector and ext4 FS is exposed with samba3.6.25.-5 This morning i was trying to find some data on HDD, and in one moment DCOM(windows service crashed) working and my Wifi(i was still connected,but with yellow sign-no internet) 5GHz/80MHz/auto/HR/psk2 stops working for any devices(i have ipad and iphone also connected in the same time and another identically the same laptop but with Ubuntu  XX on it- never had such problem, but never using samba). Then i connected to the router with LAN cable, pickup some logs, and restart it.

Here are logs(dmesg,logread).
DMESG
[   37.961556] ieee80211 phy0: change: 0x40
[   38.171461] ieee80211 phy0: change: 0x40
[   38.381455] ieee80211 phy0: change: 0x40
[   38.489475] ieee80211 phy0: change: 0x100
[   38.555761] ieee80211 phy0: change: 0x100
[   38.568494] ieee80211 phy0: change: 0x42
[   38.731464] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   38.737902] br-lan: port 3(wlan0) entered forwarding state
[   38.743440] br-lan: port 3(wlan0) entered forwarding state
[   38.836780] ieee80211 phy0: change: 0x20
[   40.741443] br-lan: port 3(wlan0) entered forwarding state
[ 1250.182983] TCP: request_sock_TCP: Possible SYN flooding on port 1194. Sending cookies.  Check SNMP counters.
[ 1250.195553] xhci-hcd f10f8000.usb3: ERROR Transfer event TRB DMA ptr not part of current TD ep_index 2 comp_code 13
[ 1250.206035] xhci-hcd f10f8000.usb3: Looking for event-dma 000000001b7d7ae0 trb-start 000000001b7d7a30 trb-end 000000001b7d7a30 seg-start 000000001b7d7000 seg-end 000000001b7d7ff0
[ 1250.222007] xhci-hcd f10f8000.usb3: ERROR Transfer event TRB DMA ptr not part of current TD ep_index 2 comp_code 13
[ 1250.232487] xhci-hcd f10f8000.usb3: Looking for event-dma 000000001b07faa0 trb-start 000000001b07f9e0 trb-end 000000001b07f9e0 seg-start 000000001b07f000 seg-end 000000001b07fff0
[ 1250.248458] xhci-hcd f10f8000.usb3: ERROR Transfer event TRB DMA ptr not part of current TD ep_index 2 comp_code 13
[ 1250.258937] xhci-hcd f10f8000.usb3: Looking for event-dma 000000001b0cfa30 trb-start 000000001b0cf970 trb-end 000000001b0cf970 seg-start 000000001b0cf000 seg-end 000000001b0cfff0
[ 1250.274908] xhci-hcd f10f8000.usb3: ERROR Transfer event TRB DMA ptr not part of current TD ep_index 2 comp_code 13
[ 1250.285387] xhci-hcd f10f8000.usb3: Looking for event-dma 000000001b7d7af0 trb-start 000000001b7d7a30 trb-end 000000001b7d7a30 seg-start 000000001b7d7000 seg-end 000000001b7d7ff0
/cut/
[ 1257.241712] xhci-hcd f10f8000.usb3: Looking for event-dma 000000001b7db400 trb-start 000000001b7d7a30 trb-end 000000001b7d7a30 seg-start 000000001b7d7000 seg-end 000000001b7d7ff0
[ 1257.257683] xhci-hcd f10f8000.usb3: ERROR Transfer event TRB DMA ptr not part of current TD ep_index 2 comp_code 13
[ 1257.268161] xhci-hcd f10f8000.usb3: Looking for event-dma 000000001b7db410 trb-start 000000001b7d7a30 trb-end 000000001b7d7a30 seg-start 000000001b7d7000 seg-end 000000001b7d7ff0
[ 1257.284132] xhci-hcd f10f8000.usb3: ERROR Transfer event TRB DMA ptr not part of current TD ep_index 2 comp_code 13
[ 1257.294611] xhci-hcd f10f8000.usb3: Looking for event-dma 000000001b7db420 trb-start 000000001b7d7a30 trb-end 000000001b7d7a30 seg-start 000000001b7d7000 seg-end 000000001b7d7ff0
[32050.062252] ieee80211 phy0: cmd 0x9125=BAStream timed out
[32050.067685] ieee80211 phy0: return code: 0x1125
[32050.072234] ieee80211 phy0: timeout: 0x1125
[32050.076451] ieee80211 phy0: destroy ba failed execution
[32328.709000] ieee80211 phy0: cmd 0x9122=UpdateEncryption timed out
[32328.715125] ieee80211 phy0: return code: 0x1122
[32328.719687] ieee80211 phy0: timeout: 0x1122
[32328.723888] ieee80211 phy0: failed execution
[32328.728187] wlan0: failed to remove key (0, 78:7e:61:e8:a5:ab) from hardware (-5)
[32332.736320] ieee80211 phy0: cmd 0x9111=SetNewStation timed out
[32332.742187] ieee80211 phy0: return code: 0x1111
[32332.746739] ieee80211 phy0: timeout: 0x1111
[32332.750945] ieee80211 phy0: failed execution
[32366.152036] ieee80211 phy0: cmd 0x9122=UpdateEncryption timed out
[32366.158165] ieee80211 phy0: return code: 0x1122
[32366.162714] ieee80211 phy0: timeout: 0x1122
[32366.166922] ieee80211 phy0: failed execution
[32366.171209] wlan0: failed to remove key (0, 18:5e:0f:2f:00:3f) from hardware (-5)
[32370.176213] ieee80211 phy0: cmd 0x9111=SetNewStation timed out
[32370.182073] ieee80211 phy0: return code: 0x1111
[32370.186632] ieee80211 phy0: timeout: 0x1111
[32370.190831] ieee80211 phy0: failed execution
[32374.191636] ieee80211 phy0: cmd 0x8050=broadcast_ssid_enable timed out
[32374.198201] ieee80211 phy0: return code: 0x0050
[32374.202749] ieee80211 phy0: timeout: 0x0050
[32374.206956] ieee80211 phy0: failed execution
[32378.208064] ieee80211 phy0: cmd 0x8127=SetInformationElements timed out
[32378.214709] ieee80211 phy0: return code: 0x0127
[32378.219268] ieee80211 phy0: timeout: 0x0127
[32378.223468] ieee80211 phy0: failed execution
[32382.221995] ieee80211 phy0: cmd 0x801c=80211RadioControl timed out
[32382.228211] ieee80211 phy0: return code: 0x001c
[32382.232760] ieee80211 phy0: timeout: 0x001c
[32382.236967] ieee80211 phy0: failed execution
[32386.239916] ieee80211 phy0: cmd 0x8126=SetFixedRate timed out
[32386.245688] ieee80211 phy0: return code: 0x0126
[32386.250246] ieee80211 phy0: timeout: 0x0126
[32386.254445] ieee80211 phy0: failed execution
[32442.152117] ieee80211 phy0: cmd 0x9122=UpdateEncryption timed out
[32442.158247] ieee80211 phy0: return code: 0x1122
[... 120 lines cut ...]
Mon May 30 08:09:59 2016 daemon.info dnsmasq-dhcp[24183]: DHCPACK(br-lan) 192.168.168.150 jj:jj:jj:jj:jj:jj HOST_WITH_ISSUE
Mon May 30 08:10:36 2016 daemon.info hostapd: wlan1: STA zz:zz:zz:zz:zz:zz WPA: group key handshake completed (RSN)
Mon May 30 08:10:36 2016 daemon.info hostapd: wlan1: STA kk:kk:kk:kk:kk:kk WPA: group key handshake completed (RSN)
Mon May 30 08:10:47 2016 kern.err kernel: [32442.152117] ieee80211 phy0: cmd 0x9122=UpdateEncryption timed out
Mon May 30 08:10:47 2016 kern.err kernel: [32442.158247] ieee80211 phy0: return code: 0x1122
Mon May 30 08:10:47 2016 kern.err kernel: [32442.162798] ieee80211 phy0: timeout: 0x1122
Mon May 30 08:10:47 2016 kern.err kernel: [32442.167007] ieee80211 phy0: failed execution
Mon May 30 08:10:47 2016 kern.err kernel: [32442.171294] wlan0: failed to remove key (1, ff:ff:ff:ff:ff:ff) from hardware (-5)
Mon May 30 08:10:51 2016 kern.err kernel: [32446.178336] ieee80211 phy0: cmd 0x9122=UpdateEncryption timed out
Mon May 30 08:10:51 2016 kern.err kernel: [32446.184463] ieee80211 phy0: return code: 0x1122
Mon May 30 08:10:51 2016 kern.err kernel: [32446.189035] ieee80211 phy0: timeout: 0x1122
Mon May 30 08:10:51 2016 kern.err kernel: [32446.193237] ieee80211 phy0: failed execution
Mon May 30 08:10:51 2016 kern.err kernel: [32446.197534] wlan0: failed to set key (1, ff:ff:ff:ff:ff:ff) to hardware (-5)
Mon May 30 08:10:51 2016 daemon.info dnsmasq-dhcp[24183]: DHCPINFORM(br-lan) 192.168.168.150 jj:jj:jj:jj:jj:jj 
Mon May 30 08:10:51 2016 daemon.info dnsmasq-dhcp[24183]: DHCPACK(br-lan) 192.168.168.150 jj:jj:jj:jj:jj:jj HOST_WITH_ISSUE
Mon May 30 08:10:51 2016 daemon.info dnsmasq-dhcp[24183]: DHCPINFORM(br-lan) 192.168.168.150 jj:jj:jj:jj:jj:jj 
Mon May 30 08:10:51 2016 daemon.info dnsmasq-dhcp[24183]: DHCPACK(br-lan) 192.168.168.150 jj:jj:jj:jj:jj:jj HOST_WITH_ISSUE
Mon May 30 08:11:56 2016 daemon.info dnsmasq-dhcp[24183]: DHCPINFORM(br-lan) 192.168.168.150 jj:jj:jj:jj:jj:jj
Mon May 30 08:11:56 2016 daemon.info dnsmasq-dhcp[24183]: DHCPACK(br-lan) 192.168.168.150 jj:jj:jj:jj:jj:jj HOST_WITH_ISSUE
Mon May 30 08:12:23 2016 user.warn ddns-scripts[2996]: jozo_noip_me: Updating IP at DDNS provider failed - starting retry 54/0
Mon May 30 08:13:00 2016 authpriv.info dropbear[32487]: Child connection from 192.168.168.150:56696
Mon May 30 08:13:06 2016 authpriv.notice dropbear[32487]: Password auth succeeded for 'root' from 192.168.168.150:56696

Regards,
Goran.

yuhhaurlin, 2016-05-30:
Please give precise steps to reproduce this problem. From the log, firmware is gone.

gsustek, 2016-05-30:
i reproduce it again, just use same samba version then share /mnt/sda1 HDD(ext4 FS)(connected to to ESATA port) and try to stream some movie with VLC on windows 7..

i didn't understand that "firmware is gone", i am currently connected  to the router through ethernet cable, what logs do you need?

yuhhaurlin, 2016-05-30:
I don't know if this is problem of WiFi. Can you do iperf stress test to see if you will encounter problem or not.

gsustek, 2016-05-30:
i would do that, but in the meantime, please tell me what debug logs can i switch on, or even create build with higher login. could you suggest me iperf command switches on server(router) and client win7?  thnx. 
i always have trouble with 5GHz, and when i change it from auto to channel 120 issue also happend but  i can still have alive ssh connection to router.

gsustek, 2016-05-30:
i do the iperf stress test, and i got it, even from linux..after 2 minutes.

yuhhaurlin, 2016-05-30:
It is very strange. The log means firmware can't response commands. I don't think the driver can't survive for two minutes iperf test.

yuhhaurlin, 2016-05-30:
Auto channel won't work now.

gsustek, 2016-05-30:
I think it is the Intel 7265 wifi card, on linux, iperf work for a while, but on windows, drop death after 3 seconds. Only 5GHz. Do you have same wifi to test? I will now try another build to check is it problem in kernel, or in packet configuration.

Do you have any suggestion? What can i enable/debugging to see what is wrong.

yuhhaurlin, 2016-05-30:
The log shows every commands can't be responsed by firmware.

wongsyrone, 2016-05-30:
How about lack of file descriptor since he is using samba

yuhhaurlin, 2016-05-30:
I suggest he just did iperf test and the problem happened about 2 minutes.

yuhhaurlin, 2016-05-31:
@gsustek Please just do iperf test and check if this is the only client has this kind of problem. Thanks.

gsustek, 2016-05-31:
It seems that i have some false configuration, after i install LEDE r471
Everything is workig fine, you can now close this issue.

Thank you

closed 2016-05-31 by yuhhaurlin

yuhhaurlin, 2016-06-01:
I think I know the root cause: Prototype of ampdu_action() of mac80211 is changed. I will modify mwlwifi to comply with mac80211 APIs of latest compatible (backports) wireless package. Later you can built this driver directly without patches.