kaloz/mwlwifi #83: mwlwifi upstreaming

thagabe, 2016-06-03:
@yuhhaurlin how's the progress into upstreaming this code to the linux kernel? Also can this code be built on CC+kernel3.18 all the way up to LEDE-kernel4.4.12?

wongsyrone, 2016-06-03:
@yuhhaurlin 

Just an off-topic point.
OpenWrt/LEDE users don't have to remove [MAX-MPDU-7991] manually, vht_capab is generated by script(https://github.com/lede-project/source/blob/master/package/kernel/mac80211/files/lib/netifd/wireless/mac80211.sh#L264)

yuhhaurlin, 2016-06-03:
The code is compiled with compatible (backports) wireless package. Backports package is used to let updated drivers to work on older kernel.
So if you can update compatible wireless package to 2016-01-10, then this driver can work on previous kernel.

yuhhaurlin, 2016-06-03:
@wongsyrone I know that. Just remind this modification must be done. Thanks.

yuhhaurlin, 2016-06-08:
Marvell will submit patch v8 recently. I think there is no need to track this activity here. If you are interesting, you can check mailing list of Linux Wirless:  https://wireless.wiki.kernel.org/en/developers/mailinglists.

closed 2016-06-08 by yuhhaurlin