kaloz/mwlwifi #84: Low wifi upload speed to router(WRT1200AC)

stamak, 2016-06-03:
I am experiencing low upload speed ( ~  500-700KB/s ) ONLY via wifi to router(WRT1200AC)
But via wire all is ok  - (max possible speed)
Download - all is nice as via wifi as via wire
Upload to other hosts via wifi - good as well

How to reproduce :
```
Stass-MacBook-Pro:~ stamak$ dd if=/dev/zero of=test bs=1024000 count=10
10+0 records in
10+0 records out
10240000 bytes transferred in 0.007036 secs (1455378434 bytes/sec)
Stass-MacBook-Pro:~ stamak$ ls -lh test
-rw-r--r--  1 stamak  staff   9,8M  3 чер 20:08 test
Stass-MacBook-Pro:~ stamak$ scp test root@router_ip:/tmp
root@router_ip's password:
test                                                  100%   10MB 555.6KB/s   00:18
Stass-MacBook-Pro:~ stamak$
```

Wire:
```
root@my-test:~$ dd if=/dev/zero of=/tmp/test_10M bs=1M count=10
10+0 records in
10+0 records out
10485760 bytes (10 MB) copied, 0.0836006 s, 125 MB/s
root@my-test:~$ ls -lh /tmp/test_10M
-rw-rw-r-- 1 root root 10M Jun  3 17:07 /tmp/test_10M
root@my-test:~$ scp /tmp/test_10M root@router_ip:/tmp
root@router_ip's password:
test_10M                                          100%   10MB  10.0MB/s   00:01
root@my-test:~$
```

Versions:
```
root@OpenWrt:~# strings /lib/modules/*.*/mwlwifi.ko | grep "^10.3"
10.3.0.17-20160531-1
10.3.0.17-20160531-1
root@OpenWrt:~# cat /etc/openwrt_version
15.05.1
```

yuhhaurlin, 2016-06-03:
Please use Iperf to do test. I think this is not problem of WiFi.

yuhhaurlin, 2016-06-05:
If upload to other clients is all right, I don't think this is a problem of WiFi. Please help to do iperf test to see if WiFi has problem.

yuhhaurlin, 2016-06-06:
I will close this one. If you find iPerf test is not good for WiFi, please update test result and I will reopen this issue.

closed 2016-06-06 by yuhhaurlin

stamak, 2016-06-06:
My iPerf results:

```
Stass-MacBook-Pro:~ stamak$ iperf -c 192.168.2.1
------------------------------------------------------------
Client connecting to 192.168.2.1, TCP port 5001
TCP window size:  129 KByte (default)
------------------------------------------------------------
[  4] local 192.168.2.177 port 51832 connected with 192.168.2.1 port 5001
[ ID] Interval       Transfer     Bandwidth
[  4]  0.0-10.0 sec   148 MBytes   124 Mbits/sec
Stass-MacBook-Pro:~ stamak$ iperf -s
------------------------------------------------------------
Server listening on TCP port 5001
TCP window size:  128 KByte (default)
------------------------------------------------------------
[  4] local 192.168.2.177 port 5001 connected with 192.168.2.1 port 36385
[ ID] Interval       Transfer     Bandwidth
[  4]  0.0-10.3 sec  19.0 MBytes  15.5 Mbits/sec
```

there are download and upload tests

stamak, 2016-06-06:
the most interesting when I disabled WMM I have got  ~ 2.0MB/s instead of  ~ 700KB/s:

```
Stass-MacBook-Pro:~ stamak$ scp test root@192.168.2.1:/tmp
root@192.168.2.1's password:
test                                             100%   10MB   2.0MB/s   00:05
Stass-MacBook-Pro:~ stamak$
```

As for me it is WiFi problem

yuhhaurlin, 2016-06-06:
192.168.2.1 is your router. The iperf should station to router is 124 Mbps, but router to station is 15.5 Mbps (looks like g rate). Does these tests are done with WMM enable? Please don't disable WMM.

stamak, 2016-06-07:
"Does these tests are done with WMM enable?"
Yes this tests are done with WMM enable. PHY mode is 802.11ac

My MacBook Pro is Retina, 13-inch, Early 2015

AP settings:
```
config wifi-device 'radio0'
	option type 'mac80211'
	option hwmode '11a'
	option path 'soc/soc:pcie-controller/pci0000:00/0000:00:01.0/0000:01:00.0'
	option txpower '20'
	option channel '36'
	option htmode 'VHT80'
	option country 'US'

config wifi-iface
	option device 'radio0'
	option network 'lan'
	option mode 'ap'
	option key '*******'
	option ssid 'Test_5GHz'
	option encryption 'psk2+ccmp'
```

yuhhaurlin, 2016-06-07:
Can you give me full iperf test result?
One for AP to station and another one for station to AP and let iperf run for a while.

stamak, 2016-06-08:
**From Laptop to AP**

AP output:
```
root@OpenWrt:~# iperf -s
------------------------------------------------------------
Server listening on TCP port 5001
TCP window size: 85.3 KByte (default)
------------------------------------------------------------
[  4] local 192.168.2.1 port 5001 connected with 192.168.2.177 port 53241
[ ID] Interval       Transfer     Bandwidth
[  4]  0.0-30.0 sec  1.59 GBytes   456 Mbits/sec
```

laptop output:
```
Stass-MBP:~ stamak$ iperf -c 192.168.2.1 -t 30
------------------------------------------------------------
Client connecting to 192.168.2.1, TCP port 5001
TCP window size:  129 KByte (default)
------------------------------------------------------------
[  4] local 192.168.2.177 port 53241 connected with 192.168.2.1 port 5001
[ ID] Interval       Transfer     Bandwidth
[  4]  0.0-30.0 sec  1.59 GBytes   457 Mbits/sec
```

**From AP to laptop**

laptop:
```
Stass-MBP:~ stamak$ iperf -s
------------------------------------------------------------
Server listening on TCP port 5001
TCP window size:  128 KByte (default)
------------------------------------------------------------
[  4] local 192.168.2.177 port 5001 connected with 192.168.2.1 port 35608
[ ID] Interval       Transfer     Bandwidth
[  4]  0.0-30.0 sec  1.95 GBytes   557 Mbits/sec
```

AP output:
```
root@OpenWrt:~# iperf -c 192.168.2.177 -t 30
------------------------------------------------------------
Client connecting to 192.168.2.177, TCP port 5001
TCP window size: 43.8 KByte (default)
------------------------------------------------------------
[  3] local 192.168.2.1 port 35608 connected with 192.168.2.177 port 5001
[ ID] Interval       Transfer     Bandwidth
[  3]  0.0-30.0 sec  1.95 GBytes   558 Mbits/sec
```

As you see iperf results are pretty good now! Maybe it is due to router reboot, I will be checking from time to time.

According to iperf we can say that there should not be any problem but I still have with scp upload via wifi from laptop to AP, results the same:
enabled WMM ~ 700KB/s
disabled WMM ~ 2.0MB/s 
wire  - good

@yuhhaurlin  taking into account **WMM enable/disable** diversity - looks like the problem with WIFI driver?

yuhhaurlin, 2016-06-08:
I don't know. Throughout of WiFi is good.

yuhhaurlin, 2016-06-08:
BTW, throughput dropping after running for a while is fixed after 10.3.0.17-20160324.

stamak, 2016-06-11:
@yuhhaurlin this is 100% reproducible
if you have HW you can very easily verify it

bbogey, 2016-06-11:
wrt1900acs
Lede r484 mwlwifi 10.3.0.17-20160607

root@lede:~# uptime
 14:21:24 up 4 days,  4:28,  load average: 0.00, 0.00, 0.00

root@lede:~# free -m
             total       used       free     shared    buffers     cached
Mem:        513468     219220     294248      11072     119224      36008
-/+ buffers/cache:      63988     449480
Swap:            0          0          0

```
Wireless 5GHz

Macbook-Jukka:~ jukka$ dd if=/dev/zero of=test bs=1024000 count=10
10+0 records in
10+0 records out
10240000 bytes transferred in 0.014601 secs (701322202 bytes/sec)
Macbook-Jukka:~ jukka$ scp test root@192.168.1.1:/tmp
root@192.168.1.1's password: 
test                                                                                                                  100%   10MB   9.8MB/s   00:01    

Wired 

jukka@bogey:~$ dd if=/dev/zero of=test bs=1024000 count=10
10+0 records in
10+0 records out
10240000 bytes (10 MB, 9.8 MiB) copied, 0.0133121 s, 769 MB/s
jukka@bogey:~$ scp test root@192.168.1.1:/tmp
root@192.168.1.1's password: 
test                                                                                                          100%   10MB   9.8MB/s   00:01    
```

stamak, 2016-06-12:
@bbogey it's cool but I am running OpenWRT 15.05.1 and have WRT1200AC
Looks like I have to test it on the same LEDE version as you have at least
Could you please help me with it, because I can not find any info about how I can build the same version as you have on https://www.lede-project.org/ ?
Thx in advance