kaloz/mwlwifi #85: Wireless 5GHz AC speed slow

jglooije, 2016-06-05:
Here is iperf test with with MSI 2QC apache, AC client on 5GHz. Intel dualband AC3160 
Running wrt1900acs, LEDE Reboot r492 with mwlwifi version 10.3.0.17-20160603 Latest commit 8635191

root@OpenWrt:/usr/bin# ./iperf3 -c 192.168.144.29 -i 10 -t 60 -p 5201 -R
Connecting to host 192.168.144.29, port 5201
Reverse mode, remote host 192.168.144.29 is sending
[ 4] local 192.168.144.1 port 38176 connected to 192.168.144.29 port 5201
[ ID] Interval Transfer Bandwidth
[ 4] 0.00-10.00 sec 77.5 MBytes 65.0 Mbits/sec
[ 4] 10.00-20.00 sec 86.4 MBytes 72.5 Mbits/sec
[ 4] 20.00-30.00 sec 89.6 MBytes 75.2 Mbits/sec
[ 4] 30.00-40.00 sec 86.6 MBytes 72.6 Mbits/sec
[ 4] 40.00-50.00 sec 86.0 MBytes 72.1 Mbits/sec
[ 4] 50.00-60.00 sec 79.7 MBytes 66.9 Mbits/sec

[ ID] Interval Transfer Bandwidth
[ 4] 0.00-60.00 sec 506 MBytes 70.7 Mbits/sec sender
[ 4] 0.00-60.00 sec 506 MBytes 70.7 Mbits/sec receiver

iperf Done.
root@OpenWrt:/usr/bin# ./iperf3 -c 192.168.144.29 -i 10 -t 60 -p 5201
Connecting to host 192.168.144.29, port 5201
[ 4] local 192.168.144.1 port 38180 connected to 192.168.144.29 port 5201
[ ID] Interval Transfer Bandwidth Retr Cwnd
[ 4] 0.00-10.00 sec 49.6 MBytes 41.6 Mbits/sec 0 211 KBytes
[ 4] 10.00-20.00 sec 77.9 MBytes 65.4 Mbits/sec 0 211 KBytes
[ 4] 20.00-30.00 sec 72.7 MBytes 61.0 Mbits/sec 0 211 KBytes
[ 4] 30.00-40.00 sec 73.9 MBytes 62.0 Mbits/sec 0 211 KBytes
[ 4] 40.00-50.00 sec 81.1 MBytes 68.0 Mbits/sec 0 211 KBytes
[ 4] 50.00-60.00 sec 75.1 MBytes 63.0 Mbits/sec 2 211 KBytes

[ ID] Interval Transfer Bandwidth Retr
[ 4] 0.00-60.00 sec 430 MBytes 60.2 Mbits/sec 2 sender
[ 4] 0.00-60.00 sec 430 MBytes 60.1 Mbits/sec receiver

iperf Done.

yuhhaurlin, 2016-06-05:
What is the setting of AP? This only happens for this version of code? What is the number you got with stock firmware under same setting?

yuhhaurlin, 2016-06-05:
BTW, how about other clients?

jglooije, 2016-06-05:
@yuhhaurlin On every build I used from trunk openwrt to mrfreeze his build and now my own I seem to run against the 80Mbit/s limit. (I used openwrt and lede) I started my own build with version 10.3.0.16-20160105 until the last and tried almost all version in between. 
I use the standard AP settings on channel 149 ac with a guest wifi. I tried to tune my laptop and AC point settings but it brings no higher speeds only slower(so I'm on the standard settings). 
The MCS is mostly MCS3 or lower. On the 5Ghz with an N client I reach higher speeds (above 100Mbit/s) then with my AC client.
For the factory firmware I need to do a test, I need to find the time and moment, so for that I will come back later.

yuhhaurlin, 2016-06-05:
Thanks. So this is the problem with this specific client. Can you change the title?

yuhhaurlin, 2016-06-05:
Does anyone have the same issue with this client?

NemoAlex, 2016-06-06:
I have tested 10.3.0.17-20160601 with my MBP Mid 2014, the 5GHz basically not available. 2.4GHz is fine. But has no issue with my other clients.
Now I have to use 20160531-1.
Not sure if it's same or related issue with this one.

yuhhaurlin, 2016-06-06:
@NemoAlex Not available is not a good way to describe the problem. Can you let me know precisely what the problem that you encountered?

NemoAlex, 2016-06-06:
It's like I can see the ssid and connect, but everything is crazy slow.
Ping the router, time are about 2s-3s, 50% packet loss.

yuhhaurlin, 2016-06-06:
@NemoAlex But with 20160531, everything is all right?

NemoAlex, 2016-06-06:
@yuhhaurlin Yes. It's a new problem since 10.3.0.16. 
Didn't test 10.3.0.17-20160603.

yuhhaurlin, 2016-06-06:
@NemoAlex I don't understand! 20160531-1 is also 10.3.0.17. So the problem happened between 10.3.0.17-20160531-1 and 10.3.0.17-20160601? That is 10.3.0.17-20160531-1 is all right, but 10.3.0.17-20160601 has problem.

NemoAlex, 2016-06-06:
10.3.0.17-20160531-1 is all right, but 10.3.0.17-20160601 has the problem.
Sorry about the misunderstanding.

yuhhaurlin, 2016-06-06:
@NemoAlex Thanks. It is clear to me now. The only modification should be the way to check capability of AMSDU. I will check it. But I think it is not related to this issue.

wongsyrone, 2016-06-06:
I'm using 39cf37cf5f8940dd1796c0a10b06e88748553263 on wrt1900ac, 5G works well for me. Is this issue specific to wrt1900acs?

yuhhaurlin, 2016-06-07:
@wongsyrone I think problem of NemoAlex is that his MBP Mid 2014 (11n) has problem to work with AP on 5g band after 10.3.0.17-20160531-1. But it can work on 2g band with 10.3.0.17-20160601. I will check it later.

bbogey, 2016-06-07:
What this shows for the client
 iwinfo wlan0 assoclist

24:A0:74:4A:1F:E4  -52 dBm / -89 dBm (SNR 37)  10 ms ago
        RX: 866.7 MBit/s, VHT-MCS 9, 80MHz, VHT-NSS 2      2778 Pkts.
        TX: 866.7 MBit/s, VHT-MCS 9, 80MHz, VHT-NSS 2      3889 Pkts.

jglooije, 2016-06-07:
@yuhhaurlin I flashed to factory firmware (latest from the web v1.0.0.172978) turn off/on and did a speed test, the result was very poor. 5MBit/s upload and 50MBit/s download. Compared to wired 15MBit/s upload and 150MBit/s download. Now for the **_weird_** part. Flashed back my own build with the following result:

root@OpenWrt:/usr/bin# ./iperf3 -c 192.168.144.29 -i 10 -t 60 -p 5201 -R
Connecting to host 192.168.144.29, port 5201
Reverse mode, remote host 192.168.144.29 is sending
[  4] local 192.168.144.1 port 34552 connected to 192.168.144.29 port 5201
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-60.00  sec  1.25 GBytes   180 Mbits/sec                  sender
[  4]   0.00-60.00  sec  1.25 GBytes   180 Mbits/sec                  receiver
- - - - - - - - - - - - - - - - - - - - - - - - -
root@OpenWrt:/usr/bin# ./iperf3 -c 192.168.144.29 -i 10 -t 60 -p 5201
Connecting to host 192.168.144.29, port 5201
[  4] local 192.168.144.1 port 34556 connected to 192.168.144.29 port 5201
[ ID] Interval           Transfer     Bandwidth       Retr
[  4]   0.00-60.00  sec  1.86 GBytes   266 Mbits/sec  668             sender
[  4]   0.00-60.00  sec  1.86 GBytes   266 Mbits/sec                  receiver
- - - - - - - - - - - - - - - - - - - - - - - - -

**Turn off/on the router and get the following result:** (no settings changed, only password)

root@OpenWrt:/usr/bin# ./iperf3 -c 192.168.144.29 -i 10 -t 60 -p 5201 -R
Connecting to host 192.168.144.29, port 5201
Reverse mode, remote host 192.168.144.29 is sending
[  4] local 192.168.144.1 port 56656 connected to 192.168.144.29 port 5201
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-60.00  sec  1.03 GBytes   147 Mbits/sec                  sender
[  4]   0.00-60.00  sec  1.03 GBytes   147 Mbits/sec                  receiver
- - - - - - - - - - - - - - - - - - - - - - - - -
root@OpenWrt:/usr/bin# ./iperf3 -c 192.168.144.29 -i 10 -t 60 -p 5201
Connecting to host 192.168.144.29, port 5201
[  4] local 192.168.144.1 port 56660 connected to 192.168.144.29 port 5201
[ ID] Interval           Transfer     Bandwidth       Retr
[  4]   0.00-60.00  sec   592 MBytes  82.8 Mbits/sec    8             sender
[  4]   0.00-60.00  sec   592 MBytes  82.8 Mbits/sec                  receiver

So higher speeds are possible.

yuhhaurlin, 2016-06-12:
It looks like the latest test is better. Can you try the latest driver 10.3.0.17-20160608?

jglooije, 2016-06-12:
@yuhhaurlin, Sorry no improvement for me

jglooije, 2016-07-15:
I read on the Internet that the AC-3160 for some people caused problems, so decided to replace it. Bought a BCM94352Z NGFF M.2 WiFi WLAN Bluetooth 4.0 802.11ac adapter. This adapter gives me better speed. From the router 330Mb/s and to the router 300 Mb/s. The MCS from to router is now MCS9 and to the router MCS6.

yuhhaurlin, 2016-07-15:
Thanks. I will close this issue.

closed 2016-07-15 by yuhhaurlin