kaloz/mwlwifi #86: 2.4 GHZ not coming up

thagabe, 2016-06-07:
I have just upgraded to LEDE r521  and although luci sees Marvell 88W8864 802.11bgn (radio0) nothing seems able to connect to it and it's reporting 0% ? Mbit/s. I will try to set it up manually to see if that helps but right now 100% solid on LEDE and 5GHZ wifi

yuhhaurlin, 2016-06-07:
I just run 10.3.0.17-20160607 with OpenWrt, I did not encounter this kind of problem.

root@OpenWrt:/# cat /sys/kernel/debug/ieee80211/phy1/mwlwifi/info

driver name: mwlwifi
chip type: 88W8864
hw version: 7
driver version: 10.3.0.17-20160607
firmware version: 0x07020902
mac address: 00:50:43:21:bd:e8
2g: enable
5g: disable
antenna: 4 4
irq number: 104
iobase0: f0a00000
iobase1: f0c00000
tx limit: 768
rx limit: 64
ap macid support: 0000ffff
sta macid support: 00010000
macid used: 00000001
qe trigger number: 20764

root@OpenWrt:/# iw dev
phy#1
        Interface wlan1
                ifindex 12
                wdev 0x100000003
                addr 00:50:43:21:bd:e8
                ssid OpenWrt-2g-david
                type AP
                channel 6 (2437 MHz), width: 20 MHz, center1: 2437 MHz
                txpower 30.00 dBm
phy#0
        Interface wlan0
                ifindex 11
                wdev 0x4
                addr 00:50:43:21:bd:e9
                ssid OpenWrt-5g-david
                type AP
                channel 149 (5745 MHz), width: 80 MHz, center1: 5775 MHz
                txpower 20.00 dBm

root@OpenWrt:/# iw wlan1 station dump
Station b8:e8:56:3f:12:f8 (on wlan1)
        inactive time:  10030 ms
        rx bytes:       6281
        rx packets:     69
        tx bytes:       2569
        tx packets:     11
        tx retries:     0
        tx failed:      0
        signal:         -12 dBm
        signal avg:     -13 dBm
        tx bitrate:     216.7 MBit/s MCS 23 short GI
        rx bitrate:     36.0 MBit/s
        authorized:     yes
        authenticated:  yes
        preamble:       short
        WMM/WME:        yes
        MFP:            no
        TDLS peer:      no
        connected time: 218 seconds

thagabe, 2016-06-07:
[image]
seems it's disable but ive tried to bring it up and it still seems disable. the third command doesn't output anything for me

yuhhaurlin, 2016-06-07:
One module is running as 2g and another one is running as 5g. My device is not the same as yours (phy0 is 5g and phy1 is 2g). I think your device should be phy0 is 2g and phy1 is 5g. However, I don't think this is problem of driver.

yuhhaurlin, 2016-06-07:
The third command proves that client is connected to my AP on 2g. Please report real problem for mwlwifi here.

closed 2016-06-07 by yuhhaurlin