kaloz/mwlwifi #87: Performance issues on Intel(R) Dual Band Wireless-AC 7260 2.4Ghz

thagabe, 2016-06-07:
Running the latest mwlwifi af294d17374f2acff09b0b8fc64668288b6e1dd4 (compiling lede with the newest merged code) my computer will connect to the 2.4 ssid and have connection to the internet. The problem I'm noticing is that the connection is dropped, when that doesn't happen the internet speeds are between .98mbps to 50mbps with high variability. Connecting to 2.4ghz on my nexus 6p yields, from what i have tested, a stable connection with up to 80 mbps (out of my broadband's 180 mbps). To remedied this I have used my old Asus rt-n56u running padavan's firmware (opensource) as an AP for 2.4 and the intel card is working flawlessly (stable, no drops).

yuhhaurlin, 2016-06-07:
Previous version is all right?

thagabe, 2016-06-07:
If I recall correctly I had opened a ticket before and I did test it but i guess it was switching to 5ghz since 2.4ghz was dropped (I have now deleted all associated wifi except the 2.4ghz)

thagabe, 2016-06-08:
Latest 2a4f8bab4fe4a801affede8178e8fbc578d14f45 fixed my woes! I haven't messed around with the driver past setting an ssid and a wpa2 forced aes password and solid + stable.

closed 2016-06-08 by thagabe