kaloz/mwlwifi #88: Dreadfull performance latest driver with intel 7265-AC

Razerwire, 2016-06-11:
I'm experiencing pretty dreadfull downstream performance running the latest driver with a client utilizing Intel® Dual Band Wireless-AC 7265:

```
[  5] local 192.168.1.146 port 5201 connected to 192.168.1.1 port 55586
[ ID] Interval           Transfer     Bandwidth
[  5]   0.00-1.00   sec  1.35 MBytes  11.3 Mbits/sec                  
[  5]   1.00-2.00   sec   642 KBytes  5.26 Mbits/sec              
[  5]   2.00-3.00   sec  1.64 MBytes  13.7 Mbits/sec                  
[  5]   3.00-4.00   sec   735 KBytes  6.02 Mbits/sec                  
[  5]   4.00-5.00   sec   755 KBytes  6.19 Mbits/sec                  
[  5]   5.00-6.00   sec   814 KBytes  6.67 Mbits/sec                  
[  5]   6.00-7.00   sec   677 KBytes  5.55 Mbits/sec                  
[  5]   7.00-8.00   sec   543 KBytes  4.45 Mbits/sec                  
[  5]   8.00-9.00   sec   605 KBytes  4.96 Mbits/sec                  
[  5]   9.00-10.00  sec   563 KBytes  4.61 Mbits/sec                  
[  5]  10.00-11.00  sec   550 KBytes  4.51 Mbits/sec                  
[  5]  11.00-12.00  sec   510 KBytes  4.18 Mbits/sec                  
[  5]  12.00-13.00  sec   543 KBytes  4.45 Mbits/sec                  
[  5]  13.00-14.00  sec   604 KBytes  4.95 Mbits/sec                  
[  5]  14.00-15.00  sec   567 KBytes  4.65 Mbits/sec                  
[  5]  15.00-16.00  sec   642 KBytes  5.26 Mbits/sec                  
[  5]  16.00-17.00  sec   413 KBytes  3.38 Mbits/sec                  
[  5]  17.00-18.00  sec   911 KBytes  7.46 Mbits/sec                  
[  5]  18.00-19.00  sec   612 KBytes  5.02 Mbits/sec                  
[  5]  19.00-20.00  sec   608 KBytes  4.98 Mbits/sec                  
[  5]  20.00-21.00  sec   677 KBytes  5.55 Mbits/sec                  
[  5]  21.00-22.00  sec   501 KBytes  4.10 Mbits/sec                  
[  5]  22.00-23.00  sec   308 KBytes  2.53 Mbits/sec                  
[  5]  23.00-24.00  sec   535 KBytes  4.38 Mbits/sec                  
[  5]  24.00-25.00  sec   631 KBytes  5.17 Mbits/sec                  
[  5]  25.00-26.00  sec   576 KBytes  4.71 Mbits/sec                  
[  5]  26.00-27.00  sec   853 KBytes  6.99 Mbits/sec                  
[  5]  27.00-28.00  sec   605 KBytes  4.96 Mbits/sec                  
[  5]  27.00-28.00  sec   605 KBytes  4.96 Mbits/sec                  

[ ID] Interval           Transfer     Bandwidth
[  5]   0.00-28.00  sec  0.00 Bytes  0.00 bits/sec                  sender
[  5]   0.00-28.00  sec  30.0 MBytes  8.98 Mbits/sec                  receiver
```
My other 802.11AC clients are not affected by this problem, it's just the Wireless-AC 7265.

A reboot of the WRT1900ACv2 temporarily fixes the issue - for a few minutes - before speed drops again.

yuhhaurlin, 2016-06-12:
So this problem only happens for latest driver, but it is all right for previous version?

johnnysl, 2016-06-12:
Do you use abluetooth device (mouse?) on this card as well?
your speed seems to match others that use the card with a bluetooth device at the same time.
e.g.: https://communities.intel.com/thread/61208

Razerwire, 2016-06-12:
So last night i literally moved the router 3 inches further from the wall... since then the problem is gone....

bbogey, 2016-06-13:
> So last night i literally moved the router 3 inches further from the wall... since then the problem is gone....

Interesting. Are you able to reproduce?

I think I had yesterday the same issue with AppleTV N client on 5GHz. 
Could not stream movies but it was downloading 1.5Mbit/s. HBO and Netflix were garbage.

At the same time other clients in 5GHz was also very slow and pings were dropping.
The Apple TV is behind Gyproc wall about 4 meters from router. And there is tiny PC between blocking the waves.

Then I moved Apple TV for 2 inches and boom everything works fine again. I'm not able to reproduce the issue moving the AppleTV back where it was...

yuhhaurlin, 2016-06-13:
Sorry. I will close this one.

closed 2016-06-13 by yuhhaurlin