kaloz/mwlwifi #89: memory leak

gsustek, 2016-06-15:
I have 1900acs router and i have try'd every build from CC 15.05.1 to latest LEDE and i have the same issue...
Router setup:
-SATA connected HDD(i was also connect it through USB3.0-issue remains)  http://sprunge.us/NjPP
-vsftpd(or nfs4)
When i upload through wifi (5GHz) from Laptop(Intel7625, or ipad Air2) large file(size 2-3 Gigs) it seems that OS runs out of memory(i don't have this issue, when i upload through the same wifi, through router LAN to another PC with vsftpd service). I have swap mounted to OS(sometimes i have even 4-6 MB occupied).
some logs,(free, dmesg,meminfo)
http://sprunge.us/ZOKW
http://sprunge.us/iOMJ
http://sprunge.us/CBIe
also since both nfs (which runs in kernel space) and vsftpd (which runs in userspace) are affected it is unlikely to be a userspace program leak.

[img]https://s31.postimg.org/ijp9hn5yz/wifi_issue.png[/img]  last graph is before crash wifi->vsftpd->SATA HDD, firtst three is wifi->ethernet->ethernet(PC)->vsftpd

yuhhaurlin, 2016-06-16:
Can you run iperf test to see if you will encounter this kind of problem?

gsustek, 2016-06-16:
hi, it seems that this tweak solve OOM problem, http://stackoverflow.com/questions/21374491/vm-min-free-kbytes-why-keep-minimum-reserved-memory

you can now close this issue, for now.

Thank you again.

yuhhaurlin, 2016-06-16:
This driver is used in a lot of places. We did not find OOM problem. I will close this issue. Thanks.

closed 2016-06-16 by yuhhaurlin