kaloz/mwlwifi #92: RX and TX data rates highly asymmetric

cgbaumann, 2016-06-30:
When running tests with iperf the measured data rates are highly asymmetric:

~ # cat /sys/kernel/debug/ieee80211/phy0/mwlwifi/info

driver name: mwlwifi
chip type: 88W8897
hw version: 7
driver version: 10.3.0.17-20160617
firmware version: 0x0802000c
mac address: fc:62:b9:ff:e8:d4
2g: enable
5g: enable
antenna: 2 2
irq number: 154
iobase0: ffffff8009600000
iobase1: ffffff8009800000
tx limit: 768
rx limit: 64
ap macid support: 0000ffff
sta macid support: 00010000
macid used: 00010000
qe trigger number: 1745

~ # iperf -c 192.168.0.100
------------------------------------------------------------
Client connecting to 192.168.0.100, TCP port 5001
TCP window size: 85.0 KByte (default)
------------------------------------------------------------
[  3] local 192.168.0.101 port 43050 connected with 192.168.0.100 port 5001
[ ID] Interval       Transfer     Bandwidth
[  3]  0.0-12.7 sec  1.00 MBytes   658 Kbits/sec

~ # iperf -s
------------------------------------------------------------
Server listening on TCP port 5001
TCP window size: 85.3 KByte (default)
------------------------------------------------------------
[  4] local 192.168.0.101 port 5001 connected with 192.168.0.100 port 32931
[ ID] Interval       Transfer     Bandwidth
[  4]  0.0-10.3 sec  23.8 MBytes  19.3 Mbits/sec

yuhhaurlin, 2016-06-30:
The speed is pretty low. Can you let me know 8897 is running as AP or station mode? And what is the configuration of the BSS? Thanks.

yuhhaurlin, 2016-06-30:
BTW, please give me peer station information via "cat /sys/kernel/debug/ieee80211/phy0/mwlwifi/sta". Thanks.

cgbaumann, 2016-06-30:
As AP I use a TP-Link "TL-MR3420".

```
~ # cat /sys/kernel/debug/ieee80211/phy0/mwlwifi/sta

mac address: e8:94:f6:6d:d3:0a
aid: 0
ampdu: true
amsdu: false
ht_cap: 0x006f, ampdu: 03, 00
rx_mask: 0xff, ff, 00, 00
rx_bw: 1, rx_nss: 2
tdls: 0, tdls_init: 0
wme: 1, mfp: 0
IV: 000000000001
```

yuhhaurlin, 2016-06-30:
What is the configuration for the AP (it is 11n 2x2 AP)? BTW, what is the number you can get if you run mwifiex?

yuhhaurlin, 2016-06-30:
Did you hook antenna to 88W8897?

cgbaumann, 2016-06-30:
The web-interface of the AP says "11bgn mixed". My device has antennas for 2.4 and 5 GHz installed.
Using mwifiex gives 90 MBits/s RX and max. 10 MBits/s TX. Above 10 MBits/s the driver crashes. That's why I tried mwlwifi.

yuhhaurlin, 2016-06-30:
How did you build mwlwifi?

cgbaumann, 2016-06-30:
I integrated it into a 4.6 kernel tree and built it as a module.

yuhhaurlin, 2016-06-30:
1. Please use "cat /sys/kernel/debug/ieee80211/phy0/mwlwifi/ampdu" to check if BA is established or not when you run Tx test.
2. Please build in mac80211 debug function to give more debug information.
3. I don't know if you can use sniffer to get wireless packets？

cgbaumann, 2016-06-30:
```
~ # cat /sys/kernel/debug/ieee80211/phy0/mwlwifi/ampdu

stream: 0
idx: 0
state: 0
stream: 1
idx: 0
state: 0
stream: 2
idx: 0
state: 0
stream: 3
idx: 0
state: 0
```

Sniffing is a bit tricky with my current equipment. I would need an additional device.

yuhhaurlin, 2016-06-30:
Are you sure you cat this file when you run Tx iperf test? It is very strange, BA stream does not establish. Can I know the setting of your AP?

cgbaumann, 2016-06-30:
Ok, I tried again with iperf in background:

```
~ # cat /sys/kernel/debug/ieee80211/phy0/mwlwifi/ampdu

stream: 0
idx: 0
state: 3
mac address: e8:94:f6:6d:d3:0a
tid: 0
stream: 1
idx: 0
state: 0
stream: 2
idx: 0
state: 0
stream: 3
idx: 0
state: 0
```

yuhhaurlin, 2016-06-30:
Yes, BA is established. I have no idea why throughput is so bad. Can you capture packets?

yuhhaurlin, 2016-06-30:
This test is done on 5g?

cgbaumann, 2016-06-30:
The AP only uses 2.4 GHz band and is set to 300 MBits/s max. speed.

yuhhaurlin, 2016-06-30:
Can you try 5 GHz band？

cgbaumann, 2016-06-30:
My AP unfortunately doesn't support this.
I did a sniff with wireshark on IP level and couldn't detect any problems here (e.g. loast packets)

yuhhaurlin, 2016-07-08:
It is hard for me to know if your 2.4g band is clean. Normally 11n on 2.4g is hard to run 40  MHz. About Tx is pretty worse than Rx, I suggest that you can check if Tx signal is good or not. Maybe it needs to load power table for your module. It is better to run on 5g band for less interference and wider bandwidth.

northbound1, 2016-07-09:
@yuhhaurlin

Will this help?
I have also noticed connection speed drop under heavy load in one direction only, doing iperf3
then a -R one of the two will show serious decrease depending on which is the server.
Observed In Luci under Associated Stations > connection speeds.
No one else running 5Ghz in range.
3 meters distance full power and reduced power makes no difference. Direct line of site US set channel 157.
Different locations of router and client makes no difference.
An example.

[image]

Wrt1900acv1
5 Ghz
866.7 Mbit/s, 80MHz, VHT-MCS 9, VHT-NSS 2, Short GI

root@OpenWrt:~# cat /etc/openwrt_release
DISTRIB_ID='LEDE'
DISTRIB_RELEASE='HEAD'
DISTRIB_REVISION='r918'
DISTRIB_CODENAME='reboot'
DISTRIB_TARGET='mvebu/generic'
DISTRIB_DESCRIPTION='LEDE Reboot r918'
DISTRIB_TAINTS='no-all'

root@OpenWrt:~# cat /sys/kernel/debug/ieee80211/phy1/mwlwifi/info

driver name: mwlwifi
chip type: 88W8864
hw version: 7
driver version: 10.3.0.17-20160617
firmware version: 0x07020902
mac address: 00:25:9c:13:89:fa
2g: disable
5g: enable
antenna: 4 4
irq number: 91
iobase0: d1400000
iobase1: d1580000
tx limit: 768
rx limit: 64
ap macid support: 0000ffff
sta macid support: 00010000
macid used: 00000001
qe trigger number: 99465

root@OpenWrt:~# cat /sys/kernel/debug/ieee80211/phy1/mwlwifi/sta

mac address: **:**:**:**:**:** < edited
aid: 1
ampdu: true
amsdu: false
ht_cap: 0x006f, ampdu: 03, 05
rx_mask: 0xff, ff, 00, 00
vht_cap: 0x0f815832, mcs: fffa, fffa
rx_bw: 2, rx_nss: 2
tdls: 0, tdls_init: 0
wme: 1, mfp: 0
IV: 000000001e7e

root@OpenWrt:~# cat /sys/kernel/debug/ieee80211/phy1/mwlwifi/ampdu

stream: 0
idx: 0
state: 3
mac address: **:**:**:**:**:** < edited
tid: 0
stream: 1
idx: 0
state: 0
stream: 2
idx: 0
state: 0
stream: 3
idx: 0
state: 0

yuhhaurlin, 2016-07-09:
Please do not mix this issue with 8864. It looks like your client does not support AMSDU.

northbound1, 2016-07-09:
Sorry

yuhhaurlin, 2016-07-11:
@cgbaumann Above are all information I can provide. I will close this one. Thanks for your help.

closed 2016-07-11 by yuhhaurlin