kaloz/mwlwifi #93: BAStream timed out / UpdateEncryption timed out

johnnysl, 2016-07-06:
This is a new one for me. Currently running lede build r904, with the latest wifi driver on top.
HW: WRT1900ACV2.
Had to reboot to fix the issue.

[code]
[13686.727011] ieee80211 phy0: cmd 0x9125=BAStream timed out
[13686.732438] ieee80211 phy0: return code: 0x1125
[13686.736987] ieee80211 phy0: timeout: 0x1125
[13686.741197] ieee80211 phy0: destroy ba failed execution
[13824.024087] ieee80211 phy0: cmd 0x9122=UpdateEncryption timed out
[13824.030212] ieee80211 phy0: return code: 0x1122
[13824.034811] ieee80211 phy0: timeout: 0x1122
[13824.039080] ieee80211 phy0: failed execution
[13824.043368] wlan0: failed to remove key (0, c8:f7:33:f2:xx:xx) from hardware                   (-5)
[13828.056010] ieee80211 phy0: cmd 0x9122=UpdateEncryption timed out
[13828.062292] ieee80211 phy0: return code: 0x1122
[13828.066871] ieee80211 phy0: timeout: 0x1122
[13828.071071] ieee80211 phy0: failed execution
[13828.075479] wlan0: failed to remove key (0, 04:f7:e4:d6:xx:xx) from hardware                   (-5)
[13832.083952] ieee80211 phy0: cmd 0x9122=UpdateEncryption timed out
[13832.090395] ieee80211 phy0: return code: 0x1122
[13832.094970] ieee80211 phy0: timeout: 0x1122
[13832.099282] ieee80211 phy0: failed execution
[13832.103670] wlan0: failed to remove key (2, ff:ff:ff:ff:ff:ff) from hardware                   (-5)
[13836.113842] ieee80211 phy0: cmd 0x9122=UpdateEncryption timed out
[13836.120101] ieee80211 phy0: return code: 0x1122
[13836.124665] ieee80211 phy0: timeout: 0x1122
[13836.128864] ieee80211 phy0: failed execution
[13836.133202] wlan0: failed to set key (2, ff:ff:ff:ff:ff:ff) to hardware (-5)
[13840.141766] ieee80211 phy0: cmd 0x9122=UpdateEncryption timed out
[13840.147898] ieee80211 phy0: return code: 0x1122
[13840.152447] ieee80211 phy0: timeout: 0x1122
[13840.156826] ieee80211 phy0: failed execution
[13840.161318] wlan0: failed to remove key (0, 50:2e:5c:32:xx:xx) from hardware                   (-5)
[13844.173664] ieee80211 phy0: cmd 0x9111=SetNewStation timed out
[13844.179525] ieee80211 phy0: return code: 0x1111
[13844.184087] ieee80211 phy0: timeout: 0x1111
[13844.188426] ieee80211 phy0: failed execution
[13848.193580] ieee80211 phy0: cmd 0x9111=SetNewStation timed out
[13848.199442] ieee80211 phy0: return code: 0x1111
[13848.204005] ieee80211 phy0: timeout: 0x1111
[13848.208205] ieee80211 phy0: failed execution
[13852.213503] ieee80211 phy0: cmd 0x9111=SetNewStation timed out
[13852.219366] ieee80211 phy0: return code: 0x1111
[13852.223930] ieee80211 phy0: timeout: 0x1111
[13852.228130] ieee80211 phy0: failed execution
[13856.233416] ieee80211 phy0: cmd 0x8050=broadcast_ssid_enable timed out
[13856.239976] ieee80211 phy0: return code: 0x0050
[13856.244544] ieee80211 phy0: timeout: 0x0050
[13856.248743] ieee80211 phy0: failed execution
[13860.255331] ieee80211 phy0: cmd 0x8127=SetInformationElements timed out
[13860.261979] ieee80211 phy0: return code: 0x0127
[13860.266541] ieee80211 phy0: timeout: 0x0127
[13860.270741] ieee80211 phy0: failed execution
[13864.275249] ieee80211 phy0: cmd 0x801c=80211RadioControl timed out
[13864.281459] ieee80211 phy0: return code: 0x001c
[13864.286017] ieee80211 phy0: timeout: 0x001c
[13864.290217] ieee80211 phy0: failed execution
[13868.295158] ieee80211 phy0: cmd 0x8126=SetFixedRate timed out
[13868.300930] ieee80211 phy0: return code: 0x0126
[13868.305488] ieee80211 phy0: timeout: 0x0126
[13868.309687] ieee80211 phy0: failed execution
[/code]

jklap, 2016-07-06:
I'm also seeing the same issue. Also seems like a reboot is the only resolution. 

I'm running the June 8th-ish release.

V1 hardware, OpenWrt custom build from HEAD at that time.

johnnysl, 2016-07-08:
I'm seeing these issues on a daily basis currently, with various clients.
As the same driver was stable with and earlier build of LEDE, i'm returning to that build.

bbogey, 2016-07-08:
I have also seen this once. I'm at LEDE r484 and driver was 10.3.0.17-20160608.
But didn't need a reboot. 
```
Sun Jul  3 20:11:22 2016 kern.err kernel: [336084.040860] ieee80211 phy1: cmd 0x9125=BAStream timed out
Sun Jul  3 20:11:22 2016 kern.err kernel: [336084.046373] ieee80211 phy1: return code: 0x1125
Sun Jul  3 20:11:22 2016 kern.err kernel: [336084.051020] ieee80211 phy1: timeout: 0x1125
Sun Jul  3 20:11:22 2016 kern.err kernel: [336084.055307] ieee80211 phy1: check ba failed execution
```

jklap, 2016-07-08:
What I find curious is that I started running this build Jun 10th... but for the first couple of weeks I did not encounter this issue... but since it first occurred it seems the time between occurrences has been decreasing. For example in the last 24 hours I've had to reboot 3 times... in the 3 days previously to that, twice, in the week previous to that, twice.

It seems, but could be just coincidence, that it seems to occur around times of streaming video.

yuhhaurlin, 2016-07-08:
This is the same issue as issue #55. I will close this duplicate one.

closed 2016-07-08 by yuhhaurlin