kaloz/mwlwifi #94: WPA Rekey fails / kernel errors / wifi totally dead shortly after reboot.

mystica555, 2016-07-10:
WRT1200AC
New Lede development snapshot as of today, July 9th 2016. DISTRIB_DESCRIPTION='LEDE Reboot r930'

mwlwifi is current;

# strings mwlwifi.ko |grep  10.3.0
10.3.0.17-20160617
/home/buildbot/slave-lede-local/mvebu_generic/build/build_dir/target-arm_cortex-a9+vfpv3_musl-1.1.14_eabi/linux-mvebu/mwlwifi-10.3.0.17-20160617/mac80211.c
/home/buildbot/slave-lede-local/mvebu_generic/build/build_dir/target-arm_cortex-a9+vfpv3_musl-1.1.14_eabi/linux-mvebu/mwlwifi-10.3.0.17-20160617/fwcmd.c
/home/buildbot/slave-lede-local/mvebu_generic/build/build_dir/target-arm_cortex-a9+vfpv3_musl-1.1.14_eabi/linux-mvebu/mwlwifi-10.3.0.17-20160617/tx.c
10.3.0.17-20160617

Dmesg:
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.4.14 (buildbot@builds) (gcc version 5.3.0 (LEDE GCC 5.3.0 r915) ) #1 SMP Sat Jul 9 21:25:45 UTC 2016
[    0.000000] CPU: ARMv7 Processor [414fc091] revision 1 (ARMv7), cr=10c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] Machine model: Linksys WRT1200AC
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] On node 0 totalpages: 131072
[    0.000000] free_area_init_node: node 0, pgdat c05ee080, node_mem_map dfbf9000
[    0.000000]   Normal zone: 1024 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 131072 pages, LIFO batch:31
[    0.000000] PERCPU: Embedded 12 pages/cpu @dfbcf000 s17984 r8192 d22976 u49152
[    0.000000] pcpu-alloc: s17984 r8192 d22976 u49152 alloc=12*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 130048
[    0.000000] Kernel command line: console=ttyS0,115200 root=/dev/mtdblock7 ro rootdelay=1 rootfstype=jffs2 earlyprintk mtdparts=armada-nand:2048K(uboot)ro,256K(u_env),256K(s_env),1m@9m(devinfo),40m@10m(kernel),34m@16m(rootfs),40m@50m(alt_kernel),34m@56m(alt_rootfs),80m@10m(ubifs),-@90m(syscfg)
[    0.000000] PID hash table entries: 2048 (order: 1, 8192 bytes)
[    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes)
[    0.000000] Memory: 513064K/524288K available (4190K kernel code, 178K rwdata, 1560K rodata, 264K init, 300K bss, 11224K reserved, 0K cma-reserved, 0K highmem)
[    0.000000] Virtual kernel memory layout:
[    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
[    0.000000]     fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
[    0.000000]     vmalloc : 0xe0800000 - 0xff800000   ( 496 MB)
[    0.000000]     lowmem  : 0xc0000000 - 0xe0000000   ( 512 MB)
[    0.000000]     pkmap   : 0xbfe00000 - 0xc0000000   (   2 MB)
[    0.000000]     modules : 0xbf000000 - 0xbfe00000   (  14 MB)
[    0.000000]       .text : 0xc0008000 - 0xc05a5e10   (5752 kB)
[    0.000000]       .init : 0xc05a6000 - 0xc05e8000   ( 264 kB)
[    0.000000]       .data : 0xc05e8000 - 0xc06148dc   ( 179 kB)
[    0.000000]        .bss : 0xc06148dc - 0xc065fb44   ( 301 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=2, Nodes=1
[    0.000000] Hierarchical RCU implementation.
[    0.000000]  RCU restricting CPUs from NR_CPUS=4 to nr_cpu_ids=2.
[    0.000000] RCU: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=2
[    0.000000] NR_IRQS:16 nr_irqs:16 16
[    0.000000] mvebu_mbus: [Firmware Warn]: deprecated mbus-mvebu Device Tree, suspend/resume will not work
[    0.000000] L2C-310 enabling early BRESP for Cortex-A9
[    0.000000] L2C-310 full line of zeros enabled for Cortex-A9
[    0.000000] L2C-310 D prefetch enabled, offset 1 lines
[    0.000000] L2C-310 dynamic clock gating enabled, standby mode enabled
[    0.000000] L2C-310 Coherent cache controller enabled, 16 ways, 1024 kB
[    0.000000] L2C-310 Coherent: CACHE_ID 0x410054c9, AUX_CTRL 0x56070001
[    0.000000] Switching to timer-based delay loop, resolution 40ns
[    0.000003] sched_clock: 32 bits at 25MHz, resolution 40ns, wraps every 85899345900ns
[    0.000011] clocksource: armada_370_xp_clocksource: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 76450417870 ns
[    0.000135] Calibrating delay loop (skipped), value calculated using timer frequency.. 50.00 BogoMIPS (lpj=250000)
[    0.000143] pid_max: default: 32768 minimum: 301
[    0.000185] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.000190] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.000404] CPU: Testing write buffer coherency: ok
[    0.000519] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[    0.000544] Setting up static identity map for 0x8280 - 0x82d8
[    0.000632] mvebu-soc-id: MVEBU SoC ID=0x6820, Rev=0x4
[    0.000720] mvebu-pmsu: Initializing Power Management Service Unit
[    0.001184] Booting CPU 1
[    0.001367] CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
[... 1378 lines cut ...]
Sat Jul  9 22:19:34 2016 kern.err kernel: [ 5460.952667] ieee80211 phy1: cmd 0x9122=UpdateEncryption timed out
Sat Jul  9 22:19:34 2016 kern.err kernel: [ 5460.958810] ieee80211 phy1: return code: 0x1122
Sat Jul  9 22:19:34 2016 kern.err kernel: [ 5460.963373] ieee80211 phy1: timeout: 0x1122
Sat Jul  9 22:19:34 2016 kern.err kernel: [ 5460.967573] ieee80211 phy1: failed execution
Sat Jul  9 22:19:34 2016 kern.err kernel: [ 5460.971862] wlan1-1: failed to set key (2, ff:ff:ff:ff:ff:ff) to hardware (-5)
Sat Jul  9 22:20:03 2016 kern.err kernel: [ 5489.895873] ieee80211 phy0: cmd 0x9122=UpdateEncryption timed out
Sat Jul  9 22:20:03 2016 kern.err kernel: [ 5489.902003] ieee80211 phy0: return code: 0x1122
Sat Jul  9 22:20:03 2016 kern.err kernel: [ 5489.906564] ieee80211 phy0: timeout: 0x1122
Sat Jul  9 22:20:03 2016 kern.err kernel: [ 5489.910764] ieee80211 phy0: failed execution
Sat Jul  9 22:20:03 2016 kern.err kernel: [ 5489.915092] wlan0: failed to remove key (2, ff:ff:ff:ff:ff:ff) from hardware (-5)
Sat Jul  9 22:20:07 2016 kern.err kernel: [ 5493.921739] ieee80211 phy0: cmd 0x9122=UpdateEncryption timed out
Sat Jul  9 22:20:07 2016 kern.err kernel: [ 5493.927873] ieee80211 phy0: return code: 0x1122
Sat Jul  9 22:20:07 2016 kern.err kernel: [ 5493.932460] ieee80211 phy0: timeout: 0x1122
Sat Jul  9 22:20:07 2016 kern.err kernel: [ 5493.936661] ieee80211 phy0: failed execution
Sat Jul  9 22:20:07 2016 kern.err kernel: [ 5493.940949] wlan0: failed to set key (2, ff:ff:ff:ff:ff:ff) to hardware (-5)
Sat Jul  9 22:20:11 2016 kern.err kernel: [ 5497.947687] ieee80211 phy0: cmd 0x9122=UpdateEncryption timed out
Sat Jul  9 22:20:11 2016 kern.err kernel: [ 5497.953830] ieee80211 phy0: return code: 0x1122
Sat Jul  9 22:20:11 2016 kern.err kernel: [ 5497.958380] ieee80211 phy0: timeout: 0x1122
Sat Jul  9 22:20:11 2016 kern.err kernel: [ 5497.962648] ieee80211 phy0: failed execution
Sat Jul  9 22:20:11 2016 kern.err kernel: [ 5497.966937] wlan0-1: failed to remove key (2, ff:ff:ff:ff:ff:ff) from hardware (-5)
Sat Jul  9 22:20:15 2016 kern.err kernel: [ 5501.973849] ieee80211 phy0: cmd 0x9122=UpdateEncryption timed out
Sat Jul  9 22:20:15 2016 kern.err kernel: [ 5501.979972] ieee80211 phy0: return code: 0x1122
Sat Jul  9 22:20:15 2016 kern.err kernel: [ 5501.984533] ieee80211 phy0: timeout: 0x1122
Sat Jul  9 22:20:15 2016 kern.err kernel: [ 5501.988734] ieee80211 phy0: failed execution
Sat Jul  9 22:20:15 2016 kern.err kernel: [ 5501.993031] wlan0-1: failed to set key (2, ff:ff:ff:ff:ff:ff) to hardware (-5)

johnnysl, 2016-07-10:
Same issue as mine: https://github.com/kaloz/mwlwifi/issues/55

yuhhaurlin, 2016-07-10:
If it is not happened all the time, it should be same issue as #55.

yuhhaurlin, 2016-07-11:
@johnnysl Can you help to test this image: https://drive.google.com/open?id=0B3qLWtcWB9EdYnpBTGRYQ0VkY28? Thanks.

yuhhaurlin, 2016-07-11:
@mystica555 Can you let me know how often this problem would happen?

johnnysl, 2016-07-11:
@yuhhaurlin, sure will try to download and test today.

johnnysl, 2016-07-11:
@yuhhaurlin can you confirm the new firmware version? i forgot to write down the old one for comparison :$

yuhhaurlin, 2016-07-11:
@johnnysl 7.2.9.24. Previous one: 7.2.9.2.

johnnysl, 2016-07-11:
ok, if i put this in dec. it matches:
Firmware version: 0x07020918

i'll test it for a day, usually i have the issues within an hour.

yuhhaurlin, 2016-07-11:
If you encounter problem, please let me know the way to reproduce it. I will base on this version of code to check this problem.

johnnysl, 2016-07-11:
Connection is currently still rock stable, will continue testing with various loads and clients. Only thing i experience is that the 2.4ghz performance is really bad (iperf speeds of  ~30mbit up, ~50 mbit down, on a 20mhz channel, seems that the clients doesn't wan't to connect on more then 72mbit), but that is a different issue i'll open a seperate case on later.

mystica555, 2016-07-11:
I'll try the version you have posted.

The crux appears to be with Samsung clients.
Galaxy Note 3 on Sprint
Galaxy Tab (some exact model i forget; will check when roommate wakes up)
also procured from Sprint having Sprint LTE and whatever wifi radios this
also includes.

It is only these devices that can reliably lock up every firmware version
I've tried, it always seems to be within the rekeying process, and numerous
bugreports prior to this one all seem to hinge on the security rekeying
failing or having hiccups causing problems.

i _really_ hope this new firmware addresses all of these rekeying issues,
as they have been constant in some form or other since I first started
using mwlwifi, wrt1200ac, and seeing my roommate immediately have issues.

It's so bad right now, I've un-bridged my comcast modem and just have the
ssid she uses on -that-.

mystica555, 2016-07-11:
Of note, I never see the same rekeying issues with the Murata or
Qualcomm/Atheros chips in my 2 phones, nor the atheros chip in my laptop.

The only time the Blu Vivo Air LTE (qualcomm 100% integrated SoC) and the
LG G3 D851 (tmobile) w/Epigram wifi.

My apologies; i mixed up the 2.

Epigram, LG - mostly works.
Murata, Samsung - doesn't

mystica555, 2016-07-11:
Gah...
Mixing up thoughts here.

1: Samsung / Murata does not work.
2: Epigram / Atheros / QCA work better.
3: There is a random issue still present that I have not correlated yet to
the above rekeying process where even the Atheros clients stop passing
data. But it happens quite rare, and without the voracity that is Murata
when my roommate is attempting to watch Netflix on her phone.

johnnysl, 2016-07-11:
@yuhhaurlin: have seen the issue again on the new FW. it only seems to happen on the 5ghz interface. Have rebooted and am trying to reproduce, but it is a painfull issue to trigger so it seems.

mystica555, 2016-07-11:
Hmmm.

Interesting you mention the 5ghz interface. I need to see if someone has a
MEID-blacklisted sprint note3 to sell so i can do my own testing without my
roommate yelling at me that her netflix is dead every hour.

yuhhaurlin, 2016-07-11:
@johnnysl
1. Under same environment, throughput of 7.2.9.24 is worst than 7.2.9.2 on 2.4g?
2. Can you give me kernel log messages? Same as before?

johnnysl, 2016-07-11:
oh and firmware is still the new one of course:

```
 dmesg | grep firmware
[    9.886316] ieee80211 phy0: firmware version: 0x7020918
[   10.926315] ieee80211 phy1: firmware version: 0x7020918

```

yuhhaurlin, 2016-07-11:
@johnnysl the first message to let firmware command timeout is still related to "remove key"?

johnnysl, 2016-07-11:
@yuhhaurlin: sorry for my amateurism, can't confirm that 100% as i might have missed the first line while cutting and pasting it from my log, and have rebooted in the meanwhile.

will provide better logging next try.

johnnysl, 2016-07-12:
@yuhhaurlin: issue hasn't reocurred yet.
something else did though. it didn't gave any outages, but it might be related?

```
[ 8814.757582] ------------[ cut here ]------------
[ 8814.762329] WARNING: CPU: 0 PID: 1560 at compat-wireless-2016-06-20/net/mac80211/agg-tx.c:347 ___ieee80211_stop_tx_ba_session+0x13c/0x200 [mac80211]()
[ 8814.775885] Modules linked in: pppoe ppp_async iptable_nat pppox ppp_generic nf_nat_ipv4 nf_conntrack_ipv6 nf_conntrack_ipv4 ipt_REJECT ipt_MASQUERADE xt_time xt_tcpudp xt_state xt_nat xt_multiport xt_mark xt_mac xt_limit xt_id xt_conntrack xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_CT slhc nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_nat nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache nf_conntrack iptable_raw iptable_mangle iptable_filter ip_tables crc_ccitt mwlwifi mac80211 cfg80211 compat cryptodev ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_raw ip6table_mangle ip6table_filter ip6_tables x_tables tun cbc authenc sd_mod gpio_button_hotplug ext4 jbd2 mbcache aead crypto_null
[ 8814.841997] CPU: 0 PID: 1560 Comm: hostapd Not tainted 4.4.14 #2
[ 8814.848027] Hardware name: Marvell Armada 380/385 (Device Tree)
[ 8814.853982] [<c001fa20>] (unwind_backtrace) from [<c001c388>] (show_stack+0x10/0x14)
[ 8814.861761] [<c001c388>] (show_stack) from [<c01998a4>] (dump_stack+0x8c/0xa0)
[ 8814.869017] [<c01998a4>] (dump_stack) from [<c0028ae4>] (warn_slowpath_common+0x94/0xb0)
[ 8814.877143] [<c0028ae4>] (warn_slowpath_common) from [<c0028b9c>] (warn_slowpath_null+0x1c/0x24)
[ 8814.885989] [<c0028b9c>] (warn_slowpath_null) from [<bf1216c8>] (___ieee80211_stop_tx_ba_session+0x13c/0x200 [mac80211])
[ 8814.896946] [<bf1216c8>] (___ieee80211_stop_tx_ba_session [mac80211]) from [<bf121b80>] (__ieee80211_stop_tx_ba_session+0x30/0x44 [mac80211])
[ 8814.909728] [<bf121b80>] (__ieee80211_stop_tx_ba_session [mac80211]) from [<bf120814>] (ieee80211_sta_tear_down_BA_sessions+0x48/0x70 [mac80211])
[ 8814.922858] [<bf120814>] (ieee80211_sta_tear_down_BA_sessions [mac80211]) from [<bf117938>] (ieee80211_sta_eosp+0x1d8/0x54c [mac80211])
[ 8814.935114] [<bf117938>] (ieee80211_sta_eosp [mac80211]) from [<bf11a9d8>] (__sta_info_destroy+0xc/0x28 [mac80211])
[ 8814.945626] [<bf11a9d8>] (__sta_info_destroy [mac80211]) from [<bf11aa64>] (sta_info_destroy_addr_bss+0x2c/0x44 [mac80211])
[ 8814.956834] [<bf11aa64>] (sta_info_destroy_addr_bss [mac80211]) from [<bf0e0718>] (nl80211_del_station+0xe8/0xf8 [cfg80211])
[ 8814.968115] [<bf0e0718>] (nl80211_del_station [cfg80211]) from [<c033b6ac>] (genl_rcv_msg+0x2dc/0x384)
[ 8814.977461] [<c033b6ac>] (genl_rcv_msg) from [<c033a9cc>] (netlink_rcv_skb+0x54/0xb4)
[ 8814.985325] [<c033a9cc>] (netlink_rcv_skb) from [<c033b3c4>] (genl_rcv+0x28/0x34)
[ 8814.992839] [<c033b3c4>] (genl_rcv) from [<c033a39c>] (netlink_unicast+0x13c/0x1f4)
[ 8815.000528] [<c033a39c>] (netlink_unicast) from [<c033a7dc>] (netlink_sendmsg+0x2cc/0x340)
[ 8815.008829] [<c033a7dc>] (netlink_sendmsg) from [<c02fcee4>] (sock_sendmsg+0x14/0x24)
[ 8815.016694] [<c02fcee4>] (sock_sendmsg) from [<c02fd530>] (___sys_sendmsg+0x20c/0x220)
[ 8815.024644] [<c02fd530>] (___sys_sendmsg) from [<c02fe328>] (__sys_sendmsg+0x3c/0x60)
[ 8815.032508] [<c02fe328>] (__sys_sendmsg) from [<c0009c80>] (ret_fast_syscall+0x0/0x3c)
[ 8815.040590] ---[ end trace 7816f585ec7c0860 ]---
```

yuhhaurlin, 2016-07-12:
This if some kind of warning. I think your system is still all right. Please help to get kernel log related to firmware command timeout. Thanks.

yuhhaurlin, 2016-07-12:
@mystica555 Can you also help to test this version of firmware to see if it resolves your problem? If yes and original issue #55 is hard to reproduce, I will close this one and only leave issue #55 as open.

mystica555, 2016-07-12:
Due to the inability for me with my own devices to reproduce, and only the
Galaxy Note 3 w/Murata wifi being the bad egg means I have to wait until
approximately 2:30 AM for my roommate to arrive back home from her job at
the airport to be able to do any testing...

The testing consists of me changing the wireless SSID from "not the
wrt1200ac" to "the wrt1200ac" and watching all devices on the ssid
reconnect, and waiting between 10 minutes and 2 hours for her to come
upstairs banging on my door "THE INTERNET ISNT WORKING AGAIN"

Needless to say this is moderately untenable...

She does have days off mid-week, so I'll do some testing then... Please
close my new bug, as it does seem this is quite related to this exact
report, just different manifestations of what I personally think is a bug
in the mac80211 layer of your driver.

mystica555, 2016-07-12:
However, from my own dmesg, BAStream was the first command that timed out.

[ 78.096604] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[ 78.103139] br-lan: port 2(wlan0) entered forwarding state
[ 78.108683] br-lan: port 2(wlan0) entered forwarding state
[ 78.150047] device wlan0-1 entered promiscuous mode
[ 78.208907] IPv6: ADDRCONF(NETDEV_UP): wlan0-1: link is not ready
[ 78.215080] br-lan: port 5(wlan0-1) entered forwarding state
[ 78.220788] br-lan: port 5(wlan0-1) entered forwarding state
[ 78.370613] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0-1: link becomes ready
[ 78.469542] ieee80211 phy0: change: 0x20
[ 80.101571] br-lan: port 2(wlan0) entered forwarding state
[ 80.211563] br-lan: port 5(wlan0-1) entered forwarding state
[ 1311.982691] ieee80211 phy1: cmd 0x9125=BAStream timed out
[ 1311.988127] ieee80211 phy1: return code: 0x1125
[ 1311.992676] ieee80211 phy1: timeout: 0x1125
[ 1311.996891] ieee80211 phy1: destroy ba failed execution
[ 1563.049137] ieee80211 phy0: radar detected by firmware
[ 1563.582229] ieee80211 phy0: channel switch is done
[ 1563.587056] ieee80211 phy0: change: 0x40
[ 1575.199406] ieee80211 phy1: cmd 0x9122=UpdateEncryption timed out
[ 1575.205543] ieee80211 phy1: return code: 0x1122
[ 1575.210109] ieee80211 phy1: timeout: 0x1122
[ 1575.214309] ieee80211 phy1: failed execution
[ 1575.218597] wlan1-1: failed to remove key (0, b0:e8:92:06:2b:15) from
hardware (-5)
[ 1579.243490] ieee80211 phy1: cmd 0x9111=SetNewStation timed out
[ 1579.249354] ieee80211 phy1: return code: 0x1111
[ 1579.253918] ieee80211 phy1: timeout: 0x1111
[ 1579.258118] ieee80211 phy1: failed execution

mystica555, 2016-07-12:
On Tue, Jul 12, 2016 at 10:48 AM, Mike Hodson <mystica@gmail.com> wrote:

> b0:e8:92

Interestingly enough, the Murata client is _not_ the associated first MAC
address ; it is my epson printer.  'B0-E8-92' : SEIKO EPSON CORPORATION

But, it is definitely not the last if you continue from 1579 onward for a
few seconds

johnnysl, 2016-07-12:
@yuhhaurlin: system is not showing issues in the last 24 hrs. tried reboots, various clients, streaming data (netflix), putting devices in standby, etc, etc. 
Besides significantly lowering these issues, the new firmware is definitively an improvement on stability for old devices on the 2.4ghz as well.

as soon as i experience the issue again, i will report back.

yuhhaurlin, 2016-07-12:
@mystica555 Your test is on new firmware (7.2.9.24)?

yuhhaurlin, 2016-07-12:
@johnnysl Thanks.

mystica555, 2016-07-12:
So far the test I did was prior to this new firmware; i have not yet had a
chance to test the new firmware. I will have a chance in approximately 2
days to thoroughly test it, with perhaps a chance tonight or tomorrow night

jklap, 2016-07-13:
I've always seen the BAStream timeout as the first indication of failure.

Note that it occurs most frequently for me while using an Apple TV and streaming.

yuhhaurlin, 2016-07-13:
@jklap Yes. And it looks like new firmware fixed it.

jklap, 2016-07-13:
Awesome! And idea on when it will get pushed here so I can build and test?

yuhhaurlin, 2016-07-13:
@jklap Under verification. You can download the test image from the URL I mentioned in this issue list.

johnnysl, 2016-07-13:
@yuhhaurlin: finally able to trigger the event again.
Macbook Air Mid 2012, watching netflix
Ipad air 2, watching youtube
both on the 5ghz channel.

```
[ 8815.040590] ---[ end trace 7816f585ec7c0860 ]---
[174553.304241] ieee80211 phy0: cmd 0x9125=BAStream timed out
[174553.309755] ieee80211 phy0: return code: 0x1125
[174553.314401] ieee80211 phy0: timeout: 0x1125
[174553.318687] ieee80211 phy0: destroy ba failed execution
[174620.392948] ieee80211 phy0: cmd 0x9122=UpdateEncryption timed out
[174620.399159] ieee80211 phy0: return code: 0x1122
[174620.403808] ieee80211 phy0: timeout: 0x1122
[174620.408094] ieee80211 phy0: failed execution
[174620.412469] wlan0: failed to remove key (0, 50:2e:5c:32:xx:xx) from hardware (-5)
[174624.420343] ieee80211 phy0: cmd 0x9122=UpdateEncryption timed out
[174624.426595] ieee80211 phy0: return code: 0x1122
[174624.431247] ieee80211 phy0: timeout: 0x1122
[174624.435542] ieee80211 phy0: failed execution
[174624.439916] wlan0: failed to remove key (0, 70:56:81:c4:xx:xx) from hardware (-5)
[174628.446280] ieee80211 phy0: cmd 0x9122=UpdateEncryption timed out
[174628.452490] ieee80211 phy0: return code: 0x1122
[174628.457136] ieee80211 phy0: timeout: 0x1122
[174628.461422] ieee80211 phy0: failed execution
[174628.465803] wlan0: failed to remove key (2, ff:ff:ff:ff:ff:ff) from hardware (-5)
[174632.472223] ieee80211 phy0: cmd 0x9122=UpdateEncryption timed out
[174632.478439] ieee80211 phy0: return code: 0x1122
[174632.483083] ieee80211 phy0: timeout: 0x1122
[174632.487369] ieee80211 phy0: failed execution
[174632.491743] wlan0: failed to set key (2, ff:ff:ff:ff:ff:ff) to hardware (-5)
[174636.498183] ieee80211 phy0: cmd 0x9122=UpdateEncryption timed out
[174636.504401] ieee80211 phy0: return code: 0x1122
[174636.509037] ieee80211 phy0: timeout: 0x1122
[174636.513330] ieee80211 phy0: failed execution
[174636.517704] wlan0: failed to remove key (0, 78:7e:61:b8:xx:xx) from hardware (-5)
[174640.524556] ieee80211 phy0: cmd 0x9111=SetNewStation timed out
[174640.530504] ieee80211 phy0: return code: 0x1111
[174640.535148] ieee80211 phy0: timeout: 0x1111
[174640.539434] ieee80211 phy0: failed execution
[174644.544477] ieee80211 phy0: cmd 0x9111=SetNewStation timed out
[174644.550426] ieee80211 phy0: return code: 0x1111
[174644.555072] ieee80211 phy0: timeout: 0x1111
[174644.559359] ieee80211 phy0: failed execution
[174648.566401] ieee80211 phy0: cmd 0x9111=SetNewStation timed out
[174648.572350] ieee80211 phy0: return code: 0x1111
[174648.576996] ieee80211 phy0: timeout: 0x1111
[174648.581283] ieee80211 phy0: failed execution
[174652.586323] ieee80211 phy0: cmd 0x8050=broadcast_ssid_enable timed out
[174652.592975] ieee80211 phy0: return code: 0x0050
[174652.597611] ieee80211 phy0: timeout: 0x0050
[174652.601897] ieee80211 phy0: failed execution
[174656.605918] ieee80211 phy0: cmd 0x8127=SetInformationElements timed out
[174656.612656] ieee80211 phy0: return code: 0x0127
[174656.617291] ieee80211 phy0: timeout: 0x0127
[174656.621577] ieee80211 phy0: failed execution
[174660.627909] ieee80211 phy0: cmd 0x801c=80211RadioControl timed out
[174660.634214] ieee80211 phy0: return code: 0x001c
[174660.638849] ieee80211 phy0: timeout: 0x001c
[174660.643143] ieee80211 phy0: failed execution
[174664.647839] ieee80211 phy0: cmd 0x8126=SetFixedRate timed out
[174664.653707] ieee80211 phy0: return code: 0x0126
[174664.658342] ieee80211 phy0: timeout: 0x0126
[174664.662637] ieee80211 phy0: failed execution

```

johnnysl, 2016-07-20:
@yuhhaurlin I'm now running a build with kernel 4.4.15 for the last 5 days and since then not one issue anymore.
it seems that whatever triggered the issue is fixed in this new(er) Lede build.

yuhhaurlin, 2016-07-20:
@johnnysl So you mean this issue and issue #55 are gone without the need to modify driver and firmware. The Lede build uses the latest driver (10.3.0.17-20160617) and firmware 7.2.9.2?

yuhhaurlin, 2016-07-20:
@johnnysl I just checked repository of Lede. It deletes all patches and modify the Makefile to use 10.3.0.17-20160617 (The firmware should be 7.2.9.2). Can you share the image you used for mystica555 and jklap to verify if their problems are gone. Thanks.

yuhhaurlin, 2016-07-20:
@johnnysl By the way, can  you help to check what is the number for "/proc/sys/vm/min_free_kbytes".

yuhhaurlin, 2016-07-20:
@johnnysl Another modification of Lede is that mac80211 had been updated to 2016/06/20. I wonder if you can build Lede with old mac80211 and new mac80211 to check if this problem will happen on these two builds?

yuhhaurlin, 2016-07-20:
@mystica555 & jklap Can you issue "echo 16384 > /proc/sys/vm/min_free_kbyte" on your current build and check if this problem is still there? Thanks.

yuhhaurlin, 2016-07-20:
@johnnysl I just download main trunk of OpenWrt, it looks like mac80211 and mwlwifi do not have any modifications. Maybe I will also build Lede to test mwlwifi later.

thagabe, 2016-07-20:
@yuhhaurlin Yes, I have pushed the newest wireless driver to LEDE and I have not had this problem with any of my builds

yuhhaurlin, 2016-07-20:
@thagabe Can you check what is the value you got for "/proc/sys/vm/min_free_kbyte"?

thagabe, 2016-07-20:
@yuhhaurlin $ cat /proc/sys//vm/min_free_kbytes    results in: 2013

yuhhaurlin, 2016-07-20:
@thagabe Can you share your image for mystica555 and jklap to verify if their problems are gone? Thanks.

thagabe, 2016-07-20:
https://drive.google.com/open?id=0B0oFVQPGy6B7NGtON2JmNFhWNE0

LEDE r958 with latest mwlwifi and kernel 4.4.14 (might build again with 4.4.15)

yuhhaurlin, 2016-07-20:
@thagabe
mystica555 uses WRT1200AC. I don't know what device used by jklap.

thagabe, 2016-07-20:
I'll compile for 1200 since i have yet to update my build folder

yuhhaurlin, 2016-07-20:
@thagabe Thanks.

thagabe, 2016-07-20:
[https://drive.google.com/open?id=0B0oFVQPGy6B7aUxWOGZablNuc1E](LEDE Images) I ended up building an exact copy of my current image for all the versions of the WRT. 

@yuhhaurlin can you check the builds are in the correct folders?

yuhhaurlin, 2016-07-20:
@thagabe Yes, I can find four folders. By the way, I am confused about the process to submit code to LEDE. You are the developer of LEDE? The code modified in LEDE will be updated back to OpenWrt? If OpenWrt added the support for future WRT series products, LEDE will also update the code from OpenWrt?

thagabe, 2016-07-20:
@yuhhaurlin 
1) I am not a core developer of either LEDE or OpenWRT. 
2) LEDE (and now OpenWRT) is trying to gain more commits by moving the repo to github and accepting patches via the pull request feature (very useful, you might want to look into this :+1: ) since it allows for code and sanity check. 
3) LEDE is a fork of OpenWRT, right this instant they are basically the same. However as time goes on they might change and deviate. Since the split many "developers" of OpenWRT moved to LEDE because of the difficult contribution guideline OpenWRT impose and their core developers (with the powers to push upstream) took too long to push things. Thus LEDE was born. Since the loss of developers OpenWRT hasn't seen much active development but they have been porting patches from LEDE to openwrt. 
4) Not sure if LEDE is doing the same, probably but don't quote me on that. The exception seems to be my patch to LEDE which hasn't been pushed back to OpenWRT maybe I should create a PR and send it OpenWRT's way for inclusion but right now I'll be sticking with LEDE as development is more active (They really want to nail the first release by being extremely stable with 4.4.XX before moving on to the next kernel most likely 4.7 or another LTS kernel )

yuhhaurlin, 2016-07-20:
@thagabe Thanks for your information.

thagabe, 2016-07-20:
@yuhhaurlin no problem

yuhhaurlin, 2016-07-20:
@mystica555 & jklap Please use the images posted by thagabe to verify if your problems are gone. Thanks.

thagabe, 2016-07-20:
@yuhhaurlin Can you explain what was changed in the firmware? Was it the FCC's rules on radio channels and stuff?

yuhhaurlin, 2016-07-20:
@thagabe Yes. It also has enhancement for BA related code.

thagabe, 2016-07-20:
@yuhhaurlin what is BA? Haha trying to learn the code

yuhhaurlin, 2016-07-20:
@thagabe Sorry. It is Block ACK stream.

thagabe, 2016-07-20:
@yuhhaurlin Thanks

[23 more replies]

closed 2016-08-13 by yuhhaurlin