kaloz/mwlwifi #95: Crash on WRT1200AC when transfering a lot of data

lancelot666de, 2016-07-15:
**Setup**
* Linksys WRT1200AC 
* OpenWrt DESIGNATED DRIVER (Bleeding Edge, mrf16060412@r49377)
   - Image available at https://openwrt.nc-network.io/
* On usb2, an external HDD with Btrfs is attached and shared via smb.
* On usb1, a usb pen is attached providing some additional swap space - not a prerequisite

```
cat /sys/kernel/debug/ieee80211/phy0/mwlwifi/info

driver name: mwlwifi
chip type: 88W8864
hw version: 7
driver version: 10.3.0.17-20160603
firmware version: 0x07020902
mac address: 00:25:9c:13:cd:96
2g: disable
5g: enable
antenna: 2 2
irq number: 104
iobase0: e1280000
iobase1: e1400000
tx limit: 768
rx limit: 64
ap macid support: 0000ffff
sta macid support: 00010000
macid used: 00000001
qe trigger number: 80379
```

**Original Problem**
When copying a lot of data (4GB - 12 GB) over the Wifi to the smb share,  logread puts the following line into the log:

> ieee80211 phy1: failed to reallocate TX buffer

After this line hits the logs, the copying continues for a while before the system crashes. The crash is not deterministic. I observed crashes:
> * Unable to handle kernel paging request at virtual address xxx  -> reboot shortly afterwards
> * Illegal instruction printed on the console -> reboot immediately
> * reboot
> * Wifi 2.4 GhZ dies

To pin down the issue, I ran iperf on the wifi interface. No crash could be observered but again the following error message:

> ieee80211 phy1: failed to reallocate TX buffer

**Reproduing**

Next, I created two test cases:
1. Disable Wifi, run iperf over ethernet while copying data on HDD via terminal. 
2. Enabled Wifi, run iperf over Wifi while copying data on HDD via terminal. 

In the first test case, I could not observe a single crash. In the second case, a crash occurs regularly after some GB transmitted. Most often, I can oberserve the message in the logs:
> ieee80211 phy1: failed to reallocate TX buffer

It seems that copying data in the background increases the probability that an error causes a crash.

Here are some logs from a test run (copying data via SMB):

```
Fri Jul 15 08:25:40 2016 daemon.info hostapd: wlan1: STA 00:04:20:2a:fe:ed WPA: group key handshake completed (RSN)
Fri Jul 15 08:25:40 2016 daemon.info hostapd: wlan1: STA 84:34:97:0c:4a:14 WPA: group key handshake completed (RSN)
Fri Jul 15 08:25:40 2016 daemon.info hostapd: wlan1: STA 7c:7a:91:37:22:24 WPA: group key handshake completed (RSN)
Fri Jul 15 08:25:41 2016 daemon.info hostapd: wlan1: STA 54:72:4f:06:69:ca WPA: group key handshake completed (RSN)
Fri Jul 15 08:25:49 2016 daemon.info hostapd: wlan0: STA 78:f8:82:b3:24:7e WPA: group key handshake completed (RSN)
Fri Jul 15 08:26:02 2016 kern.debug kernel: [ 2457.466201] ieee80211 phy0: failed to reallocate TX buffer
Fri Jul 15 08:26:02 2016 kern.debug kernel: [ 2457.467829] ieee80211 phy1: failed to reallocate TX buffer
Fri Jul 15 08:26:02 2016 kern.debug kernel: [ 2457.477245] ieee80211 phy1: failed to reallocate TX buffer
Fri Jul 15 08:29:28 2016 kern.debug kernel: [ 2662.956617] ieee80211 phy0: failed to reallocate TX buffer
Fri Jul 15 08:29:28 2016 kern.debug kernel: [ 2662.962141] ieee80211 phy1: failed to reallocate TX buffer
Fri Jul 15 08:30:00 2016 cron.info crond[1427]: USER root pid 2579 cmd /sbin/fan_ctrl.sh
Fri Jul 15 08:30:03 2016 kern.debug kernel: [ 2697.604851] ieee80211 phy0: failed to reallocate TX buffer
Fri Jul 15 08:30:03 2016 kern.debug kernel: [ 2697.606732] ieee80211 phy1: failed to reallocate TX buffer
Fri Jul 15 08:30:03 2016 kern.debug kernel: [ 2697.615897] ieee80211 phy1: failed to reallocate TX buffer
Fri Jul 15 08:30:32 2016 kern.debug kernel: [ 2726.908704] ieee80211 phy1: failed to reallocate TX buffer
Fri Jul 15 08:30:32 2016 kern.debug kernel: [ 2726.914470] ieee80211 phy0: failed to reallocate TX buffer
Fri Jul 15 08:31:00 2016 kern.debug kernel: [ 2754.752832] ieee80211 phy1: failed to reallocate TX buffer
Fri Jul 15 08:33:03 2016 kern.debug kernel: [ 2877.672179] ieee80211 phy1: failed to reallocate TX buffer
Fri Jul 15 08:35:00 2016 cron.info crond[1427]: USER root pid 2589 cmd /sbin/fan_ctrl.sh
Fri Jul 15 08:35:40 2016 daemon.info hostapd: wlan1: STA 7c:7a:91:37:22:24 WPA: group key handshake completed (RSN)
Fri Jul 15 08:35:40 2016 daemon.info hostapd: wlan1: STA 00:04:20:2a:fe:ed WPA: group key handshake completed (RSN)
Fri Jul 15 08:35:40 2016 daemon.info hostapd: wlan1: STA 84:34:97:0c:4a:14 WPA: group key handshake completed (RSN)
Fri Jul 15 08:35:42 2016 daemon.info hostapd: wlan1: STA 54:72:4f:06:69:ca WPA: group key handshake completed (RSN)
Fri Jul 15 08:35:49 2016 daemon.info hostapd: wlan0: STA 78:f8:82:b3:24:7e WPA: group key handshake completed (RSN)
Fri Jul 15 08:38:03 2016 kern.debug kernel: [ 3177.852916] ieee80211 phy0: failed to reallocate TX buffer
Fri Jul 15 08:38:03 2016 kern.debug kernel: [ 3177.858438] ieee80211 phy1: failed to reallocate TX buffer
Fri Jul 15 08:40:00 2016 cron.info crond[1427]: USER root pid 2601 cmd /sbin/fan_ctrl.sh
Fri Jul 15 08:41:03 2016 kern.debug kernel: [ 3357.952210] ieee80211 phy1: failed to reallocate TX buffer
Fri Jul 15 08:41:32 2016 kern.debug kernel: [ 3386.628644] ieee80211 phy0: failed to reallocate TX buffer
Fri Jul 15 08:41:32 2016 kern.debug kernel: [ 3386.634172] ieee80211 phy1: failed to reallocate TX buffer
```

yuhhaurlin, 2016-07-15:
Memory is exhausted. This kind of problem had been reported many times, but it should not be problem of mwlwifi. Mwlwifi does not have memory leakage problem. I suggest that you can try to get help from community.

closed 2016-07-15 by yuhhaurlin

lancelot666de, 2016-07-15:
Memory exhaustion was also my first guess thus I added the the USB pen with 8GB as swap. The router crashed once with htop enabled. 42 of 502 MB were used. The 8 GB of the swap were not touched at all. I personally don't think that the bug is a memory leakage or memory exhaustion issue. Some overflow or double free issue is for me more likely. Based on this information, do you think that this is a memory exhaustion issue?

yuhhaurlin, 2016-07-15:
Yes.

yuhhaurlin, 2016-07-15:
Maybe you can get some prebuilt versions to do test.

lancelot666de, 2016-07-15:
Now I'm puzzled. The images I used  are prebuilt. I tested with the stable CC release, stable CC release patched with a prebuilt version newest mwlwifi driver and the image of mrfreeze referenced above. Do you have specific prebuilts in mind?

I'll dig in the forums if I find more information on memory exhaustion and how to overcome it.

yuhhaurlin, 2016-07-15:
No. I just know the same problem had been reported before. From our own test, mwlwifi should not have memory leakage problem.

lancelot666de, 2016-07-17:
You were right! If other people stumble upon the same issue, it seems that the following change resolves the issue:
`echo 16384 > /proc/sys/vm/min_free_kbytes`

At least I could no longer observe a crash with this set. Thanks for pointing me in the right direction.

yuhhaurlin, 2016-07-17:
Thanks for your information.