kaloz/mwlwifi #96: New manifestation of wireless client timing out- still seems rekey related.

mystica555, 2016-07-17:
The dmesg shows nothing of consequence at this time. This is the mostly-working G3 D851. 
I've recently added a second 5ghz capable phone to the ssid. It was associated and performing a speedtest concurrently with this, as both wifi clients stopped concurrently. 

The secondary client did not disassociate, but did stop passing data. I immediately switched it to the other wifi so do not have further logs of its own activity. 

00:90:4c = epigram G3 D851
68:c4:4d = unknown-make-chipset(assumed qca?) lenovo moto g(4) 2016 variant model XT1625. 

Dmesg indicates the channel switched from 132 as configured > 149 as evidenced by 'iw dev wlan0 info' far after bootup (similar problem to another bug report; i don't mind -as- much as some might) ; this change happens far before the problem manifests itself by both clients stopping data transfer. 

[   75.372990] ieee80211 phy0: change: 0x40
[   75.480050] ieee80211 phy0: change: 0x100
[   75.558683] ieee80211 phy0: change: 0x100
[   75.571752] ieee80211 phy0: change: 0x42
[   75.679201] ieee80211 phy0: change: 0x40
[   75.863349] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   75.869794] br-lan: port 2(wlan0) entered forwarding state
[   75.875340] br-lan: port 2(wlan0) entered forwarding state
[   75.977293] ieee80211 phy0: change: 0x20
[   77.874692] br-lan: port 2(wlan0) entered forwarding state
[34449.252368] ieee80211 phy0: radar detected by firmware
[34449.850292] ieee80211 phy0: channel switch is done
[34449.855115] ieee80211 phy0: change: 0x40
root@OpenWrt:~# uptime
 04:52:43 up 1 day, 15:27,  load average: 0.00, 0.00, 0.00
root@OpenWrt:~# cat /proc/uptime
142042.63 283551.73

Below, the brief G(4) log, followed by the very long lg g3 log.

(logread |grep _the_mac_address_)

Sun Jul 17 03:54:06 2016 daemon.info hostapd: wlan0: STA 68:c4:4d:a5:33:2f IEEE 802.11: authenticated
Sun Jul 17 03:54:06 2016 daemon.info hostapd: wlan0: STA 68:c4:4d:a5:33:2f IEEE 802.11: associated (aid 1)
Sun Jul 17 03:54:06 2016 daemon.info hostapd: wlan0: STA 68:c4:4d:a5:33:2f WPA: pairwise key handshake completed (RSN)
Sun Jul 17 03:54:06 2016 daemon.info dnsmasq-dhcp[2304]: DHCPDISCOVER(br-lan) 68:c4:4d:a5:33:2f 
Sun Jul 17 03:54:06 2016 daemon.info dnsmasq-dhcp[2304]: DHCPOFFER(br-lan) 192.168.17.157 68:c4:4d:a5:33:2f 
Sun Jul 17 03:54:06 2016 daemon.info dnsmasq-dhcp[2304]: DHCPREQUEST(br-lan) 192.168.17.157 68:c4:4d:a5:33:2f 
Sun Jul 17 03:54:06 2016 daemon.info dnsmasq-dhcp[2304]: DHCPACK(br-lan) 192.168.17.157 68:c4:4d:a5:33:2f android-feaad9918787595f
Sun Jul 17 03:56:44 2016 daemon.info hostapd: wlan0: STA 68:c4:4d:a5:33:2f WPA: group key handshake completed (RSN)
Sun Jul 17 04:06:44 2016 daemon.info hostapd: wlan0: STA 68:c4:4d:a5:33:2f WPA: group key handshake completed (RSN)
Sun Jul 17 04:16:44 2016 daemon.info hostapd: wlan0: STA 68:c4:4d:a5:33:2f WPA: group key handshake completed (RSN)
Sun Jul 17 04:26:45 2016 daemon.info hostapd: wlan0: STA 68:c4:4d:a5:33:2f WPA: group key handshake completed (RSN)

And the less-than-happy LG G3 log:

Sun Jul 17 04:26:06 2016 daemon.info hostapd: wlan0: STA 00:90:4c:c5:12:38 IEEE 802.11: authenticated
Sun Jul 17 04:26:06 2016 daemon.info hostapd: wlan0: STA 00:90:4c:c5:12:38 IEEE 802.11: associated (aid 2)
Sun Jul 17 04:26:06 2016 daemon.info hostapd: wlan0: STA 00:90:4c:c5:12:38 WPA: pairwise key handshake completed (RSN)
Sun Jul 17 04:26:06 2016 daemon.info dnsmasq-dhcp[2304]: DHCPDISCOVER(br-lan) 00:90:4c:c5:12:38 
Sun Jul 17 04:26:06 2016 daemon.info dnsmasq-dhcp[2304]: DHCPOFFER(br-lan) 192.168.17.151 00:90:4c:c5:12:38 
Sun Jul 17 04:26:06 2016 daemon.info dnsmasq-dhcp[2304]: DHCPREQUEST(br-lan) 192.168.17.151 00:90:4c:c5:12:38 
Sun Jul 17 04:26:06 2016 daemon.info dnsmasq-dhcp[2304]: DHCPACK(br-lan) 192.168.17.151 00:90:4c:c5:12:38 mY5t1c4-pH0n3
Sun Jul 17 04:26:42 2016 daemon.info hostapd: wlan0: STA 00:90:4c:c5:12:38 WPA: group key handshake completed (RSN)
Sun Jul 17 04:31:45 2016 daemon.info hostapd: wlan0: STA 00:90:4c:c5:12:38 IEEE 802.11: disassociated
Sun Jul 17 04:31:46 2016 daemon.info hostapd: wlan0: STA 00:90:4c:c5:12:38 IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Sun Jul 17 04:31:48 2016 daemon.info hostapd: wlan0: STA 00:90:4c:c5:12:38 IEEE 802.11: authenticated
Sun Jul 17 04:31:48 2016 daemon.info hostapd: wlan0: STA 00:90:4c:c5:12:38 IEEE 802.11: associated (aid 2)
Sun Jul 17 04:31:48 2016 daemon.info hostapd: wlan0: STA 00:90:4c:c5:12:38 WPA: pairwise key handshake completed (RSN)
Sun Jul 17 04:32:25 2016 daemon.info hostapd: wlan0: STA 00:90:4c:c5:12:38 IEEE 802.11: disassociated
Sun Jul 17 04:32:26 2016 daemon.info hostapd: wlan0: STA 00:90:4c:c5:12:38 IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Sun Jul 17 04:32:28 2016 daemon.info hostapd: wlan0: STA 00:90:4c:c5:12:38 IEEE 802.11: authenticated
Sun Jul 17 04:32:28 2016 daemon.info hostapd: wlan0: STA 00:90:4c:c5:12:38 IEEE 802.11: associated (aid 2)
Sun Jul 17 04:32:28 2016 daemon.info hostapd: wlan0: STA 00:90:4c:c5:12:38 WPA: pairwise key handshake completed (RSN)
Sun Jul 17 04:33:04 2016 daemon.info hostapd: wlan0: STA 00:90:4c:c5:12:38 IEEE 802.11: disassociated
Sun Jul 17 04:33:05 2016 daemon.info hostapd: wlan0: STA 00:90:4c:c5:12:38 IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Sun Jul 17 04:33:08 2016 daemon.info hostapd: wlan0: STA 00:90:4c:c5:12:38 IEEE 802.11: authenticated
Sun Jul 17 04:33:08 2016 daemon.info hostapd: wlan0: STA 00:90:4c:c5:12:38 IEEE 802.11: associated (aid 2)
Sun Jul 17 04:33:08 2016 daemon.info hostapd: wlan0: STA 00:90:4c:c5:12:38 WPA: pairwise key handshake completed (RSN)
Sun Jul 17 04:33:44 2016 daemon.info hostapd: wlan0: STA 00:90:4c:c5:12:38 IEEE 802.11: disassociated
[... 19 lines cut ...]
Sun Jul 17 04:36:28 2016 daemon.info hostapd: wlan0: STA 00:90:4c:c5:12:38 IEEE 802.11: authenticated
Sun Jul 17 04:36:28 2016 daemon.info hostapd: wlan0: STA 00:90:4c:c5:12:38 IEEE 802.11: associated (aid 1)
Sun Jul 17 04:36:28 2016 daemon.info hostapd: wlan0: STA 00:90:4c:c5:12:38 WPA: pairwise key handshake completed (RSN)
Sun Jul 17 04:36:50 2016 daemon.info hostapd: wlan0: STA 00:90:4c:c5:12:38 IEEE 802.11: authenticated
Sun Jul 17 04:36:50 2016 daemon.info hostapd: wlan0: STA 00:90:4c:c5:12:38 IEEE 802.11: associated (aid 1)
Sun Jul 17 04:36:50 2016 daemon.info hostapd: wlan0: STA 00:90:4c:c5:12:38 WPA: pairwise key handshake completed (RSN)
Sun Jul 17 04:37:26 2016 daemon.info hostapd: wlan0: STA 00:90:4c:c5:12:38 IEEE 802.11: disassociated
Sun Jul 17 04:37:27 2016 daemon.info hostapd: wlan0: STA 00:90:4c:c5:12:38 IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Sun Jul 17 04:37:30 2016 daemon.info hostapd: wlan0: STA 00:90:4c:c5:12:38 IEEE 802.11: authenticated
Sun Jul 17 04:37:30 2016 daemon.info hostapd: wlan0: STA 00:90:4c:c5:12:38 IEEE 802.11: associated (aid 1)
Sun Jul 17 04:37:30 2016 daemon.info hostapd: wlan0: STA 00:90:4c:c5:12:38 WPA: pairwise key handshake completed (RSN)
Sun Jul 17 04:38:06 2016 daemon.info hostapd: wlan0: STA 00:90:4c:c5:12:38 IEEE 802.11: disassociated
Sun Jul 17 04:38:07 2016 daemon.info hostapd: wlan0: STA 00:90:4c:c5:12:38 IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Sun Jul 17 04:38:10 2016 daemon.info hostapd: wlan0: STA 00:90:4c:c5:12:38 IEEE 802.11: authenticated
Sun Jul 17 04:38:10 2016 daemon.info hostapd: wlan0: STA 00:90:4c:c5:12:38 IEEE 802.11: associated (aid 1)
Sun Jul 17 04:38:10 2016 daemon.info hostapd: wlan0: STA 00:90:4c:c5:12:38 WPA: pairwise key handshake completed (RSN)
Sun Jul 17 04:38:46 2016 daemon.info hostapd: wlan0: STA 00:90:4c:c5:12:38 IEEE 802.11: disassociated
Sun Jul 17 04:38:47 2016 daemon.info hostapd: wlan0: STA 00:90:4c:c5:12:38 IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Sun Jul 17 04:41:44 2016 daemon.info hostapd: wlan0: STA 00:90:4c:c5:12:38 IEEE 802.11: authenticated
Sun Jul 17 04:41:44 2016 daemon.info hostapd: wlan0: STA 00:90:4c:c5:12:38 IEEE 802.11: associated (aid 1)
Sun Jul 17 04:41:44 2016 daemon.info hostapd: wlan0: STA 00:90:4c:c5:12:38 WPA: pairwise key handshake completed (RSN)
Sun Jul 17 04:41:44 2016 daemon.info dnsmasq-dhcp[2304]: DHCPDISCOVER(br-lan) 00:90:4c:c5:12:38 
Sun Jul 17 04:41:44 2016 daemon.info dnsmasq-dhcp[2304]: DHCPOFFER(br-lan) 192.168.17.151 00:90:4c:c5:12:38 
Sun Jul 17 04:41:44 2016 daemon.info dnsmasq-dhcp[2304]: DHCPREQUEST(br-lan) 192.168.17.151 00:90:4c:c5:12:38 
Sun Jul 17 04:41:44 2016 daemon.info dnsmasq-dhcp[2304]: DHCPACK(br-lan) 192.168.17.151 00:90:4c:c5:12:38 mY5t1c4-pH0n3

yuhhaurlin, 2016-07-17:
I don't know what the problem is.

yuhhaurlin, 2016-07-18:
I will close this one first. If you want to report problem, please:
1. Let me know what version of mwlwifi driver you are tested.
2. Configuration of you AP.
3. Setup of your test environment.
4. What problem you encounter and the way to reproduce it.
5. If you can do some basic clarification of the problem would be better.

Thanks.

closed 2016-07-18 by yuhhaurlin