kaloz/mwlwifi #97: Massive WiFi speed drop few meters away WRT1200AC

Driver007, 2016-07-22:
Hi,
Biggest problem whit WRT1200AC WiFi speed drop few meters away.
I have 600Mbps line, and near my router I reach about 500Mbps whit Intel 7260AC
Problems starts over 2 meters away drops till 300Mbps in other room which is 3-4 meters away I will have 150Mbps.
So whats the problem, in other room I could understand walls or something but 2 meters on open space and drop is from 500Mbps to 300Mbps? thats make no sense, and on 2.4Ghz speeds drops also quite crazy.

And is pretty the same on all my devices.

now I am on:
root@lede:~# opkg list-installed | grep mwlwifi
kmod-mwlwifi - 4.4.15+10.3.0.17-20160617-1
Kernel 4.4.15

Same problems whit every version.

What logs do you need for more information?
thanks

Chadster766, 2016-07-22:
This sounds more like signal interference than a driver issue.

Use a Wifi Analzer app or use the terminal commands in OpenWRT to look for overlapping wireless channels on both the 2.4Ghz and 5Ghz frequencies.

Do you have 2.4Ghz and 5Ghz cordless phones at the location? Those probably wouldn't show in your standard wifi analyzer.

Driver007, 2016-07-22:
on 2.4Ghz I could understand, there are few other routers, previous asus router did 200/200Mbps on 2.4Ghz N, now 100Mbps on same conditions.

But on 5Ghz there aren't any other router, all channels are free, so I am sitting 4 meters ahead my router and speeds are 50Mbps sometimes better, sometimes it worse (whit no wall, just open space).

I did buy this router for IPTV purpose, but didn't expect such bad wifi signal.
Now I am just thinking whats wrong.

And no, I don't have cordless phone or any other devices that could interfere

config wifi-device 'radio0'
        option type 'mac80211'
        option hwmode '11a'
        option path 'soc/soc:pcie-controller/pci0000:00/0000:00:01.0/0000:01:00.0'
        option htmode 'VHT80'
        option country 'DE'
        option channel '100'
        option txpower '27'

config wifi-device 'radio1'
        option type 'mac80211'
        option path 'soc/soc:pcie-controller/pci0000:00/0000:00:02.0/0000:02:00.0'
        option txpower '20'
        option hwmode '11g'
        option country 'DE'
        option noscan '1'
        option htmode 'HT40' (whit HT20 is terrible, so no, that's not an option, I need speed, as I said asus did better whit same settings)
        option channel '6'

sorry if I look rude, but I am desperate to find answer for this situation.

Chadster766, 2016-07-22:
what is it like if you don't specify the txpower on the 5Ghz?

Channel 100 seems odd to me. If you don't have any other interfering SSIDs than the default of 36 should be fine. I would recommend at least trying channel 36.

Do you have any range extender devices?

Driver007, 2016-07-22:
whit no TX same problems, tried every channel and I guess those channels makes no sense, country regulations maybe mean something.

no range extenders.

Chadster766, 2016-07-22:
Well your troubleshooting has basically ruled out configuration and interference issues.

yuhhaurlin, 2016-07-22:
What is the number you got if running stock firmware?

Driver007, 2016-07-23:
never used stock firmware but for testing purpose: it was better a little
tried ddwrt and it had most stable 2.4ghz 190/190Mbps but 5ghz had big problems

was playing whit openwrt almost all day whit region settings, sometimes 2.4gz works quite good but 5ghz sucks and opposite

yuhhaurlin, 2016-07-23:
1. 2.4g would be easily has interference.
2. Please test stock firmware as reference.
3. This only happened for one client?
4. If using the same client and test with other AP, does the problem happen?

Driver007, 2016-07-25:
1. I don't mind now what speeds are on 2.4ghz, it's quite good, and somehow managed to expand range.
2. Was testing stock, it did a little better. 5ghz range was better.
3. It happens on multiple clients, sometimes speeds are better sometimes worse on same position. Like from 170Mbps to 350Mbps when downloading torrents etc. 
4. Have wr1043nd it has rock stable WiFi on same conditions

I did bought a new dual band WiFi USB, when it arrives I will test more, I have some thoughts that my main PC has some problems whit intel WiFi card.

yuhhaurlin, 2016-07-25:
Your description confuses me. Maybe when you get new client, you can do more tests.

thagabe, 2016-07-29:
@Driver007 Intel wifi+bluetooth seem to have driver issues that sometimes spring up (a while back i had solid connection with my asus rt-n56u my ap at another place in the house but when connected to wrt1900ac's 2.4GHz it slowed and disconnected, but now the connection seems to have been solid to my wrt1900ac). Try this:
-Make sure to update drivers and pci bus on DEVICE MANGER by right-clicking the device and clicking "Update driver" 
-is your Intel 7260 card running "18.33.3.2" driver (Windows 10)?
If linux:
-Which distro are you using? Arch seems solid atm otherwise look into System76's patches for better performance on the wifi while using battery (Not needed in arch)

Driver007, 2016-07-29:
Thanks for suggestions.
Now I am using latest drivers for windows, sometimes I use linux but very often as I remember speeds are same but can't say anything about distance.

Still thinking that my intel card is not working properly.
Was fixing some friends PC whit atheros N card and at my kitchen where is furthest distance from my router it had 7MB/s wheres my intel had 1MB/s
Waiting for AC usb adapter will see wheres the problem for final conclusion, router or my intel card.

closed 2016-10-14 by yuhhaurlin