kaloz/mwlwifi #98: Driver crashing

borkra, 2016-07-28:
Latest driver from Lede (week old) firmware 24. Driver is crashing sometimes. 
Crash dump below:

[383968.013555] ------------[ cut here ]------------
[383968.018347] WARNING: CPU: 0 PID: 30382 at compat-wireless-2016-06-20/net/mac80211/agg-tx.c:347 ___ieee80211_stop_tx_ba_session+0x13c/0x200 [mac80211]()
[383968.032087] Modules linked in: pppoe ppp_async iptable_nat pppox ppp_generic nf_nat_ipv4 nf_conntrack_ipv6 nf_conntrack_ipv4 ipt_REJECT ipt_MASQUERADE xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_recent xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_id xt_hl xt_helper xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_HL xt_DSCP xt_CT xt_CLASSIFY slhc nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_nat nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache nf_conntrack iptable_raw iptable_mangle iptable_filter ipt_ECN ip_tables crc_ccitt sch_teql sch_tbf sch_sfq sch_red sch_prio sch_pie sch_htb sch_gred sch_fq sch_dsmark sch_codel em_text em_nbyte em_meta em_cmp cls_basic act_police act_ipt act_skbedit act_mirred em_u32 cls_u32 cls_tcindex cls_flow cls_route cls_fw sch_hfsc sch_ingress mwlwifi mac80211 cfg80211 compat ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_raw ip6table_mangle ip6table_filter ip6_tables x_tables ifb cbc aead crypto_null gpio_button_hotplug
[383968.128776] CPU: 0 PID: 30382 Comm: hostapd Not tainted 4.4.15 #2
[383968.134981] Hardware name: Marvell Armada 380/385 (Device Tree)
[383968.141022] [<c001fab8>] (unwind_backtrace) from [<c001c420>] (show_stack+0x10/0x14)
[383968.148890] [<c001c420>] (show_stack) from [<c019b5ec>] (dump_stack+0x8c/0xa0)
[383968.156233] [<c019b5ec>] (dump_stack) from [<c0028b84>] (warn_slowpath_common+0x94/0xb0)
[383968.164446] [<c0028b84>] (warn_slowpath_common) from [<c0028c3c>] (warn_slowpath_null+0x1c/0x24)
[383968.173379] [<c0028c3c>] (warn_slowpath_null) from [<bf0926c8>] (___ieee80211_stop_tx_ba_session+0x13c/0x200 [mac80211])
[383968.184429] [<bf0926c8>] (___ieee80211_stop_tx_ba_session [mac80211]) from [<bf092b80>] (__ieee80211_stop_tx_ba_session+0x30/0x44 [mac80211])
[383968.197302] [<bf092b80>] (__ieee80211_stop_tx_ba_session [mac80211]) from [<bf091814>] (ieee80211_sta_tear_down_BA_sessions+0x48/0x70 [mac80211])
[383968.210524] [<bf091814>] (ieee80211_sta_tear_down_BA_sessions [mac80211]) from [<bf088938>] (ieee80211_sta_eosp+0x1d8/0x54c [mac80211])
[383968.222872] [<bf088938>] (ieee80211_sta_eosp [mac80211]) from [<bf08b9d8>] (__sta_info_destroy+0xc/0x28 [mac80211])
[383968.233477] [<bf08b9d8>] (__sta_info_destroy [mac80211]) from [<bf08ba64>] (sta_info_destroy_addr_bss+0x2c/0x44 [mac80211])
[383968.244775] [<bf08ba64>] (sta_info_destroy_addr_bss [mac80211]) from [<bf050718>] (nl80211_del_station+0xe8/0xf8 [cfg80211])
[383968.256144] [<bf050718>] (nl80211_del_station [cfg80211]) from [<c0344edc>] (genl_rcv_msg+0x2dc/0x384)
[383968.265578] [<c0344edc>] (genl_rcv_msg) from [<c03441fc>] (netlink_rcv_skb+0x54/0xb4)
[383968.273528] [<c03441fc>] (netlink_rcv_skb) from [<c0344bf4>] (genl_rcv+0x28/0x34)
[383968.281129] [<c0344bf4>] (genl_rcv) from [<c0343bcc>] (netlink_unicast+0x13c/0x1f4)
[383968.288904] [<c0343bcc>] (netlink_unicast) from [<c034400c>] (netlink_sendmsg+0x2cc/0x340)
[383968.297291] [<c034400c>] (netlink_sendmsg) from [<c03017e8>] (sock_sendmsg+0x14/0x24)
[383968.305242] [<c03017e8>] (sock_sendmsg) from [<c0301e34>] (___sys_sendmsg+0x20c/0x220)
[383968.313278] [<c0301e34>] (___sys_sendmsg) from [<c0302c2c>] (__sys_sendmsg+0x3c/0x60)
[383968.321230] [<c0302c2c>] (__sys_sendmsg) from [<c0009c80>] (ret_fast_syscall+0x0/0x3c)
[383968.329295] ---[ end trace b4f16c12187ec159 ]---

thagabe, 2016-07-29:
Which model do you have? You should rebuild LEDE since some important changes have gone into it. mwlwifi, for me, has been flawless since i merged the latest to LEDE trunk. What config are you using and how did you flash (sysupgrade, usb-tty,etc)?

borkra, 2016-07-29:
Thank you for the response.

1. Router is 1900ACv2. 
2. I have the latest driver and the latest firmware, as you can see in the logs. There is nothing newer in lede git. BTW, Lede build points to this project, for driver source.
3. As you can see from the crash report, the crash happens during BA processing. And this happens to be exactly the area yuhhaurlin changed in the firmware v24.... According to his statements in issue #94.
4. I do monitor Lede commits and did not see anything relevant in there. No changes to nl80211, or hostapd. There was one patch to hostapd yesterday, but even that flagged as change for mesh.
5. Yes this build is the most stable build I seen. And it is the first build that worked longer then 2 days. And I wouldn't even noticed the crash if I haven't checked the logs. 

 So yes I can change the build, and wait whatever time for the crash to reappear. Again this is random and very infrequent. Doubt though it will be qualified as a solution.

yuhhaurlin, 2016-07-29:
This is not crash, it is warning that mac80211 will try to delete a BA which is not existed. It is very hard to happen and it won't affect normal operation. I will close it. Thanks.

closed 2016-07-29 by yuhhaurlin