kaloz/mwlwifi #99: about Marvell Avastar 88W8897 for surface pro 4 hackintosh

yninan, 2016-07-29:
Guys ，I'm not sure is here the right place to ask this question.
but I saw you guys do something about the Marvell Avastar 88W8897 chip ,that also in surface pro 4.
I have installed el capitian on my SP4 but the wifi chip still can't drived all over the world ,so I come here hope you guys can make the first thing over the globe to drive it for hackintosher .

thanks.

thagabe, 2016-07-29:
@yninan what does "guys can make the first thing over the globe to drive it for hackintosher" mean?

lantis1008, 2016-07-30:
He's installed Mac OS X onto a surface pro 4 making it a hackintosh. It is notoriously difficult to get reliable/any drivers for hardware that wasn't originally found in an apple product. 
He wants a driver for the wifi chip that will work for him, and it sounds like there is no alternatives available.

yuhhaurlin, 2016-08-01:
I looks like this item should not be checked here. I will close this one. Thanks.

closed 2016-08-01 by yuhhaurlin