kaloz/mwlwifi #102: BAStream timed out

jklap, 2016-08-19:
Also see commentary on #94.

Running firmware 0x7020819.

During streaming (Apple TV) I'll see the following error(s). Note too, once the complete list of errors below are logged it seems to stop logging any further errors.

The Apple TV is connected via 5GHz.

Once the error occurs, any devices connected on 5GHz show as connected by can't do anything. Yet ethernet and 2.4GHz seem fine.

What is curious about this is that I'd been having this issue pretty consistently but then dropped the new firmware into place about a week ago and had no issues (but still lots of streaming with the Olympics happening). Then yesterday morning I rebooted the router (had to re-route a power cord) and within 5 minutes of streaming this afternoon it errored. Reboot and within 5 minutes it errored again.

Besides the reboot nothing has changed on the router since the firmware was updated.

[ 1101.388664] ieee80211 phy1: cmd 0x9125=BAStream timed out
[ 1101.394147] ieee80211 phy1: return code: 0x1125
[ 1101.398717] ieee80211 phy1: timeout: 0x1125
[ 1101.402979] ieee80211 phy1: destroy ba failed execution
[ 1234.022071] ieee80211 phy1: cmd 0x9122=UpdateEncryption timed out
[ 1234.028301] ieee80211 phy1: return code: 0x1122
[ 1234.032928] ieee80211 phy1: timeout: 0x1122
[ 1234.037150] ieee80211 phy1: failed execution
[ 1234.041459] wlan1: failed to remove key (0, b0:34:95:2b:38:11) from hardware (-5)
[ 1238.071392] ieee80211 phy1: cmd 0x9122=UpdateEncryption timed out
[ 1238.077584] ieee80211 phy1: return code: 0x1122
[ 1238.082205] ieee80211 phy1: timeout: 0x1122
[ 1238.086444] ieee80211 phy1: failed execution
[ 1238.090749] wlan1: failed to remove key (0, cc:20:e8:13:4f:33) from hardware (-5)
[ 1242.524740] ieee80211 phy1: cmd 0x9122=UpdateEncryption timed out
[ 1242.532235] ieee80211 phy1: return code: 0x1122
[ 1242.538151] ieee80211 phy1: timeout: 0x1122
[ 1242.543810] ieee80211 phy1: failed execution
[ 1242.549537] wlan1: failed to remove key (0, 7c:d1:c3:d2:54) from hardware (-5)
[ 1246.966445] ieee80211 phy1: cmd 0x9122=UpdateEncryption timed out
[ 1246.972636] ieee80211 phy1: return code: 0x1122
[ 1246.977202] ieee80211 phy1: timeout: 0x1122
[ 1246.981413] ieee80211 phy1: failed execution
[ 1246.985784] wlan1: failed to remove key (0, c8:69:cd:3c:38:78) from hardware (-5)
[ 1251.389246] ieee80211 phy1: cmd 0x9122=UpdateEncryption timed out
[ 1251.395443] ieee80211 phy1: return code: 0x1122
[ 1251.400010] ieee80211 phy1: timeout: 0x1122
[ 1251.404289] ieee80211 phy1: failed execution
[ 1251.408596] wlan1: failed to remove key (1, ff:ff:ff:ff:ff:ff) from hardware (-5)
root@faster:/# [ 1255.703941] ieee80211 phy1: cmd 0x9122=UpdateEncryption timed out
[ 1255.710091] ieee80211 phy1: return code: 0x1122
[ 1255.714741] ieee80211 phy1: timeout: 0x1122
[ 1255.718975] ieee80211 phy1: failed execution
[ 1255.723326] wlan1: failed to set key (1, ff:ff:ff:ff:ff:ff) to hardware (-5)
[ 1260.137050] ieee80211 phy1: cmd 0x9122=UpdateEncryption timed out
[ 1260.143247] ieee80211 phy1: return code: 0x1122
[ 1260.147814] ieee80211 phy1: timeout: 0x1122
[ 1260.152102] ieee80211 phy1: failed execution
[ 1260.156403] wlan1: failed to remove key (0, 7c:c3:a1:b3:e0:54) from hardware (-5)
[ 1264.418027] ieee80211 phy1: cmd 0x9111=SetNewStation timed out
[ 1264.423943] ieee80211 phy1: return code: 0x1111
[ 1264.428514] ieee80211 phy1: timeout: 0x1111
[ 1264.432787] ieee80211 phy1: failed execution
[ 1268.453850] ieee80211 phy1: cmd 0x9111=SetNewStation timed out
[ 1268.459731] ieee80211 phy1: return code: 0x1111
[ 1268.464357] ieee80211 phy1: timeout: 0x1111
[ 1268.468577] ieee80211 phy1: failed execution
[ 1272.485886] ieee80211 phy1: cmd 0x9111=SetNewStation timed out
[ 1272.491804] ieee80211 phy1: return code: 0x1111
[ 1272.496374] ieee80211 phy1: timeout: 0x1111
[ 1272.500593] ieee80211 phy1: failed execution
[ 1276.905933] ieee80211 phy1: cmd 0x9111=SetNewStation timed out
[ 1276.911864] ieee80211 phy1: return code: 0x1111
[ 1276.916438] ieee80211 phy1: timeout: 0x1111
[ 1276.920649] ieee80211 phy1: failed execution
[ 1281.326756] ieee80211 phy1: cmd 0x9111=SetNewStation timed out
[ 1281.332685] ieee80211 phy1: return code: 0x1111
[ 1281.337259] ieee80211 phy1: timeout: 0x1111
[ 1281.341473] ieee80211 phy1: failed execution
[ 1285.750479] ieee80211 phy1: cmd 0x8050=broadcast_ssid_enable timed out
[ 1285.757113] ieee80211 phy1: return code: 0x0050
[ 1285.761745] ieee80211 phy1: timeout: 0x0050
[ 1285.765963] ieee80211 phy1: failed execution
[ 1290.174537] ieee80211 phy1: cmd 0x8127=SetInformationElements timed out
[ 1290.181197] ieee80211 phy1: return code: 0x0127
[ 1290.185842] ieee80211 phy1: timeout: 0x0127
[ 1290.190062] ieee80211 phy1: failed execution
[ 1294.503625] ieee80211 phy1: cmd 0x801c=80211RadioControl timed out
[ 1294.509852] ieee80211 phy1: return code: 0x001c
[ 1294.514468] ieee80211 phy1: timeout: 0x001c
[ 1294.518688] ieee80211 phy1: failed execution
[ 1298.539747] ieee80211 phy1: cmd 0x8126=SetFixedRate timed out
[ 1298.545579] ieee80211 phy1: return code: 0x0126
[ 1298.550150] ieee80211 phy1: timeout: 0x0126
[ 1298.554418] ieee80211 phy1: failed execution

Chadster766, 2016-08-19:
I just realized my Apple TV is hard wired. I will connect it via 5Ghz to try and confirm this issue.

Chadster766, 2016-08-20:
We streamed video all day today on the Apple TV from McDebian with mwlwifi version 10.3.0.17-20160531-1. 

None of the ieee80211 errors you posted appear in the journal.

How long does it take for this error to appear?

chinawrj, 2016-08-20:
@Chadster766 This issue happened on my side two months ago. Since I disabled Wi-Fi during the last two month, I haven't the chance to reproduce it. BTW, it seems the firmware in 88W8864 is dead at this time, but the sniffer show that the Beacon frame from 88W8864 is normal.

jklap, 2016-08-20:
There seems to be something else related here. After the ~week of no issues with the new firmware I had power cycled the router (not just a reboot but no power for ~10 minutes) and then that afternoon (~8 hours later) within 5 minutes of Apple TV streaming (first time since the power cycle), the failure occurred. A reboot and within 5 minute it occurred again. Reboot and the same (note, "5 minutes" is a rough approximation- I don't think this is happened with a set window of time after a reboot).

Now since that occurred, no failures-- yet still lots of Apple TV streaming.

My recollection is this is similar behavior to what I'd seen in the past... that things can be fine (for some random period of time, sometimes days) and then all of a sudden, it fails and will fail several times in a row soon after a reboot.

I think I've always seen the failure only during streaming, but maybe something else needs to set the stage for streaming to cause the failure? Just making sh*t up here, but maybe something else needs to have failed out earlier, but don't cause any noticeable issues until the firmware is under pressure from the high traffic/bandwidth?

I'm going to start making notes of what devices are being used when failures happen with the idea that maybe it's another device waking up or going to sleep that is causing it?

Chadster766, 2016-08-20:
@jklap do you have different firmwares in the primary and alternate image locations? If you do what could be happening is during the power cycle it's switching between firmwares.

U-boot will switch between the primary and alternate firmware if it detects a firmware problem during boot. You can trigger this manually by switching the router on and off 3 times is a row. The trick is don't switch off each time until the LAN LED starts blinking.

jklap, 2016-08-21:
Alt image is disabled. So is firmware switching. And l've also verified loaded firmware via dmesg.

Chadster766, 2016-08-21:
Can you test with McDebian? This would isolate driver code for testing.

jklap, 2016-08-21:
No.

kevle, 2016-08-22:
I'm running into this issue as well.

> 
> Aug 22 16:33:55 lede kernel: [  398.628849] ieee80211 phy0: cmd 0x9125=BAStream timed out
> Aug 22 16:33:55 lede kernel: [  398.634309] ieee80211 phy0: return code: 0x1125
> Aug 22 16:33:55 lede kernel: [  398.638874] ieee80211 phy0: timeout: 0x1125
> Aug 22 16:33:55 lede kernel: [  398.643085] ieee80211 phy0: destroy ba failed execution
> 

5 Ghz seems to be dead afterwards

Driver info:

> root@lede:~# cat /sys/kernel/debug/ieee80211/phy0/mwlwifi/info
> driver name: mwlwifi
> chip type: 88W8864
> hw version: 7
> driver version: 10.3.0.18-20160804
> firmware version: 0x0702091a
> power table loaded from dts: yes
> firmware region code: 0x0
> mac address: 00:00:00:00:00:00 ( redacted )
> 2g: disable
> 5g: enable
> antenna: 2 2
> irq number: 104
> iobase0: e0e00000
> iobase1: e0f80000
> tx limit: 768
> rx limit: 64
> ap macid support: 0000ffff
> sta macid support: 00010000
> macid used: 00000001
> qe trigger number: 6732

yuhhaurlin, 2016-08-22:
We did a lot of tests here, we did not see firmware gone in 5 minutes. Especially it can work for the first week. Can you give me exact procedure to reproduce it?

yuhhaurlin, 2016-08-22:
To be honest, we did not encounter this kind of problem!

yuhhaurlin, 2016-08-22:
@kevle Why the MAC address is all zero?

kevle, 2016-08-22:
@yuhhaurlin : I zeroed the text out manually so that wardrivers and the like can't get an advantage of knowing my driver version. I guess that was maybe too paranoid, here it is: 00:25:9c:14:09:b4

Here are my ( unverified ) findings: As long as 2 devices are connected it's fine, even when running speed tests simultaneously. As soon as only 1 device is left phy0 dies when running the speedtest. I will try this method some more times to pinpoint the actual cause. 

> Aug 22 18:23:10 lede hostapd: wlan0: STA 74:81:14:xx:xx:xx IEEE 802.11: authenticated
> Aug 22 18:23:11 lede hostapd: wlan0: STA 74:81:14:xx:xx:xx IEEE 802.11: associated (aid 2)
> Aug 22 18:23:11 lede hostapd: wlan0: STA 74:81:14:xx:xx:xx WPA: pairwise key handshake completed (RSN)
> Aug 22 18:23:11 lede hostapd: wlan0: STA 74:81:14:xx:xx:xx RADIUS: starting accounting session SESSION_ID_THINGY
> Aug 22 18:23:11 lede hostapd: wlan0: STA 74:81:14:xx:xx:xx IEEE 802.1X: authenticated - EAP type: 25 (PEAP)
> Aug 22 18:25:59 lede kernel: [  302.989615] ieee80211 phy0: cmd 0x9125=BAStream timed out
> Aug 22 18:25:59 lede kernel: [  302.995084] ieee80211 phy0: return code: 0x1125
> Aug 22 18:25:59 lede kernel: [  302.999672] ieee80211 phy0: timeout: 0x1125
> Aug 22 18:25:59 lede kernel: [  303.003919] ieee80211 phy0: destroy ba failed execution
>

kevle, 2016-08-22:
This time it happened with my 2 devices. My second device had issues connecting. Not sure how to reproduce it exactly aside from the fact that it seems to happen reliably in my setup.

2.4 Ghz device still works fine afterwards. Interestingly enough phy0 corresponds to the 5 Ghz device in my case. Probably because of the way my /etc/config/wireless is setup

> Aug 22 18:40:08 lede hostapd: wlan0: STA c0:ee:fb:xx:xx:xx IEEE 802.11: disassociated
> Aug 22 18:40:09 lede hostapd: wlan0: STA c0:ee:fb:xx:xx:xx IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
> Aug 22 18:40:22 lede hostapd: wlan0: STA c0:ee:fb:xx:xx:xx IEEE 802.11: authenticated
> Aug 22 18:40:22 lede hostapd: wlan0: STA c0:ee:fb:xx:xx:xx IEEE 802.11: associated (aid 2)
> Aug 22 18:40:37 lede hostapd: wlan0: STA c0:ee:fb:xx:xx:xx IEEE 802.11: disassociated
> Aug 22 18:40:38 lede hostapd: wlan0: STA c0:ee:fb:xx:xx:xx IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
> Aug 22 18:40:47 lede hostapd: wlan1: STA c0:ee:fb:xx:xx:xx IEEE 802.11: authenticated
> Aug 22 18:40:47 lede hostapd: wlan1: STA c0:ee:fb:xx:xx:xx IEEE 802.11: associated (aid 1)
> Aug 22 18:40:47 lede hostapd: wlan0: STA c0:ee:fb:xx:xx:xx IEEE 802.11: authenticated
> Aug 22 18:40:47 lede hostapd: wlan0: STA c0:ee:fb:xx:xx:xx IEEE 802.11: associated (aid 2)
> Aug 22 18:40:47 lede hostapd: wlan0: STA c0:ee:fb:xx:xx:xx WPA: pairwise key handshake completed (RSN)
> Aug 22 18:40:47 lede hostapd: wlan0: STA c0:ee:fb:xx:xx:xx RADIUS: starting accounting session SESSION_ID_THINGY
> Aug 22 18:40:47 lede hostapd: wlan0: STA c0:ee:fb:xx:xx:xx IEEE 802.1X: authenticated - EAP type: 25 (PEAP)
> Aug 22 18:41:21 lede kernel: [  452.708203] ieee80211 phy0: cmd 0x9125=BAStream timed out
> Aug 22 18:41:21 lede kernel: [  452.713661] ieee80211 phy0: return code: 0x1125
> Aug 22 18:41:21 lede kernel: [  452.718233] ieee80211 phy0: timeout: 0x1125
> Aug 22 18:41:21 lede kernel: [  452.722449] ieee80211 phy0: destroy ba failed execution
> Aug 22 18:41:25 lede kernel: [  456.747587] ieee80211 phy0: cmd 0x9125=BAStream timed out
> Aug 22 18:41:25 lede kernel: [  456.753022] ieee80211 phy0: return code: 0x1125
> Aug 22 18:41:25 lede kernel: [  456.757572] ieee80211 phy0: timeout: 0x1125
> Aug 22 18:41:25 lede kernel: [  456.761779] ieee80211 phy0: destroy ba failed execution
> 

After quite some time those errors appeared:

> Aug 22 18:44:39 lede kernel: [  651.482500] ieee80211 phy0: cmd 0x9122=UpdateEncryption timed out
> Aug 22 18:44:39 lede kernel: [  651.488633] ieee80211 phy0: return code: 0x1122
> Aug 22 18:44:39 lede kernel: [  651.493182] ieee80211 phy0: timeout: 0x1122
> Aug 22 18:44:39 lede kernel: [  651.497406] ieee80211 phy0: failed execution
> Aug 22 18:44:39 lede kernel: [  651.501696] wlan0: failed to remove key (0, c0:ee:fb:xx:xx:xx) from hardware (-5)
> Aug 22 18:44:43 lede kernel: [  655.510462] ieee80211 phy0: cmd 0x9122=UpdateEncryption timed out
> Aug 22 18:44:43 lede kernel: [  655.516619] ieee80211 phy0: return code: 0x1122
> Aug 22 18:44:43 lede kernel: [  655.521169] ieee80211 phy0: timeout: 0x1122
> Aug 22 18:44:43 lede kernel: [  655.525369] ieee80211 phy0: failed execution
> Aug 22 18:44:43 lede kernel: [  655.529669] wlan0: failed to set key (2, ff:ff:ff:ff:ff:ff) to hardware (-5)
> Aug 22 18:44:47 lede kernel: [  659.538421] ieee80211 phy0: cmd 0x9122=UpdateEncryption timed out
> Aug 22 18:44:47 lede kernel: [  659.544583] ieee80211 phy0: return code: 0x1122
> Aug 22 18:44:47 lede kernel: [  659.549160] ieee80211 phy0: timeout: 0x1122
> Aug 22 18:44:47 lede kernel: [  659.553359] ieee80211 phy0: failed execution
> Aug 22 18:44:47 lede kernel: [  659.557661] wlan0: failed to remove key (0, 74:81:14:xx:xx:xx) from hardware (-5)
> Aug 22 18:44:52 lede kernel: [  663.566386] ieee80211 phy0: cmd 0x9111=SetNewStation timed out
> Aug 22 18:44:52 lede kernel: [  663.572248] ieee80211 phy0: return code: 0x1111
> Aug 22 18:44:52 lede kernel: [  663.576810] ieee80211 phy0: timeout: 0x1111
> Aug 22 18:44:52 lede kernel: [  663.581011] ieee80211 phy0: failed execution
> Aug 22 18:44:52 lede hostapd: wlan0: STA c0:ee:fb:xx:xx:xx IEEE 802.11: deauthenticated due to local deauth request
> Aug 22 18:44:56 lede kernel: [  667.586346] ieee80211 phy0: cmd 0x9111=SetNewStation timed out
> Aug 22 18:44:56 lede kernel: [  667.592222] ieee80211 phy0: return code: 0x1111
> Aug 22 18:44:56 lede kernel: [  667.596826] ieee80211 phy0: timeout: 0x1111
> Aug 22 18:44:56 lede kernel: [  667.601042] ieee80211 phy0: failed execution
> Aug 22 18:44:56 lede hostapd: wlan0: STA 74:81:14:xx:xx:xx IEEE 802.11: deauthenticated due to local deauth request
> Aug 22 18:45:00 lede kernel: [  671.606316] ieee80211 phy0: cmd 0x8050=broadcast_ssid_enable timed out
> Aug 22 18:45:00 lede kernel: [  671.612877] ieee80211 phy0: return code: 0x0050
> Aug 22 18:45:00 lede kernel: [  671.617442] ieee80211 phy0: timeout: 0x0050
> Aug 22 18:45:00 lede kernel: [  671.621642] ieee80211 phy0: failed execution
> Aug 22 18:45:04 lede kernel: [  675.626273] ieee80211 phy0: cmd 0x8127=SetInformationElements timed out
> Aug 22 18:45:04 lede kernel: [  675.632920] ieee80211 phy0: return code: 0x0127
> Aug 22 18:45:04 lede kernel: [  675.637500] ieee80211 phy0: timeout: 0x0127
> Aug 22 18:45:04 lede kernel: [  675.641700] ieee80211 phy0: failed execution
> Aug 22 18:45:08 lede kernel: [  679.646238] ieee80211 phy0: cmd 0x801c=80211RadioControl timed out
[... cut ...] 0x001c
> Aug 22 18:45:08 lede kernel: [  679.657011] ieee80211 phy0: timeout: 0x001c
> Aug 22 18:45:08 lede kernel: [  679.661232] ieee80211 phy0: failed execution
> Aug 22 18:45:12 lede kernel: [  683.666200] ieee80211 phy0: cmd 0x8126=SetFixedRate timed out
> Aug 22 18:45:12 lede kernel: [  683.671976] ieee80211 phy0: return code: 0x0126
> Aug 22 18:45:12 lede kernel: [  683.676537] ieee80211 phy0: timeout: 0x0126
> Aug 22 18:45:12 lede kernel: [  683.680738] ieee80211 phy0: failed execution
>

kevle, 2016-08-22:
The issue also exists on a WRT1900 ACS device.

root@ap-wrt1900acs:~# cat /sys/kernel/debug/ieee80211/phy0/mwlwifi/info

> driver name: mwlwifi
> chip type: 88W8864
> hw version: 7
> driver version: 10.3.0.18-20160804
> firmware version: 0x0702091a
> power table loaded from dts: yes
> firmware region code: 0x0
> mac address: 00:25:9c:13:ec:bb
> 2g: disable
> 5g: enable
> antenna: 4 4
> irq number: 104
> iobase0: e0e00000
> iobase1: e0f80000
> tx limit: 768
> rx limit: 64
> ap macid support: 0000ffff
> sta macid support: 00010000
> macid used: 00000001
> qe trigger number: 132722
> 

>
> Aug 22 20:11:00 lede hostapd: wlan0: STA 74:81:14:xx:xx:xx IEEE 802.11: authenticated
> Aug 22 20:11:00 lede hostapd: wlan0: STA 74:81:14:xx:xx:xx IEEE 802.11: associated (aid 2)
> Aug 22 20:11:00 lede hostapd: wlan0: STA 74:81:14:xx:xx:xx WPA: pairwise key handshake completed (RSN)
> Aug 22 20:11:00 lede hostapd: wlan0: STA 74:81:14:xx:xx:xx RADIUS: starting accounting session SESSION_ID
> Aug 22 20:11:00 lede hostapd: wlan0: STA 74:81:14:xx:xx:xx IEEE 802.1X: authenticated - EAP type: 25 (PEAP)
> Aug 22 20:11:08 lede netifd: lan (1220): Sending renew...
> Aug 22 20:11:08 lede netifd: lan (1220): Lease of 192.168.1.120 obtained, lease time 600
> Aug 22 20:12:11 lede kernel: [ 2673.542733] ieee80211 phy0: cmd 0x9125=BAStream timed out
> Aug 22 20:12:11 lede kernel: [ 2673.548185] ieee80211 phy0: return code: 0x1125
> Aug 22 20:12:11 lede kernel: [ 2673.552736] ieee80211 phy0: timeout: 0x1125
> Aug 22 20:12:11 lede kernel: [ 2673.556951] ieee80211 phy0: destroy ba failed execution
>

yuhhaurlin, 2016-08-22:
How to reproduce? It happens quickly!

kevle, 2016-08-22:
I'm just running speedtests with the speedtest.net app until the crash happens. I mostly get transfer speeds between 200-300 Mbit/s for both download and upload. The error has only surfaced while measuring the download speed so far.

This is my configuration for lede:

https://gist.github.com/kevle/1c4f60b5bec59f3725007084fa3d6871

Here is my uci wireless configuration:

https://gist.github.com/kevle/e94eacc32aac8ff40bb927939d535ecb

yuhhaurlin, 2016-08-22:
We had done tests before releasing this version of code. We did not find this kind of problem. Can you use other release to check if you still encounter this kind of problem?

yuhhaurlin, 2016-08-22:
Or you can use other pre-built image to do test to see if you will encounter same problem.

jklap, 2016-08-23:
What ever causes the below, and whatever this failure does is what brings down all of the other clients:

[189972.199658] ieee80211 phy1: cmd 0x9125=BAStream timed out
[189972.205193] ieee80211 phy1: return code: 0x1125
[189972.209905] ieee80211 phy1: timeout: 0x1125
[189972.214212] ieee80211 phy1: destroy ba failed execution

If we can not find the root cause, is there some type of additional error handling that can be added around this?

Note too, the last three times this has happened, the first MAC reported varies.

It also seems pretty consistent there is a single MAC after the remove key ff:ff:ff.. set key ff:ff:ff... (and no, that MAC there also seems to vary).

In all three cases that I've kept notes for, there is only one device that is not showing up with the others.. an iPhone 6S. It *is* possible that is just a coincidence but thought I would note it here.

I've tried multiple images previously with the same results-- I wouldn't be able to try another image for a while... too much going on.

I can pretty much reproduce this on demand simply by turning on the Apple TV and streaming via Netflix.

yuhhaurlin, 2016-08-23:
Does anyone encounter this kind of problem (driver 10.3.0.17-20160804, firmware 7.2.9.26)?

kevle, 2016-08-23:
I've tested the older version for about an hour now and had no issues whatsoever after 20Gbyte of transfered data.

> root@OpenWrt ~# cat /sys/kernel/debug/ieee80211/phy0/mwlwifi/info
> driver name: mwlwifi
> chip type: 88W8864
> hw version: 7
> driver version: 10.3.0.17-20160324
> firmware version: 0x07020902
> mac address: 00:25:9c:13:ec:bb
> 2g: disable
> 5g: enable
> antenna: 4 4
> irq number: 103
> iobase0: e1200000
> iobase1: e1380000
> tx limit: 768
> rx limit: 64
> ap macid support: 0000ffff
> sta macid support: 00010000
> macid used: 00000001
> qe trigger number: 161378
> mfg mode: false

yuhhaurlin, 2016-08-23:
Only changing driver and firmware? The other components are the same? BTW, can you give me clear steps to reproduce this problem?

yuhhaurlin, 2016-08-24:
You can get pre-built images from https://downloads.openwrt.org/people/kaloz/. Please use this image to reproduce this problem on your device.

Once if problem is reproduced, please:
1. Specify the device model you used.
2. Cat /sys/kernel/debug/ieee80211/phy0/mwlwifi/info to reconfirm the version of tested driver and firmware.
3. Setting of AP.
4. Setting of test environment: what client device you use and what application you are running. How soon and how frequency to reproduce this problem? Test environment should be as simple as possible.

Thanks.

kevle, 2016-08-24:
No issues so far after 100GiB with the image from https://downloads.openwrt.org/people/kaloz/

It uses kernel version 3.18.36, maybe that is the catch? I will try out a selfbuilt image with that kernel.

yuhhaurlin, 2016-08-24:
Thanks. I will close this one.

closed 2016-08-24 by yuhhaurlin

jklap, 2016-08-25:
Hmmm. Given that this has been reported by several users, not just myself and has the exact same behavior on #93 with additional users reporting the issue I don't think this should be closed. Please re-read my last post. There is clearly something triggering it and just because one user isn't seeing the issue after transferring a large file doesn't mean the issue doesn't exist. It simply means they didn't trigger it.

yuhhaurlin, 2016-08-26:
Can you use the pre-built images to do test?

chinawrj, 2016-08-26:
@yuhhaurlin I think you can provide the image and let other guys do the test

yuhhaurlin, 2016-08-26:
You can get pre-built images from https://downloads.openwrt.org/people/kaloz/. Please use this image to reproduce this problem on your device.

Once if problem is reproduced, please:
1. Specify the device model you used.
2. Cat /sys/kernel/debug/ieee80211/phy0/mwlwifi/info to reconfirm the version of tested driver and firmware.
3. Setting of AP.
4. Setting of test environment: what client device you use and what application you are running. How soon and how often to reproduce this problem? Test environment should be as simple as possible.

Thanks.

jklap, 2016-09-06:
I can not use other images for testing-- this is a production device now given the major strides the wifi driver has made-- currently this is the only instability we have.

And like I mentioned, there certainly seems to be some type of trigger condition that enables this to happen with streaming the piece that pushes it over the edge. We've been up 14 days now with lots of streaming, no issues. Yet I've also not been using several computers and notebooks much in the last 14 days-- seems suspect.

Although in my previous note I noted that an iPhone 6S seems to be somehow related-- possible it still is but I've been using it heavily over the last 14 days. So unless it's the 6S in combination with another device that hasn't been used, that might be a red herring.

mystica555, 2016-09-09:
At one point in time I think someone had mentioned setting certain kernel buffer parameters to a larger size... I may take a bit more time to search for this when I'm not busy today, but if there are perhaps some very stupidly basic things that are changing in the kernel .config each build then this might just be the 'straw that breaks the mwlwifi's back' ?