kaloz/mwlwifi #106: Unable to authenticate to a WPA2 Enterprise BSS when also having a WPA2 PSK BSS on the same interface

ignisf, 2016-09-05:
Hello,

I'm having issues authenticating to a WPA2 Enterprise BSS when also having a second WPA2-PSK BSS on the same interface.

Running LEDE HEAD, r1504+2 on a Linksys WRT1900ACS, `/etc/config/wireless` [here](https://gist.github.com/ignisf/333254a9ef85331ae483c14c4e915dd8), syslog output [here](https://gist.github.com/ignisf/be119958373cf7f27b915d476b15ab02).

I also tried [enabling 802.11W support](https://gist.github.com/ignisf/4df80c71c4796eb5a62bca8b15332d34) to test if it had anything to do with this. Syslog output from the attempt: [here](https://gist.github.com/ignisf/36f685727bda6a5f6172a96ca20a3cb9).

thagabe, 2016-09-06:
https://github.com/lede-project/source/commit/c5913264e7eb9f2bee782b41ab688d002f00d196 
rebuild this commit should fix

ignisf, 2016-09-06:
Another interesting aspect -- whenever roaming from another .11i BSS on another access point (a TP-Link Archer C5 or a WDR4300), clients get associated and authenticated. All the APs have a similar configuration to the one I linked above.

@thagabe, yes, the LEDE people merged my PR.

yuhhaurlin, 2016-09-07:
@thagade, ignisf Yes, ieee80211w is supported by mwlwifi and it needs to set this configuration to let it work.

ignisf, 2017-01-26:
I can no longer replicate this issue with 10.3.2.0-20161222

closed 2017-01-26 by ignisf