kaloz/mwlwifi #107: In monitor mode, no data packet captured

yydcool, 2016-09-22:
I'm using a WRT1200AC, driver version 10.3.0.18-20160823-1.
When I turn interface to monitor mode, I can only capture beacon packets and some multicast packets, but not any data packets.

I've tried 'tcpdump' and 'iwcap' commands, and tried on 2.4GHz and 5GHz, both of them don't output any data packets.

Is the driver filtering those data packets on monitor mode?

yuhhaurlin, 2016-10-14:
Monitor interface will be checked with lower priority.

yuhhaurlin, 2017-01-11:
If issue #60 is fixed, you can try to see if this issue is still there. Thanks.

yydcool, 2017-02-15:
I will try with this week

yydcool, 2017-02-16:
Still no data packets in monitor mode.
Here is how I do the capture:
I'm running (Lede leviathan IV SNAPSHOT r3293, kernel 4.4.46) on Linksys wrt1200ac.
mwlwifi driver is 10.3.2.0-20170110.
I turn on the monitor on 2.4Ghz through LuCI gui: Network -> wireless -> Marvell 88W8864 802.11bgn (radio1) -> add -> Interface Configuration -> Mode: Monitor

Here is the info after I turn on the monitor:
root@lede:~# iwinfo
wlan1     ESSID: unknown
          Access Point: 00:00:00:00:00:00
          Mode: Monitor  Channel: 1 (2.412 GHz)
          Tx-Power: 30 dBm  Link Quality: unknown/70
          Signal: unknown  Noise: -72 dBm
          Bit Rate: unknown
          Encryption: unknown
          Type: nl80211  HW Mode(s): 802.11bgn
          Hardware: 11AB:2A55 11AB:0000 [Marvell 88W8864]
          TX power offset: none
          Frequency offset: none
          Supports VAPs: yes  PHY name: phy1

I capture with:
root@lede:~# **tcpdump -i wlan1 -v**
tcpdump: listening on wlan1, link-type IEEE802_11_RADIO (802.11 plus radiotap header), capture size 262144 bytes
04:33:14.190812 36.0 Mb/s 2412 MHz 11g -40dBm signal antenna 0 Beacon (XXXXX) [1.0* 2.0* 5.5* 11.0* 18.0 24.0 36.0 54.0 Mbit] ESS CH: 1, PRIVACY
04:33:14.197995 36.0 Mb/s 2412 MHz 11g -73dBm signal antenna 0 Beacon (XXXXX) [1.0* 2.0* 5.5* 11.0* 18.0 24.0 36.0 54.0 Mbit] ESS CH: 1, PRIVACY
04:33:14.202164 36.0 Mb/s 2412 MHz 11g -58dBm signal antenna 0 Beacon (XXXXX) [1.0* 2.0* 5.5* 11.0* 6.0* 9.0 12.0* 18.0 Mbit] ESS CH: 3, PRIVACY
04:33:14.222475 36.0 Mb/s 2412 MHz 11g -80dBm signal antenna 0 Beacon (XXXXX) [1.0* 2.0* 5.5* 11.0* 6.0 9.0 12.0 18.0 Mbit] ESS CH: 2, PRIVACY
04:33:14.227190 36.0 Mb/s 2412 MHz 11g -69dBm signal antenna 0 Beacon (XXXXX) [1.0* 2.0* 5.5* 11.0* 6.0 9.0 12.0 18.0 Mbit] ESS CH: 1, PRIVACY
04:33:14.237955 36.0 Mb/s 2412 MHz 11g -76dBm signal antenna 0 Beacon (XXXXX) [1.0* 2.0* 5.5* 11.0* 6.0 9.0 12.0 18.0 Mbit] ESS CH: 1, PRIVACY
04:33:14.239212 36.0 Mb/s 2412 MHz 11g -74dBm signal antenna 0 Beacon (XXXXX) [12.0* 18.0 24.0* 36.0 48.0 54.0 Mbit] ESS CH: 1, PRIVACY
04:33:14.247941 36.0 Mb/s 2412 MHz 11g -79dBm signal antenna 0 Beacon (XXXXX) [12.0* 18.0 24.0* 36.0 48.0 54.0 Mbit] ESS CH: 1, PRIVACY
where XXXXX is the SSID of nearby APs.
04:33:14.305001 36.0 Mb/s 2412 MHz 11g -80dBm signal antenna 0 Probe Request (unconfigured) [1.0 2.0 5.5 6.0 9.0 11.0 12.0 18.0 Mbit]

where XXXXX are some of the SSID of nearby APs. I can only see Beacon packets

mrvltest, 2017-02-16:
Issue #60 has not yet been fixed now. I will put it to lower priority.

yuhhaurlin, 2018-03-05:
Fixed by https://github.com/kaloz/mwlwifi/commit/4591708bfd92b99102976b7ae2571101cbf8518e.

closed 2018-03-05 by yuhhaurlin

bbogey, 2018-06-07:
Monitor mode still gathers only multicast packets. 

wifi down
iw phy phy0 interface add mon0 type monitor
ifconfig mon0 up
iw dev mon0 set channel 36 80MHz
tcpdump -i mon0 -n -s0 -w /mnt/sda1/mon0_wrt_test.pcap

I can only see packets sent multicast

Or am I missing some kernel option?