kaloz/mwlwifi #110: 88W8964 disable_2ghz device tree attribute will make the interface unuseable

BrainSlayer, 2016-10-13:
the WRT3200ACM has 2 interfaces. first is 5 ghz only, second is 2.4 ghz only.
if i manually disable the 2.4 ghz band (which is required) the 5 ghz interface will not work anymore. band disable of 5 ghz on the 2.4 ghz interface will work. 
only solution right now is to keep 2.4 ghz band enabled, even if the card is not able to operate on this band. but iw utility etc. will return bad channels for sure

yuhhaurlin, 2016-10-14:
DTS should provide band information for mwlwifi.

BrainSlayer, 2016-10-14:
thats clear. for the 5 ghz card disable_2g must be set on the dts. but this will get the card out of operation and i just get firmware errors. so i have to keep it enabled. otherwise it will not work

yuhhaurlin, 2016-10-14:
Can you try pre-built images from https://downloads.openwrt.org/people/kaloz/?

BrainSlayer, 2016-10-14:
okay. will try it

BrainSlayer, 2016-10-14:
tested it. firmware does not support the 88w8964 wifi chipset. firmware is not included
  9.496170] ieee80211 phy0: mwlwifi: cannot find firmware image <mwlwifi/88W8964.bin>
[    9.504038] ieee80211 phy0: firmware init fail
[    9.508498] ieee80211 phy0: mwlwifi: fail to initialize firmware

yuhhaurlin, 2016-10-14:
@kaloz Did you include 88W8964.bin to your test image?

BrainSlayer, 2016-10-14:
i used the test images provided by kaloz. but just from the dts he used i know it will not work fully and secondary he did not include the mwlwifi firmware images in his firmware for the wrt3200acm

yuhhaurlin, 2016-10-14:
I post here and I will forward your message to him.

kaloz, 2016-10-15:
the firmware file install has been fixed, but I see no issues with the dts, both interfaces seem to work fine for me

BrainSlayer, 2016-10-16:
its just my experience. setting disable_2g on phy0 will cause a unusable interface. you just see firmware errors while initialisation and use of this interface. btw. check my dts, for sdio support of the bluetooth chipset and power led support. yours is broken in many cases

yuhhaurlin, 2016-10-17:
I did not encounter this kind of problem when I build image for our internal DB board working with 88W8964.

yuhhaurlin, 2016-10-17:
Did you encounter the same problem if running Kaloz;s image?

BrainSlayer, 2016-10-17:
yes i tested it. but wont do it again. i use the same wireless /mac80211 combination and the same hostapd. you can assume that i know what i'm doing over the last 11 years of development on DD-WRT. kaloz image still has alot of wrong dts properties and is not fully compatible to the rango platform. he added the wireless firmware in between. but same issue as soon as i start the wireless interface. only work around is to ignore the disable_2g directive. .

yuhhaurlin, 2016-10-17:
So you still encounter the same problem with Kaloz's image? I need to check how to build image for WRT3200AC (Rango) first.

yuhhaurlin, 2016-11-01:
Please check LEDE image of https://drive.google.com/open?id=0B3qLWtcWB9EdRHJoSGdUa2VBMFk. I did not find this problem.

yuhhaurlin, 2016-11-03:
If no comment for this issue, I will close it.

closed 2016-11-03 by yuhhaurlin