kaloz/mwlwifi #111: 88W8964 VHT 160 not operational in client mode

BrainSlayer, 2016-10-13:
in client mode VHT160 doesnt work. the network scan works, but the device does not associate in any way. (tested with a QCA9984 as AP in vht160 mode on channel 108)

yuhhaurlin, 2016-10-14:
Can you check your region code? For FCC, only one channel segment for 160 Mhz (started from channel 36).

yuhhaurlin, 2016-10-14:
BTW, we find mac80211 module of OpenWrt is not updated for 160 Mhz. We will update patch for that later.

BrainSlayer, 2016-10-14:
its not a mac80211 issue. first we use the very latest mac80211 from 08102016. (as openwrt is doing in trunk right now). the client mode works fully with the same mac80211 with a qca9984 in vht160. and yes. FCC/US is used a regdomain data. its stored by default on the wifi hw eeprom on this device which was provided by belkin. channel 36 is not supported on channel36 by mac80211 / crda frequency set. but channel 100 upwards should work. as i said i was able to scan the access point which itself us running fine if i use a second qca9984 as client in vht160. (netgeart r7800 with dd-wrt and the same mac80211 vwas used for testing) but the 88w9864 is just able to scan the network, but does not associate

BrainSlayer, 2016-10-14:
in addition VHT80 works in client mode

yuhhaurlin, 2016-10-14:
We just test 88W8964 AP + 88W8964 STA for 160 Mhz and found mac80211 module of OpenWrt is not updated to support 160 Mhz.
If FCC is set, you can only use 36~64 as 160 MHz.
However, we will do more tests and update result later.

BrainSlayer, 2016-10-14:
then you may use a old version of openwrt. but i do not use openwrt. i develop dd-wrt which does support vht160 already. so far only tested on qca9984. 

here is a  scan entry of the 88w8964 trying to associate with a qca9984. i cannot test 88w8964 to each other since i just have one sample device. but interoperability should count as well. (the wpa supplicant just shows timeout while trying to associate)
but here is the scan output. you may check if the flags are correct or if you find something unusual

BSS dc:ef:09:e4:ce:09(on mvebu0)
        TSF: 910904579 usec (0d, 00:15:10)
        freq: 5500
        beacon interval: 100 TUs
        capability: ESS (0x0011)
        signal: -64.00 dBm
        last seen: 7670 ms ago
        Information elements from Probe Response frame:
        SSID: RHBRIDGE5
        RSN:     * Version: 1
                 * Group cipher: CCMP
                 * Pairwise ciphers: CCMP
                 * Authentication suites: PSK
                 * Capabilities: 16-PTKSA-RC 1-GTKSA-RC (0x000c)
        HT capabilities:
                Capabilities: 0x9ef
                        RX LDPC
                        HT20/HT40
                        SM Power Save disabled
                        RX HT20 SGI
                        RX HT40 SGI
                        TX STBC
                        RX STBC 1-stream
                        Max AMSDU length: 7935 bytes
                        No DSSS/CCK HT40
                Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
                Minimum RX AMPDU time spacing: 8 usec (0x06)
                HT TX/RX MCS rate indexes supported: 0-31
        HT operation:
                 * primary channel: 100
                 * secondary channel offset: above
                 * STA channel width: any
        VHT capabilities:
                VHT Capabilities (0x338000f6):
                        Max MPDU length: 11454
                        Supported Channel Width: 160 MHz
                        RX LDPC
                        short GI (80 MHz)
                        short GI (160/80+80 MHz)
                        TX STBC
                        RX antenna pattern consistency
                        TX antenna pattern consistency
                VHT RX MCS set:
                        1 streams: MCS 0-9
                        2 streams: MCS 0-9
                        3 streams: MCS 0-9
                        4 streams: MCS 0-9
                        5 streams: not supported
                        6 streams: not supported
                        7 streams: not supported
                        8 streams: not supported
                VHT RX highest supported: 0 Mbps
                VHT TX MCS set:
                        1 streams: MCS 0-9
                        2 streams: MCS 0-9
                        3 streams: MCS 0-9
                        4 streams: MCS 0-9
                        5 streams: not supported
                        6 streams: not supported
                        7 streams: not supported
                        8 streams: not supported
                VHT TX highest supported: 0 Mbps
        VHT operation:
                 * channel width: 1 (80 MHz)
                 * center freq segment 1: 106
                 * center freq segment 2: 114
                 * VHT basic MCS set: 0xfffc

yuhhaurlin, 2016-10-14:
Thanks for your information. We will check it.

BrainSlayer, 2016-10-14:
the scan output is identical with a qca9984 client which is  working here in that test case. hard to say what the issue is. consider that i developed the vht160 support for ath10k. so if you find a issue there, which i havent seen yet, i'm likelly able to fix it. for testing you may use lede-staging tree from nbd since openwrt is not maintained anymore and dead (all major openwrt maintainers left and founded lede). lede is the following up project based on the latest openwrt source, but way more ahead now git://git.lede-project.org/lede/nbd/staging.git

yuhhaurlin, 2016-10-14:
Thanks for your information.

yuhhaurlin, 2016-10-14:
BTW, does the stock firmware can connect to AP under this test case?

BrainSlayer, 2016-10-14:
the stock image does not support client mode

BrainSlayer, 2016-10-14:
and the stock firmware i have is not the final public firmware. consider that the device is not yet released to the market (expected next week). its a early sample provided by belkin some months ago

yuhhaurlin, 2016-10-14:
I checked staging of LEDE. Mac80211 module is updated to 2016-10-08 and hostapd package is updated to 2016-09-05. However, I can't find profile related to WRT3200ACS (Rango). I think I will base on LEDE to create image to verify mwlwifi (Built for our internal board). BTW, do you know when the profile for WRT3200ACS will be ready for LEDE?

Noltari, 2016-10-15:
@yuhhaurlin please check this PR -> https://github.com/lede-project/source/pull/413
It's also on my staging tree (branch mvebu-next): https://git.lede-project.org/?p=lede/noltari/staging.git;a=shortlog;h=refs/heads/mvebu-next

BrainSlayer, 2016-10-15:
consider that the dts from kaloz is buggy and not fully supporting rango. so check the dts files from dd-wrt from complete support http://svn.dd-wrt.com/browser/src/linux/universal/linux-3.18/arch/arm/boot/dts they are compatible with openwrt, but you need to replace the partition layout in this dts file since the partition names in dd-wrt are different

BrainSlayer, 2016-10-15:
i checked now vht160 in ap mode. it does also not work. my clients (laptop, iphone etc.) see the network, but are all unable to connect

yuhhaurlin, 2016-10-17:
Did you add the patch for hostapd to workaround 160 MHz operation? We had tested here with Macbook Pro and 88W8964 client mode (however, it can't work for 160 Mhz, it is connected as 80 MHz mode).

yuhhaurlin, 2016-10-17:
Can you try Kaloz's image to see if AP mode can work or not? Thanks.

BrainSlayer, 2016-10-17:
i used latest hostapd which is used in openwrt as well. so this  patch is included. all is working. except for 160 mhz.. the same driver and hostapd works with 160 mhz on qca9984 cards. so since you have never tested 160 mhz on your side with different hardware than 88w8964 i assume 160 fails to comply the official 802.11ac standard and is not inter operational with other vendors. vht 80 nd 40 mhz mode works without issues in client and ap mode. 160 fails to connect in client and no one can associate in ap mode

BrainSlayer, 2016-10-17:
my test client by the way was a samsung galaxy note edge, a iphone 6. the wireless card in my macbook broke unfortunatly.but my older macbook pro retina has no ac support unlike the iphone 6 and one hint. i did not use any encryption in setup

yuhhaurlin, 2016-10-17:
O.K. So all the issues are related to 160 MHz. I will try to do more tests, I need to get mac80211 working first, maybe I will build LEDE. Thanks.

yuhhaurlin, 2016-10-17:
But we had tested mwlwifi AP mode with our proprietary driver for 160 Mhz, it can work.

yuhhaurlin, 2016-10-17:
From our test here, mwlwifi AP mode setting for 160 MHz can work. Macbook pro and mwlwifi 88W8964 can connect to it as 80 MHz. Our proprietary driver (client mode) can connect to it as 160 Mhz. 

I will check QA to test more clients. Thanks.

yuhhaurlin, 2016-10-17:
160 fails to connect in client and no one can associate in ap mode => If client only supports 80 Mhz, it can still connect to mwlwifi 88W8964 AP setting as 160 Mhz mode (the workaround patch of hostapd is do this kind of thing). What do you mean it can't associate with mwlwifi 88W8964 setting as 160 Mhz mode?

yuhhaurlin, 2016-10-17:
BTW, if your region code is 0x10 (cat /sys/kernel/debug/ieee80211/phy0/mwlwifi/device_pwrtbl). You should only set 36~64 as 160 MHz. And you need to wait for CAC (60 seconds) timeout to let the AP work.

BrainSlayer, 2016-10-17:
mwlwifi 88w8964 vht160  AP <--  iphone 6 -- no assoc (hostapd shows timeout assoc request)
mwlwifi 88w8964 vht80 AP <-- iphone 6 -- assoc works like everything else
mwlwifi 88w8964 STA --> ath10k netgear 7800 QCA9984 vht 160 AP --> no assoc
mwlwifi 88w8964 STA --> ath10k netgear r7800 QCA9984 vht 80 AP --> assoc works

the iphone is just a example. i tested also various other clients. same result

yuhhaurlin, 2016-10-17:
I had checked our QA to test more clients (however, it is on our own DB AP board with 88W8964 module). 
I will try to build LEDE for our won DB AP board first. I will also check with Imre to see how to build code to work on WRT3200ACM (Rango).

BTW, your test is done on channels 36 to 64 (I think your region code should be 0x10)?

BrainSlayer, 2016-10-17:
vht 160 in channel 36 wont work. all my end devices are european devices. we can only run vht 160 in channel 100 >  

root@DD-WRT:/sys/kernel/debug/ieee80211/phy0/mwlwifi# cat device_pwrtbl

power table loaded from dts: yes
firmware region code: 0x10
number of channel: 0

BrainSlayer, 2016-10-17:
it may under some circumstances that belkin sended me a early beta sample device which has strange values stored in card eeprom. do you know a way to check this?

yuhhaurlin, 2016-10-17:
The image tested right now is built from OpenWrt with the patch I mentioned for hostapd (which is hostapd-2016-01-15). I had checked LEDE, hostapd package is updated to 2016-09-05 and the patch is already there. So I think I will build LEDE for our own DB AP board first before I can build code for WRT3200ACM (Rango) to see if there is any problems for AP setting as 160 Mhz.

BrainSlayer, 2016-10-17:
my test was runnin on channel 100 which works fine for other cards in vht160. i may check channel 36 now

yuhhaurlin, 2016-10-17:
You can check region code via "cat /sys/kernel/debug/ieee80211/phy0/mwlwifi/device_pwrtbl". I suggest that you must program device power table to your device, otherwise the power level is bad.

yuhhaurlin, 2016-10-17:
number of channel: 0 => It means your device does not program device power table. You need to check with Belkin to program power table to it first.

BrainSlayer, 2016-10-17:
update: channel 36 works. channel 100 > doesnt work for some reason but initialized without any error

yuhhaurlin, 2016-10-17:
If region code is 0x10, you can only set 36 to 64 as 160 MHz.

BrainSlayer, 2016-10-17:
do you have a working dts set for the device power table? i have seen that the power table format for the 88w6964 is much different. belkin uses the propertiery driver. and from the first view it may be that the mwlwifi driver just supports the dts power table for 88w6864 cards but not the 69 variant

yuhhaurlin, 2016-10-17:
So channel 36 (160 Mhz) can work? If yes, your have same test result as I have here.
You need to program power table to your device. If you program Europe region code, you can have two 160 MHz segment.

yuhhaurlin, 2016-10-17:
In order to comply with FCC rules. There is no way to provide power table from DTS for new device with device power table. Please check with Belkin to program device power table for you. If you need to change region code, you need to check with them too.

yuhhaurlin, 2016-10-17:
In order to comply with new FCC rules, client is also be restricted to work with region code. So I think if you set channel 36 (160 MHz). Mwlwifi client should connect to the AP as 80 MHz.

BrainSlayer, 2016-10-17:
from mac80211 region set vht160 is not enabled in regulatory database for EU in no channel range. i have manually patched this to enable it. but is there a reason for disabling vht160 in US for channels 1 >. because it seems to be allowed and the driver does report it as vht160 capable. this is why my gui shows it as selectable option. it uses the data returned by the driver matched with mac80211 regulatory database

BrainSlayer, 2016-10-17:
this is the official crda regulatory database  for US

country US: DFS-FCC
	(2402 - 2472 @ 40), (30)
	# 5.15 ~ 5.25 GHz: 30 dBm for master mode, 23 dBm for clients
	(5170 - 5250 @ 80), (23), AUTO-BW
	(5250 - 5330 @ 80), (23), DFS, AUTO-BW
	(5490 - 5730 @ 160), (23), DFS
	(5735 - 5835 @ 80), (30)
	# 60g band
	# reference: http://cfr.regstoday.com/47cfr15.aspx#47_CFR_15p255
	# channels 1,2,3, EIRP=40dBm(43dBm peak)
	(57240 - 63720 @ 2160), (40)

you see that it wont allow vht160 at channel 36, but from channel 100 >

yuhhaurlin, 2016-10-17:
I will check it.

BrainSlayer, 2016-10-17:
the reason is simple here. it has different dfs properties from the 36 - 64 channel space which would overlap with vht160. and for channel 100 > its clear. DFS is required, but vht160 should be enabled which is now turning to be a firmware issue in mwlwifi

yuhhaurlin, 2016-10-17:
No, there is no restriction for driver and firmware. Form the code, you can check it allows you to set three 160 MHz segment. What I need to check is if upper layer has blocked this kind of thing. I will try to change  
db.txt and to do test.

yuhhaurlin, 2016-10-17:
Did you use static regulatory database? If yes, can you send your db.txt to me?

BrainSlayer, 2016-10-17:
my db.txt is patched :-) so it seems the firmware is blocking these channels. the mwldriver has no restrictions here

yuhhaurlin, 2016-10-17:
Please send your db.txt to me and please help to confirm 36-64 is working. Thanks.

BrainSlayer, 2016-10-17:
please send  me a email to s.gottschall@dd-wrt.com i will reply with my db.txt

BrainSlayer, 2016-10-17:
and i can confirm that 36 - 64 is working with patched db.txt but channel 100 upwards is likelly a requirement which must be fullfilled. with standard mac80211 db.txt people will be out of luck and for sure openwrt uses the unpatched variant. so it needs to be clarified why channel 100 upwards is blocked

BrainSlayer, 2016-10-17:
tested now channel 36 upper with iphone 6 again. works. channel 64 lower works too. later today i will rerun vht160 tests with qca9984 on the same band in client and ap mode. but again. dont forget to discuss the channel 100 issue

yuhhaurlin, 2016-10-17:
Please let me know if 160 MHz is working for client mode with updated mac80211.

BrainSlayer, 2016-10-17:
you will get the answer within the next 60 minutes. just need to reconfigure my ap test environment

BrainSlayer, 2016-10-17:
QCA9984 AP in VHT160 US channel 36. client mode on 88w6964  associates without any issue. but. no data packet is going though. i cannot ping the ap.

yuhhaurlin, 2016-10-17:
All right. I will try to test 88W8964 160 Mhz with updated mac80211 (or patch current mac80211 module of OpenWrt). Thanks.

BrainSlayer, 2016-10-17:
try also a different platform like QCA9984. i will try to get more debug informations from ap side later today. to see whats going wrong. i got also similar client mode problems with current driver from 88w6864 users (wrt1900acv2 etc.) need to setup these devices too to validate if its a generic mwlwifi issue or chipset / firmware related. i have seen that you changed alot of client mode related code in mwlwifi.

yuhhaurlin, 2016-10-17:
I have not platform QCA9984. I will let it work for mwlwifi AP mode first. Client mode had been used by other projects and works well.

BrainSlayer, 2016-10-17:
unfortunatly the qca9984 is my only vht160 capable other end device. i just have a single rango / wrt3200acm preproduction device

yuhhaurlin, 2016-10-17:
I will make sure mwlwifi client 160 MHz can work and check your help to test it with qca9984. Thanks.

BrainSlayer, 2016-10-17:
after reveral tried (rebooted ap several times. it works now) bandwidth is also within expected range. but i will run a long test now with full bandwidth load over 24 hours

yuhhaurlin, 2016-10-17:
I think you need to  program power table to your device. Otherwise, power level is bad.

[68 more replies]

closed 2017-06-07 by yuhhaurlin