kaloz/mwlwifi #112: 5GHz stream drops after heavy load

gsustek, 2016-10-13:
http://pastebin.com/aBKNLr1N
driver version:
10.3.0.18-20160823-1

 Linux version 4.4.24 (gs@gs-E5450) (gcc version 6.2.0 (LEDE GCC 6.2.0 r1814) ) #0 SMP Sat Oct 8 11:50:54 2016

reference to fix:
http://marc.info/?l=linux-wireless&m=147560260029581&w=2

driver version

cat /sys/kernel/debug/ieee80211/phy0/mwlwifi/info

driver name: mwlwifi
chip type: 88W8864
hw version: 7
driver version: 10.3.0.18-20160823-1
firmware version: 0x0702091a
power table loaded from dts: yes
firmware region code: 0x0
mac address: 00:25:9c:13:b4:b0
2g: disable
5g: enable
antenna: 4 4
irq number: 104
iobase0: e1080000
iobase1: e1200000
tx limit: 768
rx limit: 64
ap macid support: 0000ffff
sta macid support: 00010000
macid used: 00000001
qe trigger number: 35957

gufus, 2016-10-13:
Ya

Not so fast, the new driver broke Client Bridge mode here :(

gufus, 2016-10-13:
Ya…

 

Not so fast, the new drivert broke Client Bridge mode here :(

 

http://pastebin.com/aBKNLr1N

http://marc.info/?l=linux-wireless <http://marc.info/?l=linux-wireless&m=147560260029581&w=2> &m=147560260029581&w=2

yuhhaurlin, 2016-10-14:
How to reproduce this problem?

gufus, 2016-10-14:
@yuhhaurlin 

My setup is 2 wrt1900ac 

1 unit is in AP mode, both (2g and 5g up

2g
100% 	Mode: Master | SSID: gypsynet
BSSID: 00:25:9C:13:34:6A | Encryption: WPA2 PSK (CCMP)
Channel: 6 (2.437 GHz) | Tx-Power: 30 dBm
Signal: -30 dBm | Noise: -67 dBm
Bitrate: 229.5 Mbit/s | Country: CA

5g
82% 	Mode: Master | SSID: gypsynet_5g
BSSID: 00:25:9C:13:34:6B | Encryption: WPA2 PSK (CCMP)
Channel: 161 (5.805 GHz) | Tx-Power: 30 dBm
Signal: -52 dBm | Noise: -88 dBm
Bitrate: 1053.0 Mbit/s | Country: CA

wifi driver on AP unit 3.18.20+10.3.0.3-2015..4-1

2 unit is in Client Bridge mode

wifi driver on Client Bridge unit 10.3.1.0-20160917

All wifi settings the same on both unita(s)

The Client Bridge works great

Rates on Client Bridge both 2g and 5g
http://www.gypsy-designs.com/4.jpg

yuhhaurlin, 2016-10-14:
For our internal basic acceptable test, it will do AP and STA mode performance test. For STA mode, it is setup via relayd (Routed Client with relayd). I did not receive report about host command timeout for this version of mwlwifi.

Client mode of mwlwifi also had been used by another project (Android platform with backports package 4.2.6-1), it also does not encounter host command timeout problem (running YouTube for more than 5 days).

So I need to know how to reproduce the problem that you mentioned.

yuhhaurlin, 2016-10-14:
10.3.1.0-20160917?  Did I commit this version of code?

gufus, 2016-10-14:
-----

No.

This is the dd-wrt driver, when you started working with Sebastian Gottschall

root@DD-WRT 5d:e1:f6:~# cat /sys/kernel/debug/ieee80211/phy1/mwlwifi/info

driver version: 10.3.1.0-20160917
mac address: 00:25:9c:13:0b:a1
irq number: 90
iobase0: d1000000
iobase1: d1200000
macid used: 00010001
qe trigger number: 412327

root@DD-WRT 5d:e1:f6:~#

yuhhaurlin, 2016-10-14:
If there is no precise information to reproduce this problem, I will close this one first. You can create anther one or reopen it if you can give me more precise information.

gsustek, 2016-10-14:
yuhharulin, @gufus hijacking my issue!!!!

Please do not close it.
So i just try to copy over ftp and 5GHz Wifi huge file of 43GB. after 2Gb, wifi stops working.
 if you need more additional data, please ask for it. or some logs.
Do you need my setup or build configuration?

Regards,
Goran.

yuhhaurlin, 2016-10-14:
Please use updated version to do that.

yuhhaurlin, 2016-10-14:
You can get pre-built images from https://downloads.openwrt.org/people/kaloz/. Please use this image to reproduce this problem on your device (I think it is latest version of mwlwifi driver).

gsustek, 2016-10-14:
what so you mean, to use latest driver?

yuhhaurlin, 2016-10-14:
Kaloz has pre-built images which should include latest driver. You can use these images to reproduce your problem and let me know the precise steps to reproduce it.

gufus, 2016-10-14:
@gsustek

Sorry.

I Will stay OUT of this issue.

Regards

gsustek, 2016-10-14:
hi, with latest lede and newest wifi driver, i have no issue...
for now, you can close this issue.

Regards,

closed 2016-10-17 by yuhhaurlin