kaloz/mwlwifi #113: Temperature is reported incorrectly for the CPU (WRT1900ACS v1) with 10.3.2.0-20161011

shmerl, 2016-10-20:
I'm using latest DD-WRT build (10-18-2016-r3077) which uses driver 10.3.2.0-20161011 with  Marvell Armada 385 and 88W8864.

It reports CPU temperature incorrectly (0°C or even negative).

shmerl, 2016-10-20:
Probably incorrect place to open it, since it's not WiFi related.

shmerl, 2017-01-11:
Reopening, since it can be related to thermal monitoring in thermal.* Feel free to close it if it's not relevant.

yuhhaurlin, 2017-01-11:
I don't think it is related to WiFi.

closed 2017-01-11 by yuhhaurlin

BrainSlayer, 2017-01-11:
its not related to the driver. it was a bug in my firmware code which has been already resolved

BrainSlayer, 2017-01-11:
as side node. under some conditions the cpu temperature is reported wrong. but this is related to the armada cpu and not to the wifi chipset. this happens spuriously and can only be fixed by power switching the device. its not clear why this happens.

shmerl, 2017-01-11:
It has been happening to me pretty regularly after certain update, so it must be some regression. Also reported [here](http://svn.dd-wrt.com/ticket/5592).

yuhhaurlin, 2017-01-11:
If stock firmware is all right, I think maybe you need to check with Belkin. This GitHub only handles WiFi problems.

BrainSlayer, 2017-01-11:
stock firmware does not offer to monitor the cpu temperature

BrainSlayer, 2017-01-11:
consider that im the developer of DD-WRT. i'm aware of this issue, but i dont know the reason. i just know that its working again after reboot

BrainSlayer, 2017-01-11:
on the second hand its not the latest version of dd-wrt which you're used

yuhhaurlin, 2017-01-11:
Maybe you can check with LEDE or OpenWrt (Someone from these two groups should work for BSP).

shmerl, 2017-01-11:
I'm using the latest from here: ftp://ftp.dd-wrt.com/betas/2016/12-15-2016-r30949/
If there is something later, i probably have missed that.

BrainSlayer, 2017-01-11:
just checked it again. consider this kernel log error for the sensor driver. "thermal thermal_zone0: Temperature sensor reading not valid"
root@DD-WRT:/sys/devices/virtual/hwmon/hwmon0# cat temp1_input
cat: read error: I/O error

yuhhaurlin, 2017-01-11:
In fact, thermal function of mwlwifi does not work for 88W8864. So this problem should not be related to mwlwifi.

int mwl_thermal_register(struct mwl_priv *priv)
{
	struct thermal_cooling_device *cdev;
	struct device *hwmon_dev;
	int ret;

	if (priv->chip_type != MWL8897)
		return 0;

shmerl, 2017-01-11:
I see it's using armada_thermal module. May be there was some update to the kernel that make the module not compatible?

yuhhaurlin, 2017-01-11:
Basically, for 88W8897, this function is used to get temperature of WiFi chip and another function allow user to set quiet mode of 88W8897. However, 88W8864 does not support these things. So if you use 88W8864, mwlwifi does not register anything to Linux thermal framework.

shmerl, 2017-01-11:
Yeah, it's a separate module from mwlwifi.

BrainSlayer, 2017-01-11:
i just discovered this. should fix the issue now http://svn.dd-wrt.com/changeset/31126

shmerl, 2017-01-11:
Ah, interesting. Thanks.