kaloz/mwlwifi #114: No information about channel RX/busy time

ignisf, 2016-10-24:
This means that ACS does not work properly.

```
Mon Oct 24 08:35:03 2016 kern.info kernel: [    0.000000] Booting Linux on physical CPU 0x0
Mon Oct 24 08:35:03 2016 kern.notice kernel: [    0.000000] Linux version 4.4.27 (openwrt-build@eric) (gcc version 5.4.0 (LEDE GCC 5.4.0 r1777) ) #0 SMP Mon Oct 24 08:34:59 2016
Mon Oct 24 08:35:03 2016 kern.info kernel: [    0.000000] CPU: ARMv7 Processor [414fc091] revision 1 (ARMv7), cr=10c5387d
Mon Oct 24 08:35:03 2016 kern.info kernel: [    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
Mon Oct 24 08:35:03 2016 kern.info kernel: [    0.000000] Machine model: Linksys WRT1900ACS
Mon

.....

Mon Oct 24 08:35:03 2016 kern.debug kernel: [    9.098583] ieee80211 phy1: priv->iobase0 = e1180000
Mon Oct 24 08:35:03 2016 kern.debug kernel: [    9.098595] ieee80211 phy1: priv->iobase1 = e1300000
Mon Oct 24 08:35:03 2016 kern.debug kernel: [    9.098631] ieee80211 phy1: priv->pcmd_buf = dda44000  priv->pphys_cmd_buf = 1da44000
Mon Oct 24 08:35:03 2016 kern.debug kernel: [    9.105061] ieee80211 phy1: fw download start
Mon Oct 24 08:35:03 2016 kern.debug kernel: [    9.224039] ieee80211 phy1: FwSize = 118776 downloaded Size = 118776 curr_iteration 65511
Mon Oct 24 08:35:03 2016 kern.debug kernel: [   10.087538] ieee80211 phy1: fw download complete
Mon Oct 24 08:35:03 2016 kern.debug kernel: [   10.088551] ieee80211 phy1: pcmd = dda44000
Mon Oct 24 08:35:03 2016 kern.info kernel: [   10.106539] ieee80211 phy1: firmware version: 0x702091a
Mon Oct 24 08:35:03 2016 kern.info kernel: [   10.129817] ieee80211 phy1: firmware region code: 0

......................

Mon Oct 24 08:35:06 2016 kern.debug kernel: [   14.981550] ieee80211 phy1: change: 0x40
Mon Oct 24 08:35:06 2016 kern.debug kernel: [   15.171538] ieee80211 phy1: change: 0x40
Mon Oct 24 08:35:06 2016 kern.debug kernel: [   15.361536] ieee80211 phy1: change: 0x40
Mon Oct 24 08:35:06 2016 kern.debug kernel: [   15.551538] ieee80211 phy1: change: 0x40
Mon Oct 24 08:35:06 2016 kern.debug kernel: [   15.741540] ieee80211 phy1: change: 0x40
Mon Oct 24 08:35:07 2016 kern.debug kernel: [   15.931536] ieee80211 phy1: change: 0x40
Mon Oct 24 08:35:07 2016 kern.info kernel: [   16.101532] br-wireless: port 2(wlan0) entered forwarding state
Mon Oct 24 08:35:07 2016 kern.debug kernel: [   16.121540] ieee80211 phy1: change: 0x40
Mon Oct 24 08:35:07 2016 kern.debug kernel: [   16.311540] ieee80211 phy1: change: 0x40
Mon Oct 24 08:35:07 2016 daemon.notice netifd: radio1 (1362): handle_probe_req: send failed
Mon Oct 24 08:35:07 2016 daemon.notice netifd: radio1 (1362): handle_probe_req: send failed
Mon Oct 24 08:35:07 2016 kern.debug kernel: [   16.501542] ieee80211 phy1: change: 0x40
Mon Oct 24 08:35:07 2016 daemon.notice netifd: radio1 (1362): handle_probe_req: send failed
Mon Oct 24 08:35:07 2016 daemon.notice netifd: radio1 (1362): handle_probe_req: send failed
Mon Oct 24 08:35:07 2016 kern.debug kernel: [   16.691541] ieee80211 phy1: change: 0x40
Mon Oct 24 08:35:07 2016 kern.debug kernel: [   16.881538] ieee80211 phy1: change: 0x40
Mon Oct 24 08:35:08 2016 kern.debug kernel: [   17.071538] ieee80211 phy1: change: 0x40
Mon Oct 24 08:35:08 2016 kern.debug kernel: [   17.160551] ieee80211 phy1: change: 0x100
Mon Oct 24 08:35:08 2016 kern.debug kernel: [   17.169579] ieee80211 phy1: change: 0x100
Mon Oct 24 08:35:08 2016 kern.debug kernel: [   17.178663] ieee80211 phy1: change: 0x40
Mon Oct 24 08:35:08 2016 kern.debug kernel: [   17.371537] ieee80211 phy1: change: 0x40
Mon Oct 24 08:35:08 2016 kern.debug kernel: [   17.561539] ieee80211 phy1: change: 0x40
Mon Oct 24 08:35:08 2016 kern.debug kernel: [   17.751536] ieee80211 phy1: change: 0x40
Mon Oct 24 08:35:09 2016 kern.debug kernel: [   17.941537] ieee80211 phy1: change: 0x40
Mon Oct 24 08:35:09 2016 kern.debug kernel: [   18.131535] ieee80211 phy1: change: 0x40
Mon Oct 24 08:35:09 2016 kern.debug kernel: [   18.321538] ieee80211 phy1: change: 0x40
Mon Oct 24 08:35:09 2016 kern.debug kernel: [   18.511538] ieee80211 phy1: change: 0x40
Mon Oct 24 08:35:09 2016 kern.debug kernel: [   18.701536] ieee80211 phy1: change: 0x40
Mon Oct 24 08:35:10 2016 kern.debug kernel: [   18.891537] ieee80211 phy1: change: 0x40
Mon Oct 24 08:35:10 2016 kern.debug kernel: [   19.081538] ieee80211 phy1: change: 0x40
Mon Oct 24 08:35:10 2016 kern.debug kernel: [   19.271538] ieee80211 phy1: change: 0x40
Mon Oct 24 08:35:10 2016 kern.debug kernel: [   19.461549] ieee80211 phy1: change: 0x40
Mon Oct 24 08:35:10 2016 kern.debug kernel: [   19.651539] ieee80211 phy1: change: 0x40
Mon Oct 24 08:35:10 2016 kern.debug kernel: [   19.740551] ieee80211 phy1: change: 0x100
Mon Oct 24 08:35:10 2016 kern.debug kernel: [   19.749579] ieee80211 phy1: change: 0x100
Mon Oct 24 08:35:10 2016 kern.debug kernel: [   19.758612] ieee80211 phy1: change: 0x40
Mon Oct 24 08:35:11 2016 kern.debug kernel: [   19.951539] ieee80211 phy1: change: 0x40
Mon Oct 24 08:35:11 2016 kern.debug kernel: [   20.141537] ieee80211 phy1: change: 0x40
Mon Oct 24 08:35:11 2016 kern.debug kernel: [   20.331537] ieee80211 phy1: change: 0x40
Mon Oct 24 12:39:23 2016 kern.debug kernel: [   20.521566] ieee80211 phy1: change: 0x40
Mon Oct 24 12:39:23 2016 kern.debug kernel: [   20.711540] ieee80211 phy1: change: 0x40
Mon Oct 24 12:39:23 2016 kern.debug kernel: [   20.901538] ieee80211 phy1: change: 0x40
Mon Oct 24 12:39:23 2016 daemon.notice netifd: radio1 (1362): handle_probe_req: send failed
Mon Oct 24 12:39:23 2016 kern.debug kernel: [   21.091539] ieee80211 phy1: change: 0x40
Mon Oct 24 12:39:23 2016 kern.debug kernel: [   21.281537] ieee80211 phy1: change: 0x40
Mon Oct 24 12:39:24 2016 kern.debug kernel: [   21.471536] ieee80211 phy1: change: 0x40
[... 35 lines cut ...]
Mon Oct 24 12:39:29 2016 kern.debug kernel: [   27.201603] ieee80211 phy1: change: 0x40
Mon Oct 24 12:39:29 2016 daemon.notice netifd: radio1 (1362): ACS: Survey is missing channel time
Mon Oct 24 12:39:29 2016 daemon.notice netifd: radio1 (1362): ACS: Survey is missing RX and busy time (at least one is required)
Mon Oct 24 12:39:29 2016 daemon.notice netifd: radio1 (1362): ACS: Survey is missing channel time
Mon Oct 24 12:39:29 2016 daemon.notice netifd: radio1 (1362): ACS: Survey is missing RX and busy time (at least one is required)
Mon Oct 24 12:39:29 2016 daemon.notice netifd: radio1 (1362): ACS: Survey is missing channel time
Mon Oct 24 12:39:29 2016 daemon.notice netifd: radio1 (1362): ACS: Survey is missing RX and busy time (at least one is required)
Mon Oct 24 12:39:29 2016 daemon.notice netifd: radio1 (1362): ACS: Survey is missing channel time
Mon Oct 24 12:39:29 2016 daemon.notice netifd: radio1 (1362): ACS: Survey is missing RX and busy time (at least one is required)
Mon Oct 24 12:39:29 2016 daemon.notice netifd: radio1 (1362): ACS: Survey is missing channel time
Mon Oct 24 12:39:29 2016 daemon.notice netifd: radio1 (1362): ACS: Survey is missing RX and busy time (at least one is required)
Mon Oct 24 12:39:29 2016 daemon.notice netifd: radio1 (1362): ACS: Survey is missing channel time
Mon Oct 24 12:39:29 2016 daemon.notice netifd: radio1 (1362): ACS: Survey is missing RX and busy time (at least one is required)
Mon Oct 24 12:39:29 2016 daemon.notice netifd: radio1 (1362): ACS: Survey is missing channel time
Mon Oct 24 12:39:29 2016 daemon.notice netifd: radio1 (1362): ACS: Survey is missing RX and busy time (at least one is required)
Mon Oct 24 12:39:29 2016 daemon.notice netifd: radio1 (1362): ACS: Survey is missing channel time
Mon Oct 24 12:39:29 2016 daemon.notice netifd: radio1 (1362): ACS: Survey is missing RX and busy time (at least one is required)
Mon Oct 24 12:39:29 2016 daemon.notice netifd: radio1 (1362): ACS: Survey is missing channel time
Mon Oct 24 12:39:29 2016 daemon.notice netifd: radio1 (1362): ACS: Survey is missing RX and busy time (at least one is required)
Mon Oct 24 12:39:29 2016 daemon.notice netifd: radio1 (1362): ACS: Survey is missing channel time
Mon Oct 24 12:39:29 2016 daemon.notice netifd: radio1 (1362): ACS: Survey is missing RX and busy time (at least one is required)
Mon Oct 24 12:39:29 2016 daemon.notice netifd: radio1 (1362): ACS: Survey is missing channel time
Mon Oct 24 12:39:29 2016 daemon.notice netifd: radio1 (1362): ACS: Survey is missing RX and busy time (at least one is required)
Mon Oct 24 12:39:29 2016 daemon.notice netifd: radio1 (1362): wlan1: ACS-COMPLETED freq=2412 channel=1
```

yuhhaurlin, 2016-10-24:
Can you try pre-built images from https://downloads.openwrt.org/people/kaloz/?

ignisf, 2016-10-24:
sure, will update you when I do.

ignisf, 2016-10-25:
Hello again,

Unfortunately I'm still seeing no information about the channel use:

```
Tue Oct 25 18:03:03 2016 daemon.notice netifd: radio1 (3288): ACS: Survey is missing channel time
Tue Oct 25 18:03:03 2016 daemon.notice netifd: radio1 (3288): ACS: Survey is missing channel time
Tue Oct 25 18:03:03 2016 daemon.notice netifd: radio1 (3288): ACS: Survey is missing channel time
Tue Oct 25 18:03:03 2016 daemon.notice netifd: radio1 (3288): ACS: Survey is missing channel time
Tue Oct 25 18:03:03 2016 daemon.notice netifd: radio1 (3288): ACS: Survey is missing channel time
Tue Oct 25 18:03:03 2016 daemon.notice netifd: radio1 (3288): ACS: Channel 1 has insufficient survey data
Tue Oct 25 18:03:03 2016 daemon.notice netifd: radio1 (3288): ACS: Survey is missing channel time
Tue Oct 25 18:03:03 2016 daemon.notice netifd: radio1 (3288): ACS: Survey is missing channel time
Tue Oct 25 18:03:03 2016 daemon.notice netifd: radio1 (3288): ACS: Survey is missing channel time
Tue Oct 25 18:03:03 2016 daemon.notice netifd: radio1 (3288): ACS: Survey is missing channel time
Tue Oct 25 18:03:03 2016 daemon.notice netifd: radio1 (3288): ACS: Survey is missing channel time
Tue Oct 25 18:03:03 2016 daemon.notice netifd: radio1 (3288): ACS: Channel 1 has insufficient survey data
Tue Oct 25 18:03:03 2016 daemon.notice netifd: radio1 (3288): ACS: Survey is missing channel time
Tue Oct 25 18:03:03 2016 daemon.notice netifd: radio1 (3288): ACS: Survey is missing channel time
Tue Oct 25 18:03:03 2016 daemon.notice netifd: radio1 (3288): ACS: Survey is missing channel time
Tue Oct 25 18:03:03 2016 daemon.notice netifd: radio1 (3288): ACS: Survey is missing channel time
Tue Oct 25 18:03:03 2016 daemon.notice netifd: radio1 (3288): ACS: Survey is missing channel time
Tue Oct 25 18:03:03 2016 daemon.notice netifd: radio1 (3288): ACS: Channel 1 has insufficient survey data
Tue Oct 25 18:03:03 2016 daemon.notice netifd: radio1 (3288): ACS: Survey is missing channel time
Tue Oct 25 18:03:03 2016 daemon.notice netifd: radio1 (3288): ACS: Survey is missing channel time
Tue Oct 25 18:03:03 2016 daemon.notice netifd: radio1 (3288): ACS: Survey is missing channel time
Tue Oct 25 18:03:03 2016 daemon.notice netifd: radio1 (3288): ACS: Survey is missing channel time
Tue Oct 25 18:03:03 2016 daemon.notice netifd: radio1 (3288): ACS: Survey is missing channel time
Tue Oct 25 18:03:03 2016 daemon.notice netifd: radio1 (3288): ACS: Channel 1 has insufficient survey data
Tue Oct 25 18:03:03 2016 daemon.notice netifd: radio1 (3288): ACS: Survey is missing channel time
Tue Oct 25 18:03:03 2016 daemon.notice netifd: radio1 (3288): ACS: Survey is missing channel time
Tue Oct 25 18:03:03 2016 daemon.notice netifd: radio1 (3288): ACS: Survey is missing channel time
Tue Oct 25 18:03:03 2016 daemon.notice netifd: radio1 (3288): ACS: Survey is missing channel time
Tue Oct 25 18:03:03 2016 daemon.notice netifd: radio1 (3288): ACS: Survey is missing channel time
Tue Oct 25 18:03:03 2016 daemon.notice netifd: radio1 (3288): ACS: Channel 1 has insufficient survey data
Tue Oct 25 18:03:03 2016 daemon.notice netifd: radio1 (3288): ACS: Survey is missing channel time
Tue Oct 25 18:03:03 2016 daemon.notice netifd: radio1 (3288): ACS: Survey is missing channel time
Tue Oct 25 18:03:03 2016 daemon.notice netifd: radio1 (3288): ACS: Survey is missing channel time
```

same for radio0.

```
root@OpenWrt:/# ethtool -S wlan0
NIC statistics:
     rx_packets: 0
     rx_bytes: 0
     rx_duplicates: 0
     rx_fragments: 0
     rx_dropped: 0
     tx_packets: 0
     tx_bytes: 0
     tx_filtered: 0
     tx_retry_failed: 0
     tx_retries: 0
     sta_state: 0
     txrate: 0
     rxrate: 0
     signal: 0
     channel: 5180
     noise: 168
     ch_time: 18446744073709551615
     ch_time_busy: 18446744073709551615
     ch_time_ext_busy: 18446744073709551615
     ch_time_rx: 18446744073709551615
     ch_time_tx: 18446744073709551615
root@OpenWrt:/# ethtool -S wlan1
NIC statistics:
     rx_packets: 0
     rx_bytes: 0
     rx_duplicates: 0
     rx_fragments: 0
     rx_dropped: 0
     tx_packets: 0
     tx_bytes: 0
     tx_filtered: 0
     tx_retry_failed: 0
     tx_retries: 0
     sta_state: 0
     txrate: 0
     rxrate: 0
     signal: 0
     channel: 2412
     noise: 170
     ch_time: 18446744073709551615
     ch_time_busy: 18446744073709551615
     ch_time_ext_busy: 18446744073709551615
     ch_time_rx: 18446744073709551615
     ch_time_tx: 18446744073709551615
root@OpenWrt:/# 

```
root@OpenWrt:/# iw wlan0 survey dump
Survey data from wlan0
	frequency:			5180 MHz
	noise:				-89 dBm
root@OpenWrt:/# iw wlan1 survey dump
Survey data from wlan1
	frequency:			2412 MHz
	noise:				-87 dBm
root@OpenWrt:/# 
```

yuhhaurlin, 2016-10-25:
Sorry. Auto channel selection is not working.

yuhhaurlin, 2016-11-23:
ACS can work now. Please test https://github.com/kaloz/mwlwifi/commit/90e99a735923a92a770afe3d84a2e4b834e010c0. Thanks.

ignisf, 2016-11-23:
Thanks, will do so the first chance I get

yuhhaurlin, 2016-11-24:
I think this issue is fixed. Thanks.

closed 2016-11-24 by yuhhaurlin