kaloz/mwlwifi #115: Fails to build due to dropped config_enabled in linux next-20161028

seragh, 2016-10-28:
Commit ab494c6adcff01c8c56dfed616ebb5434a7c671e finally drops config_enabled macro

https://git.kernel.org/cgit/linux/kernel/git/next/linux-next.git/commit/?id=ab494c6adcff01c8c56dfed616ebb5434a7c671e

yuhhaurlin, 2016-11-01:
Thanks. We will check it later.

seragh, 2016-11-05:
Was added to mainline as c0a0aba8e478229b2f0956918542152fbad3f794

Therefore won't compile on Linux 4.9-rc3 and later.

yuhhaurlin, 2016-11-10:
Please check commit https://github.com/kaloz/mwlwifi/commit/ee0d12c3ea794ac4f9af125f5e9607daca9c93f5. Thanks.

closed 2016-11-10 by yuhhaurlin

seragh, 2016-11-10:
On Wed, 09 Nov 2016 19:02:43 -0800
yuhhaurlin <notifications@github.com> wrote:

> Please check commit
> https://github.com/kaloz/mwlwifi/commit/ee0d12c3ea794ac4f9af125f5e9607daca9c93f5.
> Thanks.
> 

Works, the second hunk fixes another issue tho :)

Thanks
Ralph