kaloz/mwlwifi #117: WRT3200ACM no test example for 160Mhz

Chadster766, 2016-11-11:
I'm trying to get 160Mhz to work in hostapd.conf on the WRT3200ACM but can't figure it out.

Please add an example to the mwlwifi/test/ directory.

yuhhaurlin, 2016-11-12:
Please check LEDE image of https://drive.google.com/open?id=0B3qLWtcWB9EdRHJoSGdUa2VBMFk. 160 MHz can work for AP and STA mode on channel 36 (160 MHz). About channel 100 (160 MHz), AP can allow this setting but client can't connect to it. We will check this problem. You can load this image and get configuration file by yourself.

Chadster766, 2016-11-12:
Thanks I saw that post in the other issue thread and had already gotten the 160Mhz config from that LEDE image :smiley: 

Still the 160Mhz hostapd conf documentation should be added to the driver module.

stuwilkins, 2016-11-13:
@yuhhaurlin is that image the same as the latest lede build? I need the sysupgrade version not the factory flash

yuhhaurlin, 2016-11-14:
This is just for verification of problem. You can build image by yourself or get pre-built image from community.

yuhhaurlin, 2016-11-14:
I will close this one.

closed 2016-11-14 by yuhhaurlin

Chadster766, 2016-11-14:
Why close this documentation issue before adding the 160Mhz documentation like the other hostapd configuration examples in the repo?

yuhhaurlin, 2016-11-14:
OpenWRT and LEDE will create configuration file. There is no need to add this kind of sample file in mwlwifi.

Chadster766, 2016-11-14:
But the repo already has this:

https://github.com/kaloz/mwlwifi/tree/master/test

Also are you now purposing that this driver be proprietary to OpenWRT and LEDE?

yuhhaurlin, 2016-11-14:
You can check hostapd to know how to configure it.