kaloz/mwlwifi #120: eee80211 phy1: cmd 0x8050=broadcast_ssid_enable timed out

acarlo79, 2016-11-16:
The upgrade to fstools within LEDE leads somehow to break the WiFI driver.
`
Wed Nov 16 08:29:11 2016 kern.err kernel: [  439.774217] ieee80211 phy1: cmd 0x8050=broadcast_ssid_enable timed out
Wed Nov 16 08:29:11 2016 kern.err kernel: [  439.780775] ieee80211 phy1: return code: 0x0050
Wed Nov 16 08:29:11 2016 kern.err kernel: [  439.785335] ieee80211 phy1: timeout: 0x0050
Wed Nov 16 08:29:11 2016 kern.err kernel: [  439.789534] ieee80211 phy1: failed execution
Wed Nov 16 08:29:15 2016 kern.err kernel: [  443.789032] ieee80211 phy1: cmd 0x8127=SetInformationElements timed out
Wed Nov 16 08:29:15 2016 kern.err kernel: [  443.795683] ieee80211 phy1: return code: 0x0127
Wed Nov 16 08:29:15 2016 kern.err kernel: [  443.800232] ieee80211 phy1: timeout: 0x0127
Wed Nov 16 08:29:15 2016 kern.err kernel: [  443.804438] ieee80211 phy1: failed execution
Wed Nov 16 08:29:15 2016 daemon.info dnsmasq-dhcp[2928]: DHCPREQUEST(br-lan) 192.168.20.123 a4:5e:60:e7:47:9d 
Wed Nov 16 08:29:15 2016 daemon.info dnsmasq-dhcp[2928]: DHCPACK(br-lan) 192.168.20.123 a4:5e:60:e7:47:9d Carlos-MBP
Wed Nov 16 08:29:19 2016 kern.err kernel: [  447.809028] ieee80211 phy1: cmd 0x801c=80211RadioControl timed out
Wed Nov 16 08:29:19 2016 kern.err kernel: [  447.815246] ieee80211 phy1: return code: 0x001c
Wed Nov 16 08:29:19 2016 kern.err kernel: [  447.819794] ieee80211 phy1: timeout: 0x001c
Wed Nov 16 08:29:19 2016 kern.err kernel: [  447.824001] ieee80211 phy1: failed execution
Wed Nov 16 08:29:23 2016 kern.err kernel: [  451.829023] ieee80211 phy1: cmd 0x8126=SetFixedRate timed out
Wed Nov 16 08:29:23 2016 kern.err kernel: [  451.834801] ieee80211 phy1: return code: 0x0126
Wed Nov 16 08:29:23 2016 kern.err kernel: [  451.839349] ieee80211 phy1: timeout: 0x0126
Wed Nov 16 08:29:23 2016 kern.err kernel: [  451.843556] ieee80211 phy1: failed execution
Wed Nov 16 08:29:27 2016 kern.err kernel: [  455.847011] ieee80211 phy1: cmd 0x8050=broadcast_ssid_enable timed out
Wed Nov 16 08:29:27 2016 kern.err kernel: [  455.853579] ieee80211 phy1: return code: 0x0050
Wed Nov 16 08:29:27 2016 kern.err kernel: [  455.858169] ieee80211 phy1: timeout: 0x0050
Wed Nov 16 08:29:27 2016 kern.err kernel: [  455.862375] ieee80211 phy1: failed execution
Wed Nov 16 08:29:31 2016 kern.err kernel: [  459.867007] ieee80211 phy1: cmd 0x8127=SetInformationElements timed out
Wed Nov 16 08:29:31 2016 kern.err kernel: [  459.873659] ieee80211 phy1: return code: 0x0127
Wed Nov 16 08:29:31 2016 kern.err kernel: [  459.878208] ieee80211 phy1: timeout: 0x0127
Wed Nov 16 08:29:31 2016 kern.err kernel: [  459.882414] ieee80211 phy1: failed execution
Wed Nov 16 08:29:35 2016 kern.err kernel: [  463.886998] ieee80211 phy1: cmd 0x801c=80211RadioControl timed out
Wed Nov 16 08:29:35 2016 kern.err kernel: [  463.893217] ieee80211 phy1: return code: 0x001c
Wed Nov 16 08:29:35 2016 kern.err kernel: [  463.897765] ieee80211 phy1: timeout: 0x001c
Wed Nov 16 08:29:35 2016 kern.err kernel: [  463.901973] ieee80211 phy1: failed execution
Wed Nov 16 08:29:39 2016 kern.err kernel: [  467.907002] ieee80211 phy1: cmd 0x8126=SetFixedRate timed out
Wed Nov 16 08:29:39 2016 kern.err kernel: [  467.912782] ieee80211 phy1: return code: 0x0126
Wed Nov 16 08:29:39 2016 kern.err kernel: [  467.917330] ieee80211 phy1: timeout: 0x0126
Wed Nov 16 08:29:39 2016 kern.err kernel: [  467.921536] ieee80211 phy1: failed execution
`

Found out which is exactly the change that introduced the issue with the wifi within LEDE.

d15dc7e85dbf38c07dd5a7f8068ee6655830548e is the first bad commit
commit d15dc7e85dbf38c07dd5a7f8068ee6655830548e
Author: John Crispin john@phrozen.org Date: Mon Oct 31 16:38:26 2016 +0100

  fstools: update to latest git HEAD
  
  adds f2fs support
  
  Signed-off-by: John Crispin <john@phrozen.org>

:040000 040000 d10d6f1edb2d916c88f844529897be401c0415af a654e081b6bea3460a4cd606c64dc3fd9b49d0af M package

The issue can be triggered in my case doing these steps:

1) attach a USB 3.0 HDD formatted ext4
2) create a samba share pointing to a folder on the HDD
3) try to transfer a large file from the HDD to the smb client.

The transfer will die after few hundreds megabytes and the systemlog will show the aforementioned kernel error messages on the wifi.

All the tests have been done using the 5GHz network in ac.

yuhhaurlin, 2016-11-23:
Can you help to test https://github.com/kaloz/mwlwifi/commit/fb68d695dca4c77894a914a5d85804dbc7629c28 to see if you will still encounter timeout of host command.

thagabe, 2016-11-23:
Testing as we speak

yuhhaurlin, 2016-11-23:
@thagabe You also encounter timeout of host command?

thagabe, 2016-11-23:
@yuhhaurlin I mostly work on ethernet as i find the ping more reliable but i have streamed and transfered large files to my estata drive formatted in xfs and have noticed messages in dmesg

yuhhaurlin, 2016-11-23:
Yes. This is one possible fix for firmware crash. Please help to test it. Thanks.

thagabe, 2016-11-23:
@yuhhaurlin 4GB file from HDD attached to WRT to PC (esata XFS)
           Ethernet          WIFI (5GHz <1 meter)
Read:  50 MB/s               12 MB/s
Write: 40 MB/s               10 MB/s

yuhhaurlin, 2016-11-23:
Did you encounter crash problem before for this test?

acarlo79, 2016-11-23:
I can build with this new commit tonight and run some tests.

acarlo79, 2016-11-23:
the new driver looks good and seems to have fixed the issue.
I transferred 50GB file over wifi with an average of 30MB/s (the hdd is not doing really much more).

awesome job!

acarlo79, 2016-11-23:
tested with: 
LEDE Reboot r2299 
4.4.32+10.3.2.0-20161123-1

closed 2016-11-23 by yuhhaurlin