kaloz/mwlwifi #121: Linksys WRT1200AC v2. 2.4Ghz goes down

gassan, 2016-11-19:
... if Ext.HDD Toshiba (formatted as ext4) connected to USB 3.0
It is only possible to scan and find other Wifi. No connects, No adversiting in ap mode.
With combo Usb2+ExtSata there are no problems. It is also possible to connect and mount Usb Flash stick(ntfs) via USB3.0

yuhhaurlin, 2016-11-21:
WiFi should be independent with USB. I can't understand your problem.

gassan, 2016-11-21:
[hier](https://youtu.be/cA2ILxrquL4) a small video.
SSID GGR_2.4GHz
at 1:30 goes down (silent?) and at end of video if reconnected to usb2+esata up again.

yuhhaurlin, 2016-11-22:
I did not encounter this kind of problem. Can you load pre-built image to do test?

gassan, 2016-11-22:
If this can be done with sysupgrade only (I have not USB-TTL etc), yes I can.
Can you please give me a link for my Hardware?
P.S. I suspect, you need dmesg output?

yuhhaurlin, 2016-11-23:
Can you try pre-built image form  https://downloads.openwrt.org/people/kaloz/? Thanks.

gassan, 2016-11-23:
Already done. it does not work to.
The last Driver, that has no conflicts with usb 3 is an Update for CC 15.05.1
The Driver from 13 Okt 2016 do not work in any prebuild. I think I have tested they all (4 or 5)

thagabe, 2016-11-23:
@gassan Would you like to try my build? esata/usb works well here with both radios still working well (tho i would have to rebuild since I had not included ext4 only btrfs and xfs)

yuhhaurlin, 2016-11-23:
@gassan Can you give me more debug information about your test? I don't think current driver will have this kind of problem.

gassan, 2016-11-23:
@thagabe Thanks you. I think i don't need it. what I need is a prebuild with a lot of debug info.
@yuhhaurlin What I have to send? dmesg? Content of /proc/sys/debug ?

yuhhaurlin, 2016-11-23:
Yes. Please use "dmesg" to get more information.

gassan, 2016-11-23:
I dont know, what is happening. but with new **LEDE Reboot r0+2299** works all perfect. Sorry for alarm. it was nothing changed on configs and/or installed apps.

closed 2016-11-23 by yuhhaurlin

gassan, 2016-11-23:
It does not work again
`[22838.006183] ieee80211 phy1: cmd 0x9122=UpdateEncryption timed out
[22838.012318] ieee80211 phy1: return code: 0x1122
[22838.016867] ieee80211 phy1: timeout: 0x1122
[22838.021074] ieee80211 phy1: failed execution
[22838.025363] wlan2g: failed to remove key (0, 60:21:c0:84:0c:cd) from hardware (-5)
[23028.160466] br-lan: port 2(vpn-home) entered disabled state
[23028.170940] device vpn-home left promiscuous mode
[23028.175860] br-lan: port 2(vpn-home) entered disabled state`

gassan, 2016-11-23:
@yuhhaurlin 
it don't work again

[22838.006183] ieee80211 phy1: cmd 0x9122=UpdateEncryption timed out
[22838.012318] ieee80211 phy1: return code: 0x1122
[22838.016867] ieee80211 phy1: timeout: 0x1122
[22838.021074] ieee80211 phy1: failed execution
[22838.025363] wlan2g: failed to remove key (0, 60:21:c0:84:0c:cd) from hardware (-5)
[23028.160466] br-lan: port 2(vpn-home) entered disabled state
[23028.170940] device vpn-home left promiscuous mode
[23028.175860] br-lan: port 2(vpn-home) entered disabled state

Some info. I use tinc for vpn. and yesteday was tinc upgraded.

yuhhaurlin, 2016-11-23:
If you encounter timeout of host command, please use latest driver. Issue  120.