kaloz/mwlwifi #122: Mamba 5GHz radio drops

thagabe, 2016-11-21:
`Mon Nov 21 12:52:49 2016 daemon.info hostapd: wlan0: STA d8:bb:2c:01:10:61 IEEE 802.11: authenticated
Mon Nov 21 12:52:49 2016 daemon.info hostapd: wlan0: STA d8:bb:2c:01:10:61 IEEE 802.11: associated (aid 1)
Mon Nov 21 12:52:49 2016 daemon.info hostapd: wlan0: STA d8:bb:2c:01:10:61 WPA: pairwise key handshake completed (RSN)
Mon Nov 21 12:52:51 2016 daemon.info hostapd: wlan0: STA 30:10:e4:5f:45:da IEEE 802.11: authenticated
Mon Nov 21 12:52:51 2016 daemon.info hostapd: wlan0: STA 30:10:e4:5f:45:da IEEE 802.11: associated (aid 2)
Mon Nov 21 12:52:51 2016 daemon.info hostapd: wlan0: STA 30:10:e4:5f:45:da WPA: pairwise key handshake completed (RSN)
Mon Nov 21 12:52:51 2016 daemon.info hostapd: wlan0: STA c0:11:73:00:9d:73 IEEE 802.11: authenticated
Mon Nov 21 12:52:51 2016 daemon.info hostapd: wlan0: STA c0:11:73:00:9d:73 IEEE 802.11: associated (aid 3)
Mon Nov 21 12:52:51 2016 daemon.info hostapd: wlan0: STA c0:11:73:00:9d:73 WPA: pairwise key handshake completed (RSN)
Mon Nov 21 12:52:52 2016 daemon.info dnsmasq-dhcp[3621]: DHCPREQUEST(br-lan) 192.168.1.126 c0:11:73:00:9d:73 
Mon Nov 21 12:52:52 2016 daemon.info dnsmasq-dhcp[3621]: DHCPACK(br-lan) 192.168.1.126 c0:11:73:00:9d:73 android-ac8b751371cb15b0
Mon Nov 21 12:52:53 2016 daemon.info hostapd: wlan0: STA ac:cf:85:b7:1a:3c IEEE 802.11: authenticated
Mon Nov 21 12:52:53 2016 daemon.info hostapd: wlan0: STA ac:cf:85:b7:1a:3c IEEE 802.11: associated (aid 4)
Mon Nov 21 12:52:53 2016 daemon.info hostapd: wlan0: STA ac:cf:85:b7:1a:3c WPA: pairwise key handshake completed (RSN)
Mon Nov 21 12:53:59 2016 daemon.info hostapd: wlan0: STA 30:10:e4:5f:45:da WPA: group key handshake completed (RSN)
Mon Nov 21 12:54:00 2016 daemon.info hostapd: wlan0: STA c0:11:73:00:9d:73 WPA: group key handshake completed (RSN)
Mon Nov 21 12:54:00 2016 daemon.info hostapd: wlan0: STA d8:bb:2c:01:10:61 WPA: group key handshake completed (RSN)
Mon Nov 21 12:54:00 2016 daemon.info hostapd: wlan0: STA ac:cf:85:b7:1a:3c WPA: group key handshake completed (RSN)
Mon Nov 21 12:55:00 2016 cron.info crond[3663]: USER root pid 4405 cmd /sbin/fan_ctrl.sh
Mon Nov 21 13:00:00 2016 cron.info crond[3663]: USER root pid 5081 cmd /sbin/fan_ctrl.sh
Mon Nov 21 13:03:59 2016 daemon.info hostapd: wlan0: STA c0:11:73:00:9d:73 WPA: group key handshake completed (RSN)
Mon Nov 21 13:04:00 2016 daemon.info hostapd: wlan0: STA ac:cf:85:b7:1a:3c WPA: group key handshake completed (RSN)
Mon Nov 21 13:04:07 2016 daemon.info hostapd: wlan0: STA 30:10:e4:5f:45:da IEEE 802.11: deauthenticated due to local deauth request
Mon Nov 21 13:04:07 2016 daemon.info hostapd: wlan0: STA d8:bb:2c:01:10:61 IEEE 802.11: deauthenticated due to local deauth request
Mon Nov 21 13:05:00 2016 cron.info crond[3663]: USER root pid 5827 cmd /sbin/fan_ctrl.sh
Mon Nov 21 13:10:00 2016 cron.info crond[3663]: USER root pid 6504 cmd /sbin/fan_ctrl.sh
Mon Nov 21 13:14:00 2016 daemon.info hostapd: wlan0: STA ac:cf:85:b7:1a:3c WPA: group key handshake completed (RSN)
Mon Nov 21 13:14:00 2016 daemon.info hostapd: wlan0: STA c0:11:73:00:9d:73 WPA: group key handshake completed (RSN)
Mon Nov 21 13:15:00 2016 cron.info crond[3663]: USER root pid 7249 cmd /sbin/fan_ctrl.sh
Mon Nov 21 13:20:00 2016 cron.info crond[3663]: USER root pid 8073 cmd /sbin/fan_ctrl.sh
Mon Nov 21 13:23:59 2016 daemon.info hostapd: wlan0: STA c0:11:73:00:9d:73 WPA: group key handshake completed (RSN)
Mon Nov 21 13:24:00 2016 daemon.info hostapd: wlan0: STA ac:cf:85:b7:1a:3c WPA: group key handshake completed (RSN)
Mon Nov 21 13:25:00 2016 cron.info crond[3663]: USER root pid 8745 cmd /sbin/fan_ctrl.sh
Mon Nov 21 13:30:00 2016 cron.info crond[3663]: USER root pid 9490 cmd /sbin/fan_ctrl.sh
Mon Nov 21 13:33:59 2016 daemon.info hostapd: wlan0: STA ac:cf:85:b7:1a:3c WPA: group key handshake completed (RSN)
Mon Nov 21 13:33:59 2016 daemon.info hostapd: wlan0: STA c0:11:73:00:9d:73 WPA: group key handshake completed (RSN)
Mon Nov 21 13:34:55 2016 daemon.info hostapd: wlan0: STA 68:db:ca:8f:11:91 IEEE 802.11: authenticated
Mon Nov 21 13:34:55 2016 daemon.info hostapd: wlan0: STA 68:db:ca:8f:11:91 IEEE 802.11: associated (aid 1)
Mon Nov 21 13:34:55 2016 daemon.info hostapd: wlan0: STA 68:db:ca:8f:11:91 WPA: pairwise key handshake completed (RSN)
Mon Nov 21 13:34:55 2016 daemon.info dnsmasq-dhcp[3621]: DHCPREQUEST(br-lan) 192.168.1.172 68:db:ca:8f:11:91 
Mon Nov 21 13:34:55 2016 daemon.info dnsmasq-dhcp[3621]: DHCPACK(br-lan) 192.168.1.172 68:db:ca:8f:11:91 Magalys-iPhone
Mon Nov 21 13:34:57 2016 daemon.warn odhcpd[1205]: DHCPV6 SOLICIT IA_NA from 000100011efca59b68dbca8f1191 on br-lan: ok 2601:646:c280:313::ef7/128 fd40:fa0a:7966::ef7/128 
Mon Nov 21 13:34:59 2016 daemon.warn odhcpd[1205]: DHCPV6 REQUEST IA_NA from 000100011efca59b68dbca8f1191 on br-lan: ok 2601:646:c280:313::ef7/128 fd40:fa0a:7966::ef7/128 
Mon Nov 21 13:34:59 2016 daemon.warn odhcpd[1205]: DHCPV6 REQUEST IA_NA from 000100011efca59b68dbca8f1191 on br-lan: ok 2601:646:c280:313::ef7/128 fd40:fa0a:7966::ef7/128 
Mon Nov 21 13:35:00 2016 cron.info crond[3663]: USER root pid 10166 cmd /sbin/fan_ctrl.sh
Mon Nov 21 13:37:01 2016 daemon.info dnsmasq-dhcp[3621]: DHCPREQUEST(br-lan) 192.168.1.172 68:db:ca:8f:11:91 
Mon Nov 21 13:37:01 2016 daemon.info dnsmasq-dhcp[3621]: DHCPACK(br-lan) 192.168.1.172 68:db:ca:8f:11:91 Magalys-iPhone
Mon Nov 21 13:40:00 2016 cron.info crond[3663]: USER root pid 10912 cmd /sbin/fan_ctrl.sh
Mon Nov 21 13:41:35 2016 daemon.info dnsmasq-dhcp[3621]: DHCPREQUEST(br-lan) 192.168.1.172 68:db:ca:8f:11:91 
Mon Nov 21 13:41:35 2016 daemon.info dnsmasq-dhcp[3621]: DHCPACK(br-lan) 192.168.1.172 68:db:ca:8f:11:91 Magalys-iPhone
Mon Nov 21 13:41:36 2016 daemon.info dnsmasq-dhcp[3621]: DHCPREQUEST(br-lan) 192.168.1.172 68:db:ca:8f:11:91 
Mon Nov 21 13:41:36 2016 daemon.info dnsmasq-dhcp[3621]: DHCPACK(br-lan) 192.168.1.172 68:db:ca:8f:11:91 Magalys-iPhone
Mon Nov 21 13:43:59 2016 daemon.info hostapd: wlan0: STA c0:11:73:00:9d:73 WPA: group key handshake completed (RSN)
Mon Nov 21 13:43:59 2016 daemon.info hostapd: wlan0: STA ac:cf:85:b7:1a:3c WPA: group key handshake completed (RSN)
Mon Nov 21 13:44:07 2016 daemon.info hostapd: wlan0: STA 68:db:ca:8f:11:91 IEEE 802.11: deauthenticated due to local deauth request
Mon Nov 21 13:45:00 2016 cron.info crond[3663]: USER root pid 11658 cmd /sbin/fan_ctrl.sh
Mon Nov 21 13:48:24 2016 daemon.info hostapd: wlan0: STA d8:bb:2c:01:10:61 IEEE 802.11: authenticated
Mon Nov 21 13:48:24 2016 daemon.info hostapd: wlan0: STA d8:bb:2c:01:10:61 IEEE 802.11: associated (aid 1)
Mon Nov 21 13:48:24 2016 daemon.info hostapd: wlan0: STA d8:bb:2c:01:10:61 WPA: pairwise key handshake completed (RSN)
Mon Nov 21 13:50:00 2016 cron.info crond[3663]: USER root pid 12333 cmd /sbin/fan_ctrl.sh
Mon Nov 21 13:53:59 2016 daemon.info hostapd: wlan0: STA c0:11:73:00:9d:73 WPA: group key handshake completed (RSN)
Mon Nov 21 13:54:00 2016 daemon.info hostapd: wlan0: STA ac:cf:85:b7:1a:3c WPA: group key handshake completed (RSN)
Mon Nov 21 13:54:07 2016 daemon.info hostapd: wlan0: STA d8:bb:2c:01:10:61 IEEE 802.11: deauthenticated due to local deauth request
Mon Nov 21 13:55:00 2016 cron.info crond[3663]: USER root pid 13079 cmd /sbin/fan_ctrl.sh
Mon Nov 21 14:00:00 2016 cron.info crond[3663]: USER root pid 13767 cmd /sbin/fan_ctrl.sh
Mon Nov 21 14:03:59 2016 daemon.info hostapd: wlan0: STA c0:11:73:00:9d:73 WPA: group key handshake completed (RSN)
Mon Nov 21 14:03:59 2016 daemon.info hostapd: wlan0: STA ac:cf:85:b7:1a:3c WPA: group key handshake completed (RSN)
Mon Nov 21 14:05:00 2016 cron.info crond[3663]: USER root pid 14501 cmd /sbin/fan_ctrl.sh
Mon Nov 21 14:06:13 2016 daemon.info hostapd: wlan0: STA d8:bb:2c:01:10:61 IEEE 802.11: authenticated
Mon Nov 21 14:06:13 2016 daemon.info hostapd: wlan0: STA d8:bb:2c:01:10:61 IEEE 802.11: associated (aid 1)
Mon Nov 21 14:06:13 2016 daemon.info hostapd: wlan0: STA d8:bb:2c:01:10:61 WPA: pairwise key handshake completed (RSN)
Mon Nov 21 14:10:00 2016 cron.info crond[3663]: USER root pid 15247 cmd /sbin/fan_ctrl.sh
Mon Nov 21 14:13:59 2016 daemon.info hostapd: wlan0: STA c0:11:73:00:9d:73 WPA: group key handshake completed (RSN)
Mon Nov 21 14:14:00 2016 daemon.info hostapd: wlan0: STA ac:cf:85:b7:1a:3c WPA: group key handshake completed (RSN)
Mon Nov 21 14:14:07 2016 daemon.info hostapd: wlan0: STA d8:bb:2c:01:10:61 IEEE 802.11: deauthenticated due to local deauth request
Mon Nov 21 14:15:00 2016 cron.info crond[3663]: USER root pid 15919 cmd /sbin/fan_ctrl.sh
[... cut ...]
Mon Nov 21 14:33:59 2016 daemon.info hostapd: wlan0: STA ac:cf:85:b7:1a:3c WPA: group key handshake completed (RSN)
Mon Nov 21 14:33:59 2016 daemon.info hostapd: wlan0: STA c0:11:73:00:9d:73 WPA: group key handshake completed (RSN)
Mon Nov 21 14:35:00 2016 cron.info crond[3663]: USER root pid 18837 cmd /sbin/fan_ctrl.sh`

Using the latest code on mamba with every wireless device (Nexus 6p, Intel 7260 AC card, Nexus 5). LEDE LEDE Reboot r2289 / LuCI Master (git-16.324.51057-1c27f6b) kernel 4.4.32 with BQM patches

yuhhaurlin, 2016-11-22:
Please specify how to reproduce the problem.

anomeome, 2016-11-22:
@thagabe, did some more testing, same LEDE version & kernel, fruit device as client, no issues experienced. I am wondering what you use in your build, specifically wpad(mini)/hostapd(mini).

thagabe, 2016-11-22:
@yuhhaurlin 
Problem: Wifi clients will see the wifi ssid being broadcasted but will never connect to the wifi. Tested on Windows, Linux (same pc Intel 7260 AC card), and Android (Nexus 5, 6p, 2012 7). 

Reproduce:
1) flash .img via tftp then reset
2) reset config settings after reboot
3) set up wifi 2.4 ghz with 20mhz width, then 5ghz in 80mhz on channel 36 with wpa2-forced aes password
4) wait
5) try to connect devices (On windows it just stays at "connecting ..." then says there was an error connecting)

@anomeome mentioned that apple devices stay connected so i will try to reproduce this with a new build that will be barebones to rule out any extra packages (I use luci-openssl, sqm, hd-idle, etc)

To: @anomeome Let me link my build config maybe I have some leftover package that i shouldnt have. Just to be safe ill nuke my build system and start fresh. As for the wpad and hostpad i have not touched those. I'll test with my 2010 iMac and see if I see any results.

yuhhaurlin, 2016-11-22:
1. If your device I'd not WRT3200ACM, the driver is the same as before.
2. When we release this driver, we had done tests with a lot of clients. We did not find connection problem.

thagabe, 2016-11-23:
@yuhhaurlin Unfortunately 1) does not seem to hold up and 2) this is WRT1900ACv1 (mamba)

yuhhaurlin, 2016-11-23:
For WRT1900AC, the driver is the same as before.

thagabe, 2016-11-23:
@yuhhaurlin @anomeome Unreal everything seems to be working fine (even better thanks to the BQL patches) but I have stripped down my build to just luci-openssl, nano, iptables-nat-extra, and wget-ssl. I'll build again with the rest of my packages (maybe residual downloads had something to do with this). I'm marking this as closed. Thanks!

closed 2016-11-23 by thagabe