kaloz/mwlwifi #125: Strange performance using WPA2 TKIP Encryption

ark4n631, 2016-11-28:
I have a wrt1900AC, i installed (dd-wrt, openwrt CC, DD, LEDE, etc) with all this firmwares, if you put encryption wpa2 and FORCE TKIP the connection (even in 5g) drop to 54mbps max and real connection output to internet give you 20mpbs at max.

I tested almost every firmware and took me a lot of time and effort saw this strange behavior.

If you want more info i can attach you the logs, and speedtest.

Thanks in advance and good luck

yuhhaurlin, 2016-11-29:
This is correct behavior. WiFi device does not use HT rates when using TKIP as the encryption cipher. If you want to have good performance, please use AES only. Thanks.

closed 2016-11-29 by yuhhaurlin

ark4n631, 2016-11-29:
@yuhhaurlin Thanks for answering, could you please put that info on the Readme, i could not know that info, it might help others in the future.

Thank you.

yuhhaurlin, 2016-11-30:
Done.