kaloz/mwlwifi #126: Clients dropped and unable to re-connect

davidc502, 2016-11-29:
This issue has happened twice where clients are dropped from wifi and are unable to reconnect.

Wifi Driver version 10.3.2.0-20161124   Lede 	4.4.32  Linksys 1900acs 

At the time we had a Amazon FireTV box streaming a video from my media server, and a iphone 7 on facebook at the time.

Ran down stairs and pulled the log once I noticed it had happened.

[61054.184304] ieee80211 phy0: cmd 0x9125=BAStream timed out
[61054.189743] ieee80211 phy0: return code: 0x1125
[61054.194292] ieee80211 phy0: timeout: 0x1125
[61054.198503] ieee80211 phy0: destroy ba failed execution
[61178.090217] ieee80211 phy0: cmd 0x801d=MEMAddrAccess timed out
[61178.096086] ieee80211 phy0: return code: 0x001d
[61178.100634] ieee80211 phy0: timeout: 0x001d
[61178.104839] ieee80211 phy0: failed execution
[61182.128148] ieee80211 phy0: cmd 0x801d=MEMAddrAccess timed out
[61182.134032] ieee80211 phy0: return code: 0x001d
[61182.138618] ieee80211 phy0: timeout: 0x001d
[61182.142828] ieee80211 phy0: failed execution
[61186.148064] ieee80211 phy0: cmd 0x801d=MEMAddrAccess timed out
[61186.153935] ieee80211 phy0: return code: 0x001d
[61186.158483] ieee80211 phy0: timeout: 0x001d
[61186.162690] ieee80211 phy0: failed execution

cybrnook, 2016-11-30:
I will confirm the same behavior on a WRT1900ACS(V1). 

There is also one other report of this in david's thread:
https://forum.openwrt.org/viewtopic.php?id=64949&p=24

Post 580 from markm12345

yuhhaurlin, 2016-11-30:
@davidc502 Can you let me know which commit of LEDE you used to build the code? I will try to build the code and test here. Thanks.

davidc502, 2016-11-30:
r2321    / LuCI Master (git-16.330.55844-ebb5a90)

It was from the 25th of November right after the commit to revert the mvebu simplify board.d.

thagabe, 2016-11-30:
@davidc502 I have built lede with the kernel 4.4.35 PR along with the PR for the latest MWLWIFI to fix this issue. I have had this issue on .32 and the newest mwlwifi code but it seems like there were residual builds that might have something to do with it; I took the advice of another dev and started from scratch by creating a new directory and rebuilt the buildtools, everything seems good for the moment (fruity devices, android, windows, linux)

yuhhaurlin, 2016-11-30:
@thagabe Can you share your image to davidc502?

yuhhaurlin, 2016-11-30:
@davidc502 I get latest code from "https://git.lede-project.org/source.git", but the driver is not the latest one.

davidc502, 2016-11-30:
I'll give it a try and report back the finding.

miguel-orange, 2016-12-03:
Thanks for bringing up this topic @davidc502. I'm using your build and getting the same issues. It occurs on both the WRT1900AC v1 and WRT1900ACS. I've noticed that the ACS doesn't drop as frequent as v1 and when streaming also is a culprit. It's only on the 5g side for me. I'll have my phone on the 2.5g and reboot the router when the 5g acts up.

thagabe, 2016-12-03:
@yuhhaurlin @davidc502 Here is my image running just fine with all my devices https://drive.google.com/open?id=0B0oFVQPGy6B7Qlk3ZjR4STRUMms

davidc502, 2016-12-04:
I've tried several things and they have all failed. However, I'm starting to think this is not a wifi driver issue. The reason is, I even tried 10.3.2.0-20161011 committed October 4, on the latest LEDE commits, and get the same errors. I've also tried building from scratch, so nothing was left behind, but the problem persists with both old and new drivers on the latest LEDE.

It's easy to bring it to failure because all you have to do is start downloading from more than one device at the same time and after about 5 minutes or so it fails.

Thoughts?

yuhhaurlin, 2016-12-04:
Can you let me know how to build the image? I will try to build the image and check here.

anomeome, 2016-12-04:
https://forum.lede-project.org/t/fs-284-wrt1900acs-wifi-stops-workign-under-havy-load/283/3

davidc502, 2016-12-04:
There's some preparation but after it's done, it's simply choosing targets and compiling.  Right now, I just have 1 patch to fix nslookup, but it sin't required to test with. 

The following instructions will get you compiling right away.  https://forum.openwrt.org/viewtopic.php?pid=343023#p343023

Currently this is the driver version with LEDE sources - 10.3.2.0-20161011

yuhhaurlin, 2016-12-04:
It is the way I build the LEDE image. I will check it.

davidc502, 2016-12-04:
Latest commit with kernel 4.4.35 seem to be working fine as it passed my wifi load test.  Additional testing to follow.

cybrnook, 2016-12-04:
I am also on a build with the 1011 driver on the latest 4.4.35 kernel from head, and all seems to be okay since last night.

yuhhaurlin, 2016-12-05:
All right. Please let me know if I should check this issue. Thanks.

davidc502, 2016-12-06:
I believe this issue is over with the latest commit as I've had zero problems.  Thanks for looking into it though.

closed 2016-12-06 by davidc502

kubrickfr, 2017-01-05:
This is still happening, with kernel 4.4.39 on a WRT3200ACM using @davidc502 latest build for this device.

```
Thu Jan  5 21:19:44 2017 kern.err kernel: [30177.679624] ieee80211 phy0: cmd 0x801d=MEMAddrAccess timed out
Thu Jan  5 21:19:44 2017 kern.err kernel: [30177.685500] ieee80211 phy0: return code: 0x001d
Thu Jan  5 21:19:44 2017 kern.err kernel: [30177.690052] ieee80211 phy0: timeout: 0x001d
Thu Jan  5 21:19:44 2017 kern.err kernel: [30177.694261] ieee80211 phy0: failed execution
Thu Jan  5 21:19:48 2017 kern.err kernel: [30181.697983] ieee80211 phy0: cmd 0x801d=MEMAddrAccess timed out
Thu Jan  5 21:19:48 2017 kern.err kernel: [30181.703853] ieee80211 phy0: return code: 0x001d
Thu Jan  5 21:19:48 2017 kern.err kernel: [30181.708401] ieee80211 phy0: timeout: 0x001d
Thu Jan  5 21:19:48 2017 kern.err kernel: [30181.712612] ieee80211 phy0: failed execution
Thu Jan  5 21:19:52 2017 kern.err kernel: [30185.721951] ieee80211 phy0: cmd 0x801d=MEMAddrAccess timed out
Thu Jan  5 21:19:52 2017 kern.err kernel: [30185.727812] ieee80211 phy0: return code: 0x001d
Thu Jan  5 21:19:52 2017 kern.err kernel: [30185.732370] ieee80211 phy0: timeout: 0x001d
Thu Jan  5 21:19:52 2017 kern.err kernel: [30185.736573] ieee80211 phy0: failed execution
Thu Jan  5 21:19:56 2017 kern.err kernel: [30189.739900] ieee80211 phy0: cmd 0x801d=MEMAddrAccess timed out
Thu Jan  5 21:19:56 2017 kern.err kernel: [30189.745767] ieee80211 phy0: return code: 0x001d
Thu Jan  5 21:19:56 2017 kern.err kernel: [30189.750321] ieee80211 phy0: timeout: 0x001d
Thu Jan  5 21:19:56 2017 kern.err kernel: [30189.754550] ieee80211 phy0: failed execution
```
...

miguel-orange, 2017-01-05:
@kubrickfr,
I've been using the latest build posted on 12/28 from @davidc502 and it's working well for me.
Did you make sure your settings didn't persist to your last flash? I also make sure that my opkg are up to date after flashing.

cybrnook, 2017-01-05:
Also posts of this today from my thread (3200)

https://forum.lede-project.org/t/community-snapshots-for-wrt1200ac-wrt1900acv1-wrt1900acv2-wrt1900acs-wrt3200acm/545/122?u=cybrnook

yuhhaurlin, 2017-01-05:
WRT3200ACM has problem.

kubrickfr, 2017-01-05:
This is actually the very first time I flash this device, having only received it yesterday. So no old configuration / packages. 

On 5 Jan 2017, Miguel Amaro <notifications@github.com> wrote:
>@kubrickfr,
>I've been using the latest build posted on 12/28 from @davidc502 and
>it's working well for me.
>Did you make sure your settings didn't persist to your last flash? I
>also make sure that my opkg are up to date after flashing.

-- Sent with K-@ Mail - the evolution of emailing.

kubrickfr, 2017-01-06:
After the issue occurring again an again, I tried putting the wifi down and unloading the mwlwifi kernel module, this is the log output it produces:
[syslog.txt](https://github.com/kaloz/mwlwifi/files/689947/syslog.txt)
And the modprobe hanged forever, I ended up rebooting.
The only things "specific" about my configuration are:
- 2.4GHz disabled
- 2 SSID on the 5GHz, each on a separate VLAN
- setting region to GB (but that actually doesn't get applied to the phy for some reason...)

yuhhaurlin, 2017-01-06:
I already told you WRT3200ACM has problem. Thanks.

kubrickfr, 2017-01-06:
@yuhhaurlin sorry I didn't understand it meant "It's a known issue and we don't need any extra feedback". Do you know what the problems are? Anything I could provide to help fixing these problems?

yuhhaurlin, 2017-01-06:
It needs time to let WRT3200ACM work normally.

thagabe, 2017-01-06:
@kubrickfr Basically the team at linksys is working on a new driver that will, hopefully, fix the issue. For the moment just hang tight and when the next driver comes around then you can test and provide feedback.

sorinello, 2017-11-27:
This issue still happens, I think it should be reopened.

yuhhaurlin, 2017-11-27:
You mean "host command timeout"? Can you try latest commit?

sorinello, 2017-11-27:
I get a ton of these 

`
Mon Nov 27 22:38:16 2017 kern.err kernel: [109823.812420] ieee80211 phy0: return code: 0x051d
Mon Nov 27 22:38:16 2017 kern.err kernel: [109823.817114] ieee80211 phy0: timeout: 0x001d
Mon Nov 27 22:38:20 2017 kern.err kernel: [109827.825519] ieee80211 phy0: cmd 0x801d=MEMAddrAccess timed out
Mon Nov 27 22:38:20 2017 kern.err kernel: [109827.831515] ieee80211 phy0: return code: 0x001d
Mon Nov 27 22:38:20 2017 kern.err kernel: [109827.836213] ieee80211 phy0: timeout: 0x001d
Mon Nov 27 22:38:24 2017 kern.err kernel: [109831.839482] ieee80211 phy0: cmd 0x801d=MEMAddrAccess timed out
Mon Nov 27 22:38:24 2017 kern.err kernel: [109831.845713] ieee80211 phy0: return code: 0x001d
Mon Nov 27 22:38:24 2017 kern.err kernel: [109831.850491] ieee80211 phy0: timeout: 0x001d
Mon Nov 27 22:38:24 2017 daemon.notice hostapd: nl80211: nl80211_recv_beacons->nl_recvmsgs failed: -5
Mon Nov 27 22:38:28 2017 kern.err kernel: [109835.855763] ieee80211 phy0: cmd 0x801d=MEMAddrAccess timed out
Mon Nov 27 22:38:28 2017 kern.err kernel: [109835.861818] ieee80211 phy0: return code: 0x001d
Mon Nov 27 22:38:28 2017 kern.err kernel: [109835.866458] ieee80211 phy0: timeout: 0x001d
Mon Nov 27 22:38:32 2017 kern.err kernel: [109839.869381] ieee80211 phy0: cmd 0x801d=MEMAddrAccess timed out
Mon Nov 27 22:38:32 2017 kern.err kernel: [109839.875360] ieee80211 phy0: return code: 0x001d
Mon Nov 27 22:38:32 2017 kern.err kernel: [109839.880073] ieee80211 phy0: timeout: 0x001d
Mon Nov 27 22:38:37 2017 kern.err kernel: [109843.902133] ieee80211 phy0: cmd 0x801d=MEMAddrAccess timed out
Mon Nov 27 22:38:37 2017 kern.err kernel: [109843.908120] ieee80211 phy0: return code: 0x001d
Mon Nov 27 22:38:37 2017 kern.err kernel: [109843.912776] ieee80211 phy0: timeout: 0x001d
Mon Nov 27 22:38:41 2017 kern.err kernel: [109847.917346] ieee80211 phy0: cmd 0x9111=SetNewStation timed out
Mon Nov 27 22:38:41 2017 kern.err kernel: [109847.923407] ieee80211 phy0: return code: 0x1111
Mon Nov 27 22:38:41 2017 kern.err kernel: [109847.928047] ieee80211 phy0: timeout: 0x1111
Mon Nov 27 22:38:41 2017 daemon.info hostapd: wlan0: STA 28:be:03:0a:52:61 IEEE 802.11: deauthenticated due to local deauth request
Mon Nov 27 22:38:48 2017 kern.err kernel: [109855.121983] ieee80211 phy0: cmd 0x801d=MEMAddrAccess timed out
Mon Nov 27 22:38:48 2017 kern.err kernel: [109855.127987] ieee80211 phy0: return code: 0x001d
Mon Nov 27 22:38:48 2017 kern.err kernel: [109855.132627] ieee80211 phy0: timeout: 0x001d
Mon Nov 27 22:38:52 2017 kern.err kernel: [109859.137933] ieee80211 phy0: cmd 0x801d=MEMAddrAccess timed out
Mon Nov 27 22:38:52 2017 kern.err kernel: [109859.143895] ieee80211 phy0: return code: 0x001d
Mon Nov 27 22:38:52 2017 kern.err kernel: [109859.148554] ieee80211 phy0: timeout: 0x001d
Mon Nov 27 22:38:56 2017 kern.err kernel: [109863.151873] ieee80211 phy0: cmd 0x801d=MEMAddrAccess timed out
Mon Nov 27 22:38:56 2017 kern.err kernel: [109863.157841] ieee80211 phy0: return code: 0x001d
Mon Nov 27 22:38:56 2017 kern.err kernel: [109863.162531] ieee80211 phy0: timeout: 0x001d
Mon Nov 27 22:38:56 2017 daemon.notice hostapd: nl80211: nl80211_recv_beacons->nl_recvmsgs failed: -5
Mon Nov 27 22:39:00 2017 kern.err kernel: [109867.167809] ieee80211 phy0: cmd 0x801d=MEMAddrAccess timed out
Mon Nov 27 22:39:00 2017 kern.err kernel: [109867.173810] ieee80211 phy0: return code: 0x001d
Mon Nov 27 22:39:00 2017 kern.err kernel: [109867.178499] ieee80211 phy0: timeout: 0x001d
Mon Nov 27 22:39:04 2017 kern.err kernel: [109871.183768] ieee80211 phy0: cmd 0x801d=MEMAddrAccess timed out
Mon Nov 27 22:39:04 2017 kern.err kernel: [109871.189798] ieee80211 phy0: return code: 0x001d
Mon Nov 27 22:39:04 2017 kern.err kernel: [109871.194502] ieee80211 phy0: timeout: 0x001d
Mon Nov 27 22:39:08 2017 kern.err kernel: [109875.203423] ieee80211 phy0: cmd 0x801d=MEMAddrAccess timed out
Mon Nov 27 22:39:08 2017 kern.err kernel: [109875.209420] ieee80211 phy0: return code: 0x001d
Mon Nov 27 22:39:08 2017 kern.err kernel: [109875.214060] ieee80211 phy0: timeout: 0x001d
Mon Nov 27 22:39:13 2017 kern.err kernel: [109880.797629] ieee80211 phy0: cmd 0x801d=MEMAddrAccess timed out
Mon Nov 27 22:39:13 2017 kern.err kernel: [109880.803660] ieee80211 phy0: return code: 0x001d
Mon Nov 27 22:39:13 2017 kern.err kernel: [109880.808370] ieee80211 phy0: timeout: 0x001d
Mon Nov 27 22:39:17 2017 kern.err kernel: [109884.811609] ieee80211 phy0: cmd 0x801d=MEMAddrAccess timed out
Mon Nov 27 22:39:17 2017 kern.err kernel: [109884.817574] ieee80211 phy0: return code: 0x001d
Mon Nov 27 22:39:17 2017 kern.err kernel: [109884.822209] ieee80211 phy0: timeout: 0x001d
Mon Nov 27 22:39:21 2017 kern.err kernel: [109888.833528] ieee80211 phy0: cmd 0x801d=MEMAddrAccess timed out
Mon Nov 27 22:39:21 2017 kern.err kernel: [109888.839522] ieee80211 phy0: return code: 0x001d
Mon Nov 27 22:39:21 2017 kern.err kernel: [109888.844224] ieee80211 phy0: timeout: 0x001d
Mon Nov 27 22:39:25 2017 kern.err kernel: [109892.855481] ieee80211 phy0: cmd 0x801d=MEMAddrAccess timed out
Mon Nov 27 22:39:25 2017 kern.err kernel: [109892.861446] ieee80211 phy0: return code: 0x001d
Mon Nov 27 22:39:25 2017 kern.err kernel: [109892.866081] ieee80211 phy0: timeout: 0x001d
Mon Nov 27 22:39:29 2017 kern.err kernel: [109896.871430] ieee80211 phy0: cmd 0x801d=MEMAddrAccess timed out
Mon Nov 27 22:39:29 2017 kern.err kernel: [109896.877431] ieee80211 phy0: return code: 0x001d
[... cut ...]
Mon Nov 27 22:46:29 2017 kern.err kernel: [110316.041678] ieee80211 phy0: cmd 0x801d=MEMAddrAccess timed out
Mon Nov 27 22:46:29 2017 kern.err kernel: [110316.047705] ieee80211 phy0: return code: 0x001d
Mon Nov 27 22:46:29 2017 kern.err kernel: [110316.052418] ieee80211 phy0: timeout: 0x001d
`

I am on 466368f9454250c2bc024795600d92564553d9bb, the "Free station id after firmware deletes it. " one. It happened randomly, I am monitoring the situation after the reboot see if it reappears.

If another issue must be created, please tell me and I will report one.

yuhhaurlin, 2017-11-27:
Please wait for issue 232 is fixed, then help to test the code. If you still encounter, you can create an issue to check it.