kaloz/mwlwifi #127: Add support for 802.11s

belzebub40k, 2016-12-04:
Is it still planned to add support for 802.11s? All I could find regarding this feature is a post on mailinglist that is one and a half year old.

```
> On 06/06/2015 6:01 AM, Alexis Green wrote:
> Are there plans to add support for 802.11s mesh point mode?
> 
Yes, it will be worked on in the future.

Thanks,
David
```
https://www.spinics.net/lists/linux-wireless/msg137528.html

yuhhaurlin, 2016-12-05:
Previous working is pending (VHT is not supported by mac80211 at that time). There is no plan now. Thanks.

closed 2016-12-05 by yuhhaurlin

linuskallander, 2017-12-13:
Mesh networks are becoming big. Still no plans to implement 802.11s?

yuhhaurlin, 2017-12-13:
It is done for another chip. The code will be merged here later. However there is no plan for 8864 and 8964.

linuskallander, 2017-12-14:
That's to bad... I hoped my linksys wrt1900acs would be future proof, but it seems I was wrong. (if I'm reading the specs correctly)  :(

yuhhaurlin, 2017-12-14:
Where did you get the specs? I don't think that 802.11s is supported by wrt1900acs. BTW, WEB GUI of LEDE has problem to set mesh interface. It needs patches to achieve mesh setting. I wonder which product supported by LEDE has supported 802.11s now?

linuskallander, 2017-12-14:
I found them here https://wiki.openwrt.org/toh/hwdata/linksys/linksys_wrt1900acs_v1

yuhhaurlin, 2017-12-14:
I can't find the support for 802.11s.

linuskallander, 2017-12-14:
No, put of the box it's not supported. I hoped the community would take care of that. But I realize I had to high hopes. Maybe it's not even technically possible?

linuskallander, 2017-12-14:
*out of the box

yuhhaurlin, 2017-12-14:
Mwlwifi has added code to support 802.11s and it is still under verification on another chip. I think this feature will be supported by our next chip (11ax).

s-pimenta, 2017-12-14:
The chip does need to support that protocol, right? So the implementation will be impossible unless the chipset support's it..

BrainSlayer, 2017-12-14:
unlikelly. i dont think its chipset related, but firmware related since the wifi protocol stack based functions are handled in the firmware.

s-pimenta, 2017-12-14:
So... after implementing WDS, MU-MIMO, monitor mode, and next should be 802.11s... at least for 88W8964 (WRT3200ACM) it has only almost 1 year of existence

thagabe, 2017-12-14:
@s-pimenta 
@yuhhaurlin said "there is no plan for 8864 and 8964." which is truly disappointing

s-pimenta, 2017-12-14:
@thagabe Yeah I now... companies has objectives, new products are released and will shift focus to newer produtcs to make profit... they are not interessted in supporting the actual customers... 

This is like Android devices... 2 years max of updates... unless you install a custom ROM from the XDA community...

Unless @yuhhaurlin releases the documentation of the 88W8964, we are at the mercy of what support we can get from Marvell...

yuhhaurlin, 2017-12-14:
It is not trivial for SC4 to support mesh. BTW, I wonder which device supported by LEDE or dd-WRT supports it.

belzebub40k, 2017-12-14:
@yuhhaurlin Unfortunately there aren't many platforms that support 802.11s. Most devices we use to build our community mesh networks in Germany are ar71xx-*. Some ramips devices also have support for 802.11s but the connections aren't very stable.

BrainSlayer, 2017-12-14:
@yuhhaurlin QCA chipsets and mediatek

s-pimenta, 2017-12-14:
@yuhhaurlin Mesh networks is still relatively new, especialy in home use... so we need to start somewere...

**ath9k** and the mighty **WRT54G** is still used because of open documentation...

yuhhaurlin, 2017-12-14:
Thanks for all your information. If some devices already support it, why the WEB GUI of LEDE still has problem to set mesh interface?

yuhhaurlin, 2017-12-14:
Even though document is open, I don't think you can work on this driver without source code of firmware.