kaloz/mwlwifi #128: 10.3.2.0-20161124 driver on WRT1900ACSv1 - Throughput issues with SQM enabled

cybrnook, 2016-12-14:
I noticed a rather large difference in throughput over 5GHZ when building source lede against the latest 10.3.2.0-20161124 driver, while having cake SQM enabled (cake , piece.of.cake).

10.3.2.0-20161124 driver gave results anywhere from 13Mbps to 105 Mbps (on a 125 Mbps line)

10.3.2.0-20161013 driver gave consistent performance of 105 Mbps

Cake limits are set to 122000 down and 22000 up.

yuhhaurlin, 2016-12-15:
If you do not use WRT3200ACM, driver is almost the same between 10.3.0.2-20161011 and 10.3.0.2-20161124.

jbsky, 2016-12-15:
Hi, I have the same problem.
Sometime, throughtput up around 400Mbit/s( theoricaly 866Mbit/s) and sometime 200Mbit/s  or less as 10MBits/s only.
Alone is the 36 channel...

cybrnook, 2016-12-15:
It's the strangest thing... However it was instantly noticed, and falling back to previous driver on the same kernel immediately resolved the issue.

I have the build here if you want to test it on a WRT1900ASCv1. It's in the folder labeled "NEW DRIVER"

https://www.dropbox.com/sh/2a7hkorqir0ch5f/AACq8amz9Yp3nF8K-4O2Si_2a?dl=0

yuhhaurlin, 2016-12-15:
Can you give me iperf test result?

yuhhaurlin, 2016-12-15:
BTW, you can build these two version of drivers with the same version of LEDE as package and use opkg to swap them on the same version of LEDE  with the same setting to check if the problem only happened on latest driver.

cybrnook, 2016-12-15:
Well I have two builds on the same LEDE but different driver in the link above.

yuhhaurlin, 2016-12-15:
So you are sure these two different drivers running on the same version of LIFE with same setting on the same environment and only latest driver has this problem?

If yes, please let me know your setting and please give me iperf test result. Thanks.

cybrnook, 2016-12-15:
That is correct.

I will work this weekend on getting you the info you need, stay tuned...

Chadster766, 2016-12-15:
I've started an iperf3 two hour continuous test of the latest mwlwifi driver on a WRT3200ACM. Once complete I will report results.

yuhhaurlin, 2016-12-15:
WRT3200ACM has problem.

yuhhaurlin, 2016-12-15:
_

thagabe, 2016-12-16:
@yuhhaurlin I shall start my test soon as I too experience slow downs on the new driver compared to the old driver on a WRT1900ACv1 and on more than just sqm cake (as i was running fq_codel not sch_cake until recent events told me cake was a kernel module haha)

Chadster766, 2016-12-16:
After about an hour and 20 minutes the performance drops under iPerf3 "iperf3 -c <wrt3200-ip-address> -t 7200" testing

```
[  5] 4910.00-4911.00 sec  10.7 MBytes  89.5 Mbits/sec
[  5] 4911.00-4912.00 sec  10.1 MBytes  85.0 Mbits/sec
[  5] 4912.00-4913.00 sec  8.88 MBytes  74.5 Mbits/sec
[  5] 4913.00-4914.00 sec  7.84 MBytes  65.6 Mbits/sec
[  5] 4914.00-4915.00 sec  9.09 MBytes  76.4 Mbits/sec
[  5] 4915.00-4916.00 sec  9.89 MBytes  83.0 Mbits/sec
[  5] 4916.00-4917.00 sec  9.77 MBytes  82.0 Mbits/sec
[  5] 4917.00-4918.00 sec  10.5 MBytes  88.0 Mbits/sec
[  5] 4918.00-4919.00 sec  10.2 MBytes  85.6 Mbits/sec
[  5] 4919.00-4920.00 sec  9.25 MBytes  77.6 Mbits/sec
[  5] 4920.00-4921.00 sec  7.91 MBytes  66.3 Mbits/sec
[  5] 4921.00-4922.00 sec  8.01 MBytes  67.2 Mbits/sec
[  5] 4922.00-4923.00 sec  7.47 MBytes  62.6 Mbits/sec
[  5] 4923.00-4924.00 sec  7.09 MBytes  59.5 Mbits/sec
[  5] 4924.00-4925.00 sec  9.06 MBytes  76.0 Mbits/sec
[  5] 4925.00-4926.00 sec  7.79 MBytes  65.4 Mbits/sec
[  5] 4926.00-4927.00 sec  7.59 MBytes  63.7 Mbits/sec
[  5] 4927.00-4928.00 sec  6.90 MBytes  57.8 Mbits/sec
[  5] 4928.00-4929.00 sec  7.56 MBytes  63.5 Mbits/sec
[  5] 4929.00-4930.00 sec  6.35 MBytes  53.3 Mbits/sec
[  5] 4930.00-4931.00 sec  6.06 MBytes  50.9 Mbits/sec
[  5] 4931.00-4932.00 sec  6.30 MBytes  52.9 Mbits/sec
[  5] 4932.00-4933.00 sec  7.38 MBytes  61.9 Mbits/sec
[  5] 4933.00-4934.00 sec  6.81 MBytes  57.1 Mbits/sec
[  5] 4934.00-4935.01 sec  4.12 MBytes  34.4 Mbits/sec
[  5] 4935.01-4936.01 sec  3.28 MBytes  27.4 Mbits/sec
[  5] 4936.01-4937.01 sec  2.57 MBytes  21.6 Mbits/sec
[  5] 4937.01-4938.01 sec   307 KBytes  2.51 Mbits/sec
[  5] 4938.01-4939.01 sec   468 KBytes  3.83 Mbits/sec
[  5] 4939.01-4940.01 sec  54.2 KBytes   444 Kbits/sec
[  5] 4940.01-4941.00 sec   323 KBytes  2.67 Mbits/sec
[  5] 4941.00-4942.00 sec  1.03 MBytes  8.65 Mbits/sec
[  5] 4942.00-4943.00 sec  1.16 MBytes  9.72 Mbits/sec
[  5] 4943.00-4944.00 sec  1.80 MBytes  15.1 Mbits/sec
[  5] 4944.00-4945.00 sec  1.53 MBytes  12.8 Mbits/sec
[  5] 4945.00-4946.00 sec  1.71 MBytes  14.3 Mbits/sec
[  5] 4946.00-4947.00 sec  1.61 MBytes  13.5 Mbits/sec
[  5] 4947.00-4948.00 sec  1.63 MBytes  13.7 Mbits/sec
[  5] 4948.00-4949.00 sec  1.60 MBytes  13.4 Mbits/sec
[  5] 4949.00-4950.00 sec  1.56 MBytes  13.1 Mbits/sec
[  5] 4950.00-4951.00 sec  1.69 MBytes  14.2 Mbits/sec
[  5] 4951.00-4952.00 sec  1.44 MBytes  12.1 Mbits/sec
[  5] 4952.00-4953.00 sec  1.78 MBytes  14.9 Mbits/sec
[  5] 4953.00-4954.00 sec  1.44 MBytes  12.1 Mbits/sec
```

yuhhaurlin, 2016-12-16:
WRT3200ACM has problem. There is no need to test it and show me test result.

Chadster766, 2016-12-16:
Ok :smile:

yuhhaurlin, 2016-12-16:
@cybrnook If you can build driver and swap them, I suggest that you can base on following commits to build the driver and check which commit makes the problem happened:

https://github.com/kaloz/mwlwifi/commit/57ecd81fe3d407209b261a98ca7cbfbbed363a75
https://github.com/kaloz/mwlwifi/commit/2aca3db526e7c37725648a13b7dd6b7f1f7049bc
https://github.com/kaloz/mwlwifi/commit/90e99a735923a92a770afe3d84a2e4b834e010c0

Thanks.

thagabe, 2016-12-16:
@yuhhaurlin havent tested but I have finish compiling a full image and the kmods for the different wifi drivers to help out fix this regression. They can be found here: tinyurl.com/lifehacksback-testing-cc The based image is the driver from lede for WRT1900ACv1. The rest are just kmods that can be installed on any kernel.

yuhhaurlin, 2016-12-16:
@thagabe Thanks.
@cybrnook Please check if you can leverage thagabe's full image and driver packages to do test. If yes, let me know which commit will make this problem happened. Thanks.

cybrnook, 2016-12-16:
Perhaps @thegabe could test in the meantime? He compiled for ACv1, I am currently using an ACSv1, so our platforms are different. However, I think our findings are the same.

Chadster766, 2016-12-16:
Yesterday's release of Linux 4.8.15 resolved the performance drop issue.

#118

cybrnook, 2016-12-16:
@Chadster766 , I believe you are looking at performance drops for the WRT3200ACM. We seeing erratic performance issues on ACv1, ACSv1.

yuhhaurlin, 2016-12-16:
@Chadster766 Please don't mix problem of WRT3200ACM with this issue. And I don't think it is fixed.

Chadster766, 2016-12-16:
I will also test the WRT1900AC V2 (same hardware as the ACS) WRT1200AC V1 and WRT1900AC V1.

yuhhaurlin, 2016-12-16:
@cybrnook  Wait for your test result. Only few commits between these two versions of driver.

cybrnook, 2016-12-17:
I can confirm with @Chadster766 , performance is back to being on par with latest .38 kernel, with SQM enabled.

Closing this so that we may proceed.

closed 2016-12-17 by cybrnook