kaloz/mwlwifi #129: Noise not reporting in latest driver

cybrnook, 2016-12-18:
Noise values are no longer being reported in the latest driver. 

https://github.com/openwrt/luci/issues/885#event-897928359

robd3, 2016-12-19:
Maybe related: in commit ab05f992f7b7c6dccbebcda371096adfec97b323
a "noise" variable was removed:

 -	s8 noise;                    /* Most recently reported noise in dBm */
+	/* keep survey information */
+	u32 time_period;
+	u32 time_busy;
+	u32 time_tx;

starcms, 2016-12-19:
@robd3, that's gotta be it!  In that same commit, the following was also removed (in addition to the noise variable in dev.h):

first from dev.h (as @robd3 pointed out):

-	s8 noise;                    /* Most recently reported noise in dBm */

from mac80211.c:

-	survey->filled = SURVEY_INFO_NOISE_DBM;
-	survey->noise = priv->noise;

and from rx.c:

-		priv->noise = -curr_hndl->pdesc->noise_floor;

cybrnook, 2016-12-19:
Ninja's you two are @robd3 and @starcms , I think you nailed it Rob.

Hopefully @yuhhaurlin can revert these changes moving forward in the driver.

osxest, 2016-12-19:
Can this relate to https://github.com/kaloz/mwlwifi/issues/116?

cybrnook, 2016-12-19:
Yep, seems to be the same culprit.

starcms, 2016-12-19:
I hope @yuhhaurlin will consider reverting these minor changes to get Noise working again. According to #116 , he said it was optional and apparently removed on purpose. I personally don't consider it optional; it's needed for Luci to display the SNR and fill in all the info for Associated Stations at Network->>Wireless, not to mention the Graphs and Realtime Graphs.

yuhhaurlin, 2016-12-20:
Ｉwill add it back.

BrainSlayer, 2016-12-20:
@starcms  yes that might be true. i have all other devices as well but havent reviewed them again after i found this issue. but would be good if the "noise" feature can be fixed on wrt3200acm as well

starcms, 2016-12-20:
@yuhhaurlin Thanks so much!!

@BrainSlayer I agree, in the long run I'm sure all the issues with the 3200acm will be sorted out. But it looks to be very quick and easy for @yuhhaurlin to get Noise back up and running on all the other models for the time being. Plus, on the 3200acm, it'd be better for now to show -95dBm than 0dBm for the Noise (the latter always resulting in a negative SNR).

yuhhaurlin, 2016-12-20:
Please check https://github.com/kaloz/mwlwifi/commit/bb70b40407cec7310f54d19b947bf3918222d794.

closed 2016-12-20 by yuhhaurlin

BrainSlayer, 2016-12-20:
@starcms -95 results in negative snr as well if you have a lower signal. -104 as default seem to be more optimal from my findings. (compared the signal and noise level with other chipsets to find this value)

starcms, 2016-12-20:
@yuhhaurlin Thanks again!

@BrainSlayer It's not hard coded to display -95 dBm on the 3200acs. It's just a bug that you had reported. For all the other models, Noise displays the correct value.

cybrnook, 2016-12-20:
@yuhhaurlin , thank you very much!

BrainSlayer, 2016-12-21:
@starcms but i have seen negative values. this is how i got notice of it. i compared also the same signal with a qca9984 device which showed the same rssi, but -104dbm instead of -95 like the 88w8964. since the rssi was something around -92 i got a snr of 3 and was wondering why  the connection was working. so i discovered that the noise level was totally bogus

starcms, 2016-12-21:
@BrainSlayer Did you not read what I said?  It displaying -95dBm on the 3200acm is a bug. My guess due to a bug in the 88W8964 firmware, which only Marvell can fix. On all other models (1200/1900ac/acs), the Noise value displayed is the actual/correct value. 

So this way, with the Noise value added back into the driver, it displays correctly on the other models (confirmed working again with 1220 driver). And on the 3200acm, it displays -95 dBm as opposed to 0 dBm as it was on the previous driver (1124) with the Noise value removed. 

In time, I'm sure it will be corrected on the 3200acm.

BrainSlayer, 2016-12-21:
@starcms : sure i readed it and i did not say anything different than you. i just told you what i discovered and how i discovered it. nothing else.