kaloz/mwlwifi #130: Region Code Mapping for 0x30 (EU)

con7, 2016-12-23:
Hi,

i'm running McDebian on my WRT1900ACS V2 (EU Version) with latest mwlwifi driver/firmware, but the 5GHz wifi wont work because of a non valid country code.
My firmware region code is 0x30, in the driver you defined "EU" for 0x30, which isn't a valid ISO3166 country code.

```
static const struct region_code_mapping regmap[] = {
	{"US", 0x10}, /* US FCC */
	{"CA", 0x20}, /* Canada */
	{"EU", 0x30}, /* ETSI   */
	{"ES", 0x31}, /* Spain  */
	{"FR", 0x32}, /* France */
	{"JP", 0x40}, /* Japan  */
	{"TW", 0x80}, /* Taiwan */
	{"AU", 0x81}, /* Australia */
	{"CN", 0x90}, /* China (Asia) */
};
```

CRDA wants to set country to EU, but this doesn't work because EU isn't included in db.txt / regulatory.bin.
If I add EU to db.txt and generate a new regulatory.bin, all is working fine.

Is it possible to change region code mapping for 0x30 to something valid?

The same issue already existed in an other marvell driver (https://patchwork.kernel.org/patch/7768381/)

Thanks!

lantis1008, 2016-12-26:
Unrelated to this issue, but just for my information.

Is that table now the only available country codes that are valid for the driver? Or can we still use the entire contents of regdb?
i.e. Is it now hardcoded?

yuhhaurlin, 2017-01-03:
Please check https://github.com/kaloz/mwlwifi/commits/master. Thanks.

closed 2017-01-03 by yuhhaurlin

con7, 2017-01-03:
Thanks @yuhhaurlin !

lantis1008, 2017-01-30:
@yuhhaurlin 
I have just received a device coded to region 0x10 (US). I'm in Australia. Is there anyway for me to change this?

yuhhaurlin, 2017-01-30:
Please check with Belkin. Thanks.

lantis1008, 2017-03-01:
@yuhhaurlin
Belkin and Linksys both have no suggestions. 
Are you aware of any alternative backdoors?

Chadster766, 2017-03-01:
See if you can RMA warranty swap the WRT.

Contact linksyscares@linksys.com they should be able to complete or start the RMA for you.

yuhhaurlin, 2017-03-01:
@lantis1008 There is no backdoor for it via driver. This must be modified via manufacture tool in order to comply with FCC rules.