kaloz/mwlwifi #131: How to compile new driver on OpenWrt CC.01

muronghan, 2016-12-26:
Normally I used SDK to compile the new wireless drivers (replaced the Makefile and take a shot). I could successfully get the compiling done. I even could upgrade the new driver on OpenWrt CC.01 (the old driver is 2016-1013). But when I reboot my router, WRT1200AC, the driver seems not be recognized. 

Or, I replaced the Makefile and run a full compiling. Compiling seemed successful, but when I enter into luci interface, the wireless option on menu was missed.

Strangely, I did the same on LEDE, but it looked good.

yuhhaurlin, 2017-01-03:
I just can give you information about mwlwifi driver:

Mwlwifi driver depends on backports package. Backports package is used to let updated cfg80211/mac80211 and related wireless drivers to work on previous version of kernel. For OpenWrt/Lede, mac80211 package is ported from backports package. If you can let updated mac80211 package running on CC, you can compile mwlwifi driver and let it work without any problems.

<<mac80211 package>>

Package: package/kernel/mac80211
Build directory: build_dir/target-xxxxx/linux-mvebu/compat-wireless-xxxx-xx-xx

closed 2017-01-03 by yuhhaurlin

muronghan, 2017-01-03:
@yuhhaurlin thank you for you information &happy new year!