kaloz/mwlwifi #135: mamba: 5Ghz does not start

thagabe, 2017-01-13:
- Build and compiled LEDE Reboot SNAPSHOT r2951-8935689 / LuCI Master (git-17.012.74170-de4bda6)
- mwlwifi info
```
root@lede:~# cat /sys/kernel/debug/ieee80211/phy0/mwlwifi/info

driver name: mwlwifi
chip type: 88W8864
hw version: 7
driver version: 10.3.2.0-20170110
firmware version: 0x0702091a
power table loaded from dts: yes
firmware region code: 0x0
mac address: 00:25:9c:13:1f:4d
2g: enable
5g: disable
antenna: 4 4
irq number: 90
iobase0: d1c80000
iobase1: d1e00000
tx limit: 768
rx limit: 64
ap macid support: 0000ffff
sta macid support: 00010000
macid used: 00000001
qe trigger number: 36967
```
- Syslog
```
Fri Jan 13 10:14:44 2017 kern.debug kernel: [   26.200669] ieee80211 phy1: regdomain: FCC
Fri Jan 13 10:14:44 2017 kern.debug kernel: [   26.204888] ieee80211 phy1: Channel: 36: 0x0 0x0 0xf
Fri Jan 13 10:14:44 2017 kern.debug kernel: [   26.209889] ieee80211 phy1: 8 8 8 8 8 8 8 8 9 9 9 9 9 9 9 9 
Fri Jan 13 10:14:44 2017 kern.debug kernel: [   26.215654] ieee80211 phy1: Channel: 40: 0x0 0x0 0xf
Fri Jan 13 10:14:44 2017 kern.debug kernel: [   26.220646] ieee80211 phy1: 8 8 8 8 9 9 9 9 9 9 9 9 9 9 9 9 
Fri Jan 13 10:14:44 2017 kern.debug kernel: [   26.226384] ieee80211 phy1: Channel: 44: 0x0 0x0 0xf
Fri Jan 13 10:14:44 2017 kern.debug kernel: [   26.231385] ieee80211 phy1: 8 8 8 8 9 9 9 9 9 9 9 9 9 9 9 9 
Fri Jan 13 10:14:44 2017 kern.debug kernel: [   26.237113] ieee80211 phy1: Channel: 48: 0x0 0x0 0xf
Fri Jan 13 10:14:44 2017 kern.debug kernel: [   26.242112] ieee80211 phy1: 8 8 8 8 9 9 9 9 9 9 9 9 9 9 9 9 
Fri Jan 13 10:14:44 2017 kern.debug kernel: [   26.247853] ieee80211 phy1: Channel: 52: 0x0 0x0 0xf
Fri Jan 13 10:14:44 2017 kern.debug kernel: [   26.252890] ieee80211 phy1: f f f f f f f f 12 12 12 12 12 12 12 12 
Fri Jan 13 10:14:44 2017 kern.debug kernel: [   26.259268] ieee80211 phy1: Channel: 56: 0x0 0x0 0xf
Fri Jan 13 10:14:44 2017 kern.debug kernel: [   26.264306] ieee80211 phy1: f f f f f f f f 12 12 12 12 12 12 12 12 
Fri Jan 13 10:14:44 2017 kern.debug kernel: [   26.270695] ieee80211 phy1: Channel: 60: 0x0 0x0 0xf
Fri Jan 13 10:14:44 2017 kern.debug kernel: [   26.275737] ieee80211 phy1: f f f f f f f f 12 12 12 12 12 12 12 12 
Fri Jan 13 10:14:44 2017 kern.debug kernel: [   26.282126] ieee80211 phy1: Channel: 64: 0x0 0x0 0xf
Fri Jan 13 10:14:44 2017 kern.debug kernel: [   26.287168] ieee80211 phy1: f f f f f f f f 12 12 12 12 12 12 12 12 
Fri Jan 13 10:14:44 2017 kern.debug kernel: [   26.293599] ieee80211 phy1: Channel: 100: 0x0 0x0 0xf
Fri Jan 13 10:14:44 2017 kern.debug kernel: [   26.298689] ieee80211 phy1: 14 14 14 14 14 14 14 14 14 14 14 14 14 14 14 14 
Fri Jan 13 10:14:44 2017 kern.debug kernel: [   26.305814] ieee80211 phy1: Channel: 104: 0x0 0x0 0xf
Fri Jan 13 10:14:44 2017 kern.debug kernel: [   26.310904] ieee80211 phy1: 14 14 14 14 14 14 14 14 14 14 14 14 14 14 14 14 
Fri Jan 13 10:14:44 2017 kern.debug kernel: [   26.318029] ieee80211 phy1: Channel: 108: 0x0 0x0 0xf
Fri Jan 13 10:14:44 2017 kern.debug kernel: [   26.323155] ieee80211 phy1: 14 14 14 14 14 14 14 14 14 14 14 14 14 14 14 14 
Fri Jan 13 10:14:44 2017 kern.debug kernel: [   26.330240] ieee80211 phy1: Channel: 112: 0x0 0x0 0xf
Fri Jan 13 10:14:44 2017 kern.debug kernel: [   26.335372] ieee80211 phy1: 14 14 14 14 14 14 14 14 14 14 14 14 14 14 14 14 
Fri Jan 13 10:14:44 2017 kern.debug kernel: [   26.342458] ieee80211 phy1: Channel: 116: 0x0 0x0 0xf
Fri Jan 13 10:14:44 2017 kern.debug kernel: [   26.347584] ieee80211 phy1: 14 14 14 14 14 14 14 14 14 14 14 14 14 14 14 14 
Fri Jan 13 10:14:44 2017 kern.debug kernel: [   26.354707] ieee80211 phy1: Channel: 120: 0x0 0x0 0xf
Fri Jan 13 10:14:44 2017 kern.debug kernel: [   26.359796] ieee80211 phy1: 14 14 14 14 14 14 14 14 14 14 14 14 14 14 14 14 
Fri Jan 13 10:14:44 2017 kern.debug kernel: [   26.366920] ieee80211 phy1: Channel: 124: 0x0 0x0 0xf
Fri Jan 13 10:14:44 2017 kern.debug kernel: [   26.372010] ieee80211 phy1: 14 14 14 14 14 14 14 14 14 14 14 14 14 14 14 14 
Fri Jan 13 10:14:44 2017 kern.debug kernel: [   26.379135] ieee80211 phy1: Channel: 128: 0x0 0x0 0xf
Fri Jan 13 10:14:44 2017 kern.debug kernel: [   26.384260] ieee80211 phy1: 14 14 14 14 14 14 14 14 14 14 14 14 14 14 14 14 
Fri Jan 13 10:14:44 2017 kern.debug kernel: [   26.391346] ieee80211 phy1: Channel: 132: 0x0 0x0 0xf
Fri Jan 13 10:14:44 2017 kern.debug kernel: [   26.396474] ieee80211 phy1: 14 14 14 14 14 14 14 14 14 14 14 14 14 14 14 14 
Fri Jan 13 10:14:44 2017 kern.debug kernel: [   26.403601] ieee80211 phy1: Channel: 136: 0x0 0x0 0xf
Fri Jan 13 10:14:44 2017 kern.debug kernel: [   26.408691] ieee80211 phy1: 14 14 14 14 14 14 14 14 14 14 14 14 14 14 14 14 
Fri Jan 13 10:14:44 2017 kern.debug kernel: [   26.415816] ieee80211 phy1: Channel: 140: 0x0 0x0 0xf
Fri Jan 13 10:14:44 2017 kern.debug kernel: [   26.420906] ieee80211 phy1: 14 14 14 14 14 14 14 14 14 14 14 14 14 14 14 14 
Fri Jan 13 10:14:44 2017 kern.debug kernel: [   26.428031] ieee80211 phy1: Channel: 149: 0x0 0x0 0xf
Fri Jan 13 10:14:44 2017 kern.debug kernel: [   26.433155] ieee80211 phy1: 16 16 16 16 14 14 14 14 15 15 15 15 14 14 14 14 
Fri Jan 13 10:14:44 2017 kern.debug kernel: [   26.440241] ieee80211 phy1: Channel: 153: 0x0 0x0 0xf
[... 64 lines cut ...]
Fri Jan 13 10:14:46 2017 daemon.notice netifd: radio1 (1518): wlan1: AP-ENABLED 
Fri Jan 13 10:14:46 2017 kern.debug kernel: [   28.552777] ieee80211 phy0: change: 0x40
Fri Jan 13 10:14:46 2017 daemon.err insmod: module is already loaded - act_ipt
Fri Jan 13 10:14:46 2017 kern.debug kernel: [   28.752804] ieee80211 phy0: change: 0x40
Fri Jan 13 10:14:46 2017 kern.debug kernel: [   28.952712] ieee80211 phy0: change: 0x40
Fri Jan 13 10:14:47 2017 kern.debug kernel: [   29.152677] ieee80211 phy0: change: 0x40
Fri Jan 13 10:14:47 2017 kern.debug kernel: [   29.392773] ieee80211 phy0: change: 0x40
Fri Jan 13 10:14:47 2017 kern.debug kernel: [   29.602700] ieee80211 phy0: change: 0x40
Fri Jan 13 10:14:47 2017 kern.debug kernel: [   29.882723] ieee80211 phy0: change: 0x40
Fri Jan 13 10:14:48 2017 user.notice : setting up led WAN
Fri Jan 13 10:14:48 2017 kern.debug kernel: [   30.124152] ieee80211 phy0: change: 0x40
Fri Jan 13 10:14:48 2017 daemon.notice netifd: radio0 (1517): 20/40 MHz operation not permitted on channel pri=11 sec=7 based on overlapping BSSes
Fri Jan 13 10:14:48 2017 kern.debug kernel: [   30.342728] ieee80211 phy0: change: 0x40
Fri Jan 13 10:14:48 2017 kern.debug kernel: [   30.442691] ieee80211 phy0: change: 0x100
Fri Jan 13 10:14:48 2017 daemon.notice netifd: radio0 (1517): Using interface wlan0 with hwaddr b4:75:0e:68:51:35 and ssid "Home's Network"
Fri Jan 13 10:14:48 2017 kern.debug kernel: [   30.574086] ieee80211 phy0: change: 0x100
Fri Jan 13 10:14:48 2017 kern.debug kernel: [   30.587297] ieee80211 phy0: change: 0x42
Fri Jan 13 10:14:48 2017 kern.info kernel: [   30.787070] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
Fri Jan 13 10:14:48 2017 kern.info kernel: [   30.793636] br-lan: port 2(wlan0) entered forwarding state
Fri Jan 13 10:14:48 2017 kern.info kernel: [   30.799206] br-lan: port 2(wlan0) entered forwarding state
Fri Jan 13 10:14:48 2017 daemon.notice netifd: radio0 (1517): wlan0: interface state HT_SCAN->ENABLED
Fri Jan 13 10:14:48 2017 daemon.notice netifd: radio0 (1517): wlan0: AP-ENABLED 
Fri Jan 13 10:14:48 2017 daemon.notice netifd: Network device 'wlan0' link is up
```
- 2.4 Ghz work great (80 mbps out of my 300 mbps that i get from isp [why so low?]) but 5 Ghz will not start from luci and when sending this command ```uci set wireless.@wifi-device[1].disabled=0; uci commit wireless; wifi``` then the wifi turns on but anything that is associated with it does not get internet.  One of the old logs said something to the tune of "hostpad cannot start radio1"

yuhhaurlin, 2017-01-16:
The way to get more debug information:
1. make menconfig
2. Nework->Minimum debug message priority: change 3 to 0 and save configuration.
3. hostapd-xxx/src/utils/wpa_debug.c:
    int wpa_debug_level = MSG_INFO ;  => change to MSG_DEBUG or MSG_EXCESSIVE.
4. make V=s package/network/services/hostapd/compile
5. move bin/packages/arm_cortex-a9_vfpv3/base/wpad_xxx.ipk to your target device
6. use "opkg install" to install this ipk to your target device.
7. Reboot and rerun the code, hostapd will dump more debug information.

Please also check configuration file of hostapd (/var/run/hostapd-phy0(1).conf).

I suggest that you disable 2.4g module and only run 5g mosule. Use the way described above to get more debug information and check configuration of hostapd to see what configuration to avoid hostapd to run on that module.

thagabe, 2017-01-16:
Sounds good I'll retest

thagabe, 2017-01-25:
Rebuilt and the problem is solved.

closed 2017-01-25 by thagabe

mrvltest, 2017-01-25:
You can use the way I described to check this kind of problem next time.

yuhhaurlin, 2017-01-25:
You can use the way I described to check this kind of problem if it happens again.

thagabe, 2017-01-26:
Sounds good! Just bought a 3200ACM, looking forward to the mwlwifi update for testing