kaloz/mwlwifi #136: Unbalanced and poor performance.

jbsky, 2017-01-14:
Hi,

Thanks to Chadster, Wrt1900ac run under debian. Today, I have made my own kernel and performance in globally is stable, but poor and unbalanced.

For exemple, from a Windows 10 client / Wireless Intel 7260AC (866Mbits/s), wrt kernel: 4.9.3 with lastest mwlwifi driver.

D:\Users\User>iperf3 -t 60 -c wrt  -P 8
[...]
[SUM]   0.00-60.00  sec  2.57 GBytes   368 Mbits/sec                  sender
[SUM]   0.00-60.00  sec  2.57 GBytes   368 Mbits/sec                  receiver

Calc:  368/866*100=42,5% of 866Mbits/s.

D:\Users\User>iperf3 -t 60 -c wrt  -P 8 -R
[...]
[SUM]   0.00-60.00  sec  1.33 GBytes   191 Mbits/sec  287             sender
[SUM]   0.00-60.00  sec  1.33 GBytes   190 Mbits/sec                  receiver

Calc:  190/866*100=21.9% of 866Mbits/s.

I have the objective to have the lightest core to obtain the best network performance for wrt1900ac-v1 (ex: without netfilter, vlan, dsa, ...), I can provide a .config and some script which help to compile the kernel. 

Calc:  368/866*100=42,5% of 866Mbits/s.

Can you do a better driver? Do you have you some advice?

However, wired connection is excellent! (>80% of 1Gbits/s)

Thank.

yuhhaurlin, 2017-01-16:
What is the number you got for stock firmware?

jbsky, 2017-01-16:
root@wrt:~# uname -a
Linux wrt 4.9.3 #1 SMP Fri Jan 13 22:40:50 CET 2017 armv7l GNU/Linux
root@wrt:~# modinfo mwlwifi
filename:       /lib/modules/4.9.3/kernel/drivers/net/wireless/marvell/mwlwifi/mwlwifi.ko
license:        GPL v2
author:         Marvell Semiconductor, Inc.
version:        10.3.2.0-20170110
description:    Marvell 802.11ac Wireless Network Driver
srcversion:     29C3BB49E0FD85791638A8E
alias:          pci:v000011ABd00002B40sv*sd*bc*sc*i*
alias:          pci:v000011ABd00002B38sv*sd*bc*sc*i*
alias:          pci:v000011ABd00002A55sv*sd*bc*sc*i*
depends:
intree:         Y
vermagic:       4.9.3 SMP mod_unload ARMv7 p2v8

Yesterday, I think that unbalanced is wrong, I must do a better monitoring but I'm sure that mwlwifi function to 50% of its capacity.

yuhhaurlin, 2017-01-16:
I want to know the throughput number for stock firmware under your setting and test environment. Thanks.

jbsky, 2017-01-16:
How do I find  throughput number for stock firmware ? 
I use Wrt1900ac-v1 under Debian, band 5Ghz channel 36 (1300MBits/s max) , maybe a config file for hostapd?
The client is under Windows-10 Pro- with a chip intel 7260AC (866MBits/s max)
[sys-kernel-debug-mwlwifi.txt](https://github.com/kaloz/mwlwifi/files/708033/sys-kernel-debug-mwlwifi.txt)

yuhhaurlin, 2017-01-16:
I mean the firmware you got from Belkin (the original one installed in the device).

jbsky, 2017-01-16:
I didn't pay attention to the original firmware. If i need to return to the stock firmware, I download this file http://cache-www.belkin.com/support/dl/FW_WRT1900AC_1.1.10.167514_prod.

yuhhaurlin, 2017-01-16:
Yes, please use same setting to get throughput number from stock firmware under same environment. Thanks.

jbsky, 2017-01-16:
I can't enter a shell in stock firmware (ex: ssh, serial, etc...) but I have a log of stock firmware.
[minicom-stockfirmware-wrt1900ac-1.8.txt](https://github.com/kaloz/mwlwifi/files/708174/minicom-stockfirmware-wrt1900ac-1.8.txt)
If I understand well, wrt must return to the stock firmware and after I run  iperf3 test. is-it   the right path?

yuhhaurlin, 2017-01-16:
You can test from LAN to/from WLAN.

jbsky, 2017-01-16:
Well, I can't do the test before Friday. 

Thanks for responding.

jbsky, 2017-01-20:
Hi,

I made a test today from Windows 10 client / Wireless Intel 7260AC (866Mbits/s) (lastest driver) with a wired PC (1Gbit)

D:\Tools\iperf3>iperf3.exe -c ipfire -P 6
Connecting to host ipfire, port 5201
[SUM]   0.00-10.00  sec   441 MBytes   370 Mbits/sec                  sender
[SUM]   0.00-10.00  sec   441 MBytes   370 Mbits/sec                  receiver

D:\Tools\iperf3>iperf3.exe -c ipfire -P 6 -R
Connecting to host ipfire, port 5201
[SUM]   0.00-10.00  sec   493 MBytes   413 Mbits/sec   63             sender
[SUM]   0.00-10.00  sec   492 MBytes   413 Mbits/sec                  receiver

When I bought the wrt, I was hoping that the speed would go up to 1300MBit, my impression is that I have 3 x 450Mbit / s max, is that right?

yuhhaurlin, 2017-01-21:
Intel 7260ac is 2x2. What is your channel setting?

jbsky, 2017-01-21:
channel 36.

[hostapd.conf.txt](https://github.com/kaloz/mwlwifi/files/721314/hostapd.conf.txt)
[image]

yuhhaurlin, 2017-01-21:
http://www.dell.com/Support/Article/tw/en/twbsdt1/QNA43752

closed 2017-01-21 by yuhhaurlin

jbsky, 2017-01-22:
Okay, you're right. However, I remember that with the qualcomm cards, with the ath9k driver, in 300MBit/s max, I got in the 20MBytes/s with the SMB protocol, that is 150MBits/s real with the iptables in operation. So, let's say that the maximum throughput is 50%, today I am at 40Mbytes/s with SMB protocol, that is, 320Mbits/s and without the iptables. It seems to me that you still have a margin of progression (+ 10MBytes/s).

Thank you for your attention, hoping for a further improvement of this driver.

jbsky, 2017-07-13:
I have a problem since several versions and performance has been similar for a while.
[image]

iperf3.exe -c SRV1  -P 6
[SUM]   0.00-10.00  sec   488 MBytes   409 Mbits/sec                  sender
[SUM]   0.00-10.00  sec   488 MBytes   409 Mbits/sec                  receiver

iperf3.exe -c SRV1  -P 6 -R
[SUM]   0.00-10.00  sec   322 MBytes   270 Mbits/sec  1145             sender
[SUM]   0.00-10.00  sec   318 MBytes   267 Mbits/sec                  receiver

iperf3.exe -c WRT  -P 6 
[SUM]   0.00-10.00  sec   102 MBytes  85.5 Mbits/sec                  sender
[SUM]   0.00-10.00  sec   101 MBytes  84.8 Mbits/sec                  receiver

iperf3.exe -c WRT  -P 6 -R
[SUM]   0.00-10.00  sec   320 MBytes   269 Mbits/sec   15             sender
[SUM]   0.00-10.00  sec   318 MBytes   266 Mbits/sec                  receiver

I have a significant data rate drop from PC1 to WRT1900AC.

jbsky, 2017-07-13:
Do I need to open a new ticket @yuhhaurlin ?

yuhhaurlin, 2017-07-13:
I can't tell what the problem you encounter. We did not have this kind of problem. Sorry

jbsky, 2017-07-17:
I finally find that the problem does not necessarily come from the driver but from other element of the router as the option CONFIG_ARM_MVEBU_V7_CPUIDLE which impacts the performance of wifi.