kaloz/mwlwifi #137: rewritten public history

seragh, 2017-01-16:
Commit 88550a0261ac9718eda7f28022d239c0053e2f94 (20170110 release) no longer exists in the repository and got replaced with 6bfaf15d36e8de93704692321b8aeabe171e4f03

```
Fetching origin
remote: Counting objects: 2, done.
remote: Compressing objects: 100% (2/2), done.
remote: Total 2 (delta 0), reused 0 (delta 0), pack-reused 0
Unpacking objects: 100% (2/2), done.
From https://github.com/kaloz/mwlwifi
 + 88550a0...6bfaf15 master     -> origin/master  (forced update)
```

yuhhaurlin, 2017-01-16:
Yes. I only correct commit message, The code is the same as before. Thanks.

closed 2017-01-16 by yuhhaurlin

seragh, 2017-01-16:
Thanks for confirming that it was intentional.

Note, this breaks all those which already mapped the release to the now gone commit in their build scripts and as such should be avoided in future.

yuhhaurlin, 2017-01-17:
@seragh I use rebase to correct commit message. And I think this is the only way to do that. However, I will make sure this kind of thing won't happen again. Thanks.