kaloz/mwlwifi #139: WRT1900ACS v2

fabianborg, 2017-02-15:
I am experience very poor wifi signal strengths with the updated firmware on Linksys WRT1900ACS v2, (LEDE 17.01.0-rc2) I am not sure if this is a bug. 2.4Ghz band has an RSSI of -67 and the 5ghz band is even worse with an RSSI of -75 at 5 meters line of sight distance from the router.

yuhhaurlin, 2017-02-16:
Only happened for specific version? Can you check README and check if your device is with device power table or not?

fabianborg, 2017-02-16:
This is the dump from my mwlwifi info file

<pre>
root@LEDE:~# cat /sys/kernel/debug/ieee80211/phy0/mwlwifi/info

driver name: mwlwifi
chip type: 88W8864
hw version: 7
driver version: 10.3.2.0-20170110
firmware version: 0x0702091a
power table loaded from dts: no
firmware region code: 0x30
mac address: 00:25:9c:14:1f:c0
2g: disable
5g: enable
antenna: 4 4
irq number: 104
iobase0: e0e00000
iobase1: e0f80000
tx limit: 768
rx limit: 64
ap macid support: 0000ffff
sta macid support: 00010000
macid used: 00000001
qe trigger number: 6748589
</pre>

Power table is not loaded from dts.

yuhhaurlin, 2017-02-16:
Your device is programmed with device power table, firmware of 88W8864 will not allow driver to set power table. The tx power should be all right and controlled by firmware of 88W8864.

fabianborg, 2017-02-16:
How can I improve the signal strengths please. I just ran this test literally by the router and as you can see the signals are still weak from such a very close range.

[image]

fabianborg, 2017-02-16:
I changed the tx power to auto and got a slight improvement on the 5g band, 2.4g is still the same:

[image]

bbogey, 2017-02-16:
I have WRT1900ACS (v1).  It has "power table loaded from dts: yes." (same for 2g)
Can you describe more how to load the power table and what power table I should use?

```
root@lede:/etc/config# cat /sys/kernel/debug/ieee80211/phy0/mwlwifi/info

driver name: mwlwifi
chip type: 88W8864
hw version: 7
driver version: 10.3.0.18-20160823-1
firmware version: 0x0702091a
power table loaded from dts: yes
firmware region code: 0x0
mac address: 00:25:9c:13:b5:04
2g: disable
5g: enable
antenna: 4 4
irq number: 104
iobase0: e1180000
iobase1: e1300000
tx limit: 768
rx limit: 64
ap macid support: 0000ffff
sta macid support: 00010000
macid used: 00000001
qe trigger number: 318140744

```

yuhhaurlin, 2017-02-17:
@fabianborg If device is programmed with device power table, there is no way to change tx power via driver. I think stock firmware should have the same issue. Maybe you should check with belkin to see if the device power table is programmed correctly or not.

yuhhaurlin, 2017-02-17:
@bbogey You can find parameter <marvell, powertable> in the related device DTS file. If you can't find it, you need to check with LEDE or OpenWrt community to see how to build in this parameter.

bbogey, 2017-02-17:
It is there, 
https://git.lede-project.org/?p=source.git;a=blob_plain;f=target/linux/mvebu/patches-4.4/002-add_powertables.patch;hb=HEAD

And from build dir I see it was patched ok.

yuhhaurlin, 2017-02-18:
It looks like there is no issue for driver. I will close this one.

closed 2017-02-18 by yuhhaurlin