kaloz/mwlwifi #140: The channel utilisation and noise statistics are not properly displayed by ethtool and the ethstat plugin of collectd

ignisf, 2017-02-25:
Hello,

I just updated to mwlwifi 10.3.2.0-20170110 and I experienced a regression in my monitoring. It seems that the ACS-related changes have broken the channel utilisation and noise statistics in ethtool.

Platform: LEDE SNAPSHOT, r3577-3e41afd on a Linksys WRT1900ACS

```
[    8.063138] <<Marvell 802.11ac Wireless Network Driver version 10.3.2.0-20170110>>

root@acs:~# ethtool -S wlan0
NIC statistics:
     rx_packets: 649
     rx_bytes: 72859
     rx_duplicates: 0
     rx_fragments: 645
     rx_dropped: 0
     tx_packets: 303
     tx_bytes: 118439
     tx_filtered: 0
     tx_retry_failed: 0
     tx_retries: 0
     sta_state: 0
     txrate: 0
     rxrate: 0
     signal: 0
     channel: 0
     noise: 18446744073709551615
     ch_time: 18446744073709551615
     ch_time_busy: 18446744073709551615
     ch_time_ext_busy: 18446744073709551615
     ch_time_rx: 18446744073709551615
     ch_time_tx: 18446744073709551615
```

```
[    7.045438] <<Marvell 802.11ac Wireless Network Driver version 10.3.2.0-20161222>>

root@acs:~# ethtool -S wlan0
NIC statistics:
     rx_packets: 623
     rx_bytes: 80969
     rx_duplicates: 0
     rx_fragments: 615
     rx_dropped: 0
     tx_packets: 331
     tx_bytes: 133663
     tx_filtered: 0
     tx_retry_failed: 0
     tx_retries: 0
     sta_state: 0
     txrate: 0
     rxrate: 0
     signal: 0
     channel: 5180
     noise: 168
     ch_time: 243316
     ch_time_busy: 1644
     ch_time_ext_busy: 18446744073709551615
     ch_time_rx: 18446744073709551615
     ch_time_tx: 1551
```

ignisf, 2017-02-25:
As a side note, `iw wlan0 survey dump` works fine on 20170110:

```
root@acs:~# iw wlan0 survey dump
Survey data from wlan0
	frequency:			5180 MHz [in use]
	noise:				-90 dBm
	channel active time:		1061801 ms
	channel busy time:		19736 ms
	channel transmit time:		12065 ms

root@acs:~# iw wlan0 scan
root@acs:~# iw wlan0 survey dump
Survey data from wlan0
	frequency:			5180 MHz
	noise:				-90 dBm
	channel active time:		24169 ms
	channel busy time:		260 ms
	channel transmit time:		173 ms
Survey data from wlan0
	frequency:			5200 MHz
	noise:				-90 dBm
	channel active time:		1524 ms
	channel busy time:		16 ms
	channel transmit time:		0 ms
Survey data from wlan0
	frequency:			5220 MHz
	noise:				-90 dBm
	channel active time:		1285 ms
	channel busy time:		13 ms
	channel transmit time:		0 ms
Survey data from wlan0
	frequency:			5240 MHz
	noise:				-90 dBm
	channel active time:		1305 ms
	channel busy time:		14 ms
	channel transmit time:		0 ms
Survey data from wlan0
	frequency:			5260 MHz
	noise:				-90 dBm
	channel active time:		180 ms
	channel busy time:		3 ms
	channel transmit time:		0 ms
Survey data from wlan0
	frequency:			5280 MHz
	noise:				-94 dBm
	channel active time:		220 ms
	channel busy time:		3 ms
	channel transmit time:		0 ms
Survey data from wlan0
	frequency:			5300 MHz
	noise:				-94 dBm
	channel active time:		219 ms
	channel busy time:		2 ms
	channel transmit time:		0 ms
Survey data from wlan0
	frequency:			5320 MHz
	noise:				-94 dBm
	channel active time:		219 ms
	channel busy time:		2 ms
	channel transmit time:		0 ms
Survey data from wlan0
	frequency:			5500 MHz
	noise:				-94 dBm
	channel active time:		219 ms
	channel busy time:		2 ms
	channel transmit time:		0 ms
Survey data from wlan0
	frequency:			5520 MHz
	noise:				-94 dBm
	channel active time:		220 ms
	channel busy time:		2 ms
	channel transmit time:		0 ms
Survey data from wlan0
	frequency:			5540 MHz
	noise:				-94 dBm
	channel active time:		220 ms
	channel busy time:		2 ms
	channel transmit time:		0 ms
Survey data from wlan0
	frequency:			5560 MHz
	noise:				-94 dBm
	channel active time:		219 ms
	channel busy time:		2 ms
	channel transmit time:		0 ms
Survey data from wlan0
	frequency:			5580 MHz
	noise:				-94 dBm
	channel active time:		220 ms
	channel busy time:		2 ms
	channel transmit time:		0 ms
Survey data from wlan0
	frequency:			5600 MHz
	noise:				-94 dBm
	channel active time:		219 ms
	channel busy time:		2 ms
	channel transmit time:		0 ms
Survey data from wlan0
	frequency:			5620 MHz
	noise:				-94 dBm
	channel active time:		220 ms
	channel busy time:		2 ms
	channel transmit time:		0 ms
Survey data from wlan0
	frequency:			5640 MHz
	noise:				-94 dBm
	channel active time:		219 ms
	channel busy time:		2 ms
	channel transmit time:		0 ms
Survey data from wlan0
	frequency:			5660 MHz
	noise:				-94 dBm
	channel active time:		219 ms
	channel busy time:		2 ms
	channel transmit time:		0 ms
Survey data from wlan0
	frequency:			5680 MHz
	noise:				-95 dBm
	channel active time:		220 ms
	channel busy time:		2 ms
	channel transmit time:		0 ms
Survey data from wlan0
	frequency:			5700 MHz
	noise:				-95 dBm
	channel active time:		220 ms
	channel busy time:		2 ms
	channel transmit time:		0 ms
Survey data from wlan0
	frequency:			5745 MHz
	noise:				-95 dBm
	channel active time:		219 ms
	channel busy time:		2 ms
	channel transmit time:		0 ms
Survey data from wlan0
	frequency:			5765 MHz
	noise:				-92 dBm
	channel active time:		179 ms
	channel busy time:		4 ms
	channel transmit time:		0 ms
Survey data from wlan0
	frequency:			5785 MHz
	noise:				-92 dBm
	channel active time:		170 ms
	channel busy time:		2 ms
	channel transmit time:		0 ms
Survey data from wlan0
	frequency:			5805 MHz
	noise:				-92 dBm
	channel active time:		170 ms
	channel busy time:		2 ms
	channel transmit time:		0 ms
Survey data from wlan0
	frequency:			5180 MHz
	noise:				-92 dBm
	channel active time:		169 ms
	channel busy time:		2 ms
	channel transmit time:		0 ms
root@acs:~#
```

yuhhaurlin, 2017-02-25:
I had tested via "iwlist wlan0 survey dump". The code is working.

closed 2017-02-25 by yuhhaurlin

yuhhaurlin, 2017-02-25:
BTW, from debug log of ACS of hostapd, it also shows the code is working.

ignisf, 2017-02-25:
My issue is not with `iw survey dump` but with `ethtool` not displaying sane data after the changes.

yuhhaurlin, 2017-02-25:
I have no idea about ethtool.

ignisf, 2017-02-25:
I still consider this a regression and think that this issue should be reopened. In the current state it seems that the driver is not playing well with mac80211's [ethtool hooks](http://lxr.free-electrons.com/source/net/mac80211/ethtool.c#L64).

yuhhaurlin, 2017-02-25:
ethtool is used to query and control network device driver and hardware settings, particularly for wired Ethernet devices.

ignisf, 2017-02-25:
> ethtool is used to query and control network device driver and hardware settings, particularly for wired Ethernet devices.

I am not quite sure what you're trying to say here.

Yes, this is the description of ethtool from it's project page at kernel.org. It however no longer seems to be up to date, as ethtool support has been integrated into `cfg80211` and `mac80211`. Other Wi-Fi NIC drivers use it as a convenient way to provide statistics about the operation of the device.

This driver worked well with ethtool, too until a few commits ago. I suspect it's a simple fix, however I haven't been able to figure out the exact reason why it stopped working to submit a PR.

yuhhaurlin, 2017-02-25:
I think I will only make sure iw and hostapd are working well.

ignisf, 2017-07-28:
https://github.com/kaloz/mwlwifi/commit/245005057a318f689fbe5d2c58b0eef2fb7d91e9 is to blame for future reference