kaloz/mwlwifi #141: BSSID mask Modification on Marvell 88W8864

nelcard, 2017-03-06:
Hi.

I actually working on a project and we are using Linksys WRT1900AC with Marvell 88W8864 WLAN Hardware.

We have a question for you:

In Atheros Chipsets there are the possibility to set the bssid mask through ath_hw_setbssidmask function (http://lxr.free-electrons.com/source/drivers/net/wireless/ath/hw.c). Basically, it enables the WLAN hardware to support multiple BSSes on the same Access Point.

Is there this possibility in Marvell 88W8864 WLAN Hardware?

Thanks in advance.

Chadster766, 2017-03-06:
The current driver supports multiple SSID's. They only need to be configured in the hostapd wlan.conf file.

nelcard, 2017-03-06:
Thanks for your quick answer.

One more question.

Does the current driver support multiple BSSID?.

Thanks in advance.

Chadster766, 2017-03-06:
Below is an example config file that hopefully will answer you question since I'm not sure what the difference is between an SSID and a BSSID:

```
root@MCDEBIAN:~# cat /etc/hostapd/wlan0.conf
interface=wlan0
bridge=br0
driver=nl80211
ctrl_interface=/var/run/hostapd
ignore_broadcast_ssid=0
hw_mode=a
channel=36
wmm_enabled=1
ieee80211n=1
ht_capab=[LDPC][HT40+][SHORT-GI-20][SHORT-GI-40]
vht_capab=[RXLDPC][SHORT-GI-80][RX-STBC-1][SU-BEAMFORMER][SU-BEAMFORMEE][MAX-A-MPDU-LEN-EXP7][RX-ANTENNA-PATTERN][TX-ANTENNA-PATTERN]
ieee80211ac=1
vht_oper_chwidth=1
vht_oper_centr_freq_seg0_idx=42
country_code=US

#Corp Wireless
ssid=xxxx_AC

auth_algs=1
wpa=2
wpa_passphrase=xxxxCorpWifi
wpa_key_mgmt=WPA-PSK
wpa_pairwise=TKIP
rsn_pairwise=CCMP

#Guest Wireless
bss=wlan0_4
bssid=5e:95:e1:46:52:6d
bridge=br1
ssid=xxxx_AC_Guest

auth_algs=1
wpa=2
wpa_passphrase=xxxx-Guest
wpa_key_mgmt=WPA-PSK
wpa_pairwise=TKIP
rsn_pairwise=CCMP
root@MCDEBIAN:~#
```

closed 2017-03-06 by nelcard