kaloz/mwlwifi #144: 88W8864 5GHZ Disconnect Issue

shauno100, 2017-04-05:
Is it a known issue with the 88W8864 5GHZ driver that devices connected to 5GHZ when the radio is scheduled to turn off do not disconnect and actually stay connected? I have tried stock, DD-WRT and LEDE firmwares and i have the same issue with all which is obviously due to the driver.

My router is a Linksys WRT1900ACSv2

The 2.4GHZ does not have this issue, devices connected to 2.4GHZ when the radio is switched off by the scheduler are disconnected immediately.

When both radios are meant to be switched off doing a wifi analysis from another device shows the 5ghz is still broadcasting when the schedule has meant to turn off the 5GHZ radio.

To replicate

1. Use a device with 88w8864 chip (im using wrt1900acsv2)
2. Connect a device to 5ghz band
3. Set wireless schedule to turn off both 2.4ghz and 5ghz (linksys stock or custom dd wrt script)
4. When schedule takes affect you will notice the device is still connected to 5ghz  and using wifi analyzer on other device shows 5ghz is still broadcasting whilst 2.4ghz is turned off successfully.

As mentioned if a device was connected to 2.4ghz instead the issue does not occur and the device would be disconnected immediately and both 2.4 and 5ghz bands would be turned off as expected.

shauno100, 2017-04-05:
to sum up the above the only way the 5GHZ radio will turn off successfully is if no device is connected to 5GHZ at the time the radio is scheduled to switch off

BrainSlayer, 2017-04-05:
@shauno100 one question from me. (i dont know your script) if its off, but still shown as connected, is it still working or could it be that the client adapter just hangs on that connection without getting any response. such issues may also depend on the client driver here. the linksys original driver and mwlwifi which is used in dd-wrt is completelly different

shauno100, 2017-04-05:
@BrainSlayer Sorry mate i should have known that Marvell would have developed a separate driver for the stock firmwares (closed source), I was thinking they all used the same one. Anyways i'm experiencing the issue with my Galaxy S7 Edge. I never had the issue before on my previous RTAC68U router so can only pinpoint that the issue is to do with the Linksys WRT1900ACSv2.

When wireless schedule turns off the radios and the phone is connected to 5GHZ the phone remains connected to 5GHZ with status "Connected No Internet" so the internet part disconnects but the device remains connected to the radio and the 5GHZ is still broadcasting.

The exact same happened when i was using DD-WRT. As for the wireless schedule in DD WRT i cannot find any script to turn off the wireless in DD WRT for Marvell routers so i was using the one i used on my Archer C7 (atheros) which seemed to work if nothing was connected to 5GHZ and would turn off both radios successfully but if a device was connected to 5GHZ the 5GHZ would remain connected whilst the 2.4GHZ successfully turned off. The script is below.

#!/bin/sh 
case "$1" in 
  wifioff) 
    stopservice wland 
    nvram set ath0_net_mode=disabled
   nvram set ath1_net_mode=disabled 
    ifconfig ath0 down 
    ifconfig ath1 down 
    killall hostapd 
    ;; 

  wifion) 
    stopservice lan 
    nvram set ath0_net_mode=ng-only
    nvram set ath1_net_mode=ac-only
    startservice lan 
    iwconfig ath0 txpower 25
    iwconfig ath1 txpower 25
    ;; 

  *) 
    echo "$0 {wifioff | wifion}" 
esac 

Maybe i'm way off and using the wrong script, let me know if its incorrect. I will do more testing tonight and will post more of my findings here.

BrainSlayer, 2017-04-06:
your script may not work. you need also to call stopservice lan / startservice lan in the stop procedure, otherwise the interfaces will persist and wont get disabled (you may also do that with the iw tool manually)

shauno100, 2017-04-06:
@BrainSlayer No worries thanks. I will flash your latest build shortly and amend the script and do some more testing.

shauno100, 2017-04-06:
@BrainSlayer Have done further testing after installing your latest 31791 and can report the below

1. When my Samsung Galaxy S7 is connected to 5GHZ when both the 2.4GHZ and 5GHZ radios are switched off as per schedule both radios **do** actually **turn off** successfully, have confirmed this running a wifi analyser from my tablet which shows both radios are turned off but the Galaxy S7 remains connected for some reason. My mistake when i mentioned earlier the 5GHZ stayed on.

2. Connected my tablet to 5GHZ as another test and turned off both radios, both radios turn off and my tablet immediately disconnected.

Looks like its a driver issue with my Galaxy S7 instead? very strange as i never has this issue with my previous routers, the Galaxy S7 would disconnect immediately from 5GHZ when the radio was turned off.

Is there anything different in Marvell's driver to as an example Broadcoms driver as used in an Asus AC68U when handling device disconnections that could have an impact on some devices  (galaxy S7) not dropping the connection?

As mentioned its very strange that the Galaxy S7 did not have this problem with other routers (5GHZ) and this issue does not occur when the S7 is connected to 2.4GHZ on WRT1900ACSv2 only when connected to 5GHZ.

I will close this ticket, but just curious to your thoughts on this.

shauno100, 2017-04-06:
@BrainSlayer Also forgot to mention, might be a bug with DD WRT? when my wifi is turned back on as per schedule with the script mentioned above the Wireless>Basic Settings shows the "Wireless Network Mode" for both 5GHZ and 2.4GHZ as being disabled and the channel widths are reset to 20MHZ for both bands, the channels i have also selected are reset to "Auto"

It seems like its not keeping the wireless settings prior to the wifi being turned off and back on with the script. Let me know if you need to raise a ticket on TRAC.

I have attached a screenshot showing the settings being reset.

[image]

yuhhaurlin, 2017-04-14:
This is not problem of driver. I will close it.

closed 2017-04-14 by yuhhaurlin